juj/fbcp-ili9341 #1: ili9486

artzxc, 2018-01-09:
Hi
I have a 3.5LCD ili9486(Waveshare 3.5inch RPi LCD (B)) and I tried to make your program work on it.
I changed the initialization to this:
```
void InitILI9341()
{
    BEGIN_SPI_COMMUNICATION();
  {
        SPI_TRANSFER(0xb0/*Interface Mode Control*/, 0x0);
        SPI_TRANSFER(0x11/*Sleep Out*/);
        usleep(120 * 1000);
        SPI_TRANSFER(0x3A, 0x55/*use 16 bits per pixel color*/);
        SPI_TRANSFER(0x36, 0x48/*MX, BGR == rotation 0*/);
        SPI_TRANSFER(0xC2, 0x44/*Power Control 3*/);
        SPI_TRANSFER(0xC5, 0x00, 0x00, 0x00, 0x00/*VCOM Control 1*/);
        SPI_TRANSFER(0xE0/*Positive Gamma Correction*/, 0x0F, 0x1F, 0x1C, 0x0C, 0x0F, 0x08, 0x48, 0x98, 0x37, 0x0A, 0x13, 0x04, 0x11, 0x0D, 0x00);
        SPI_TRANSFER(0xE1/*Negative Gamma Correction*/, 0x0F, 0x32, 0x2E, 0x0B, 0x0D, 0x05, 0x47, 0x75,0x37, 0x06, 0x10, 0x03, 0x24, 0x20, 0x00);
        SPI_TRANSFER(0xE2/* Digital Gamma Control 1 */, 0x0F, 0x32, 0x2E, 0x0B, 0x0D, 0x05, 0x47, 0x75, 0x37, 0x06, 0x10, 0x03, 0x24, 0x20, 0x00);
        SPI_TRANSFER(0x11/*Sleep Out*/);
        usleep(120 * 1000);
        SPI_TRANSFER(/*Display ON*/0x29);

    // Some wonky effects to try out:
//    SPI_TRANSFER(0x20/*Display Inversion OFF*/);
//    SPI_TRANSFER(0x21/*Display Inversion ON*/);
//    SPI_TRANSFER(0x38/*Idle Mode OFF*/);
//    SPI_TRANSFER(0x39/*Idle Mode ON*/); // Idle mode gives a super-saturated high contrast reduced colors mode

    // Since we are doing delta updates to only changed pixels, clear display initially to black for known starting state
    for(int y = 0; y < DISPLAY_HEIGHT; ++y)
    {
      SPI_TRANSFER(DISPLAY_SET_CURSOR_X, 0, 0, DISPLAY_WIDTH >> 8, DISPLAY_WIDTH & 0xFF);
      SPI_TRANSFER(DISPLAY_SET_CURSOR_Y, y >> 8, y & 0xFF, DISPLAY_HEIGHT >> 8, DISPLAY_HEIGHT & 0xFF);
      SPITask *clearLine = AllocTask(SCANLINE_SIZE);
      clearLine->cmd = DISPLAY_WRITE_PIXELS;
      memset(clearLine->data, 0, clearLine->size);
      CommitTask(clearLine);
      RunSPITask(clearLine);
      DoneTask(clearLine);
    }
    SPI_TRANSFER(DISPLAY_SET_CURSOR_X, 0, 0, DISPLAY_WIDTH >> 8, DISPLAY_WIDTH & 0xFF);
    SPI_TRANSFER(DISPLAY_SET_CURSOR_Y, 0, 0, DISPLAY_HEIGHT >> 8, DISPLAY_HEIGHT & 0xFF);
  }

  END_SPI_COMMUNICATION();
}
```
And the screen turns all white.
I'm a novice at these things . Could you help me ?

juj, 2018-02-15:
Hmm, not quite sure, that controller is not familiar to me either. Are you able to find links to data sheets or SPI protocol descriptions? Perhaps some other project like notro's fbtft might have clues as to what the initialization routines would need to be.

artzxc, 2018-02-25:
@juj 
Thanks for your reply. I am still trying. I bought a HDMI screen as a stopgap measure. If I can make any progress I will write it here.
Thank you again.
Happy Chinese New Year !

juj, 2018-04-21:
Added support to Waveshare32b display in https://github.com/juj/fbcp-ili9341/commit/3a98d0da2c65766d54d19786acdcf6c5c6a1665e. Perhaps that might help towards supporting Waveshare35b as well.

closed 2018-05-15 by juj

juj, 2018-05-15:
Recently got a Waveshare35b display, [this one](https://www.amazon.co.uk/Waveshare-Raspberry-Touch-Screen-Model/dp/B01N48NOXI?th=1), and added support for that display, so it should work now.

Implementing support for this was a curious experience. Going into the project, I was thinking it will be twice as slow compared to my Adafruit 320x240 2.8" Pi TFT, since the Waveshare one has twice the pixels, but that ended up being only part of the truth.

The realization was that whereas my Adafruit ILI9341 display can be driven at a whopping [73.5MHz](https://github.com/juj/fbcp-ili9341/blob/master/ili9341.h#L16) which gives a 73.5mbit/second pixel update rate, the Waveshare 3.5B already gets out of sync at 32.50MHz, and the best update rate I was able to drive it was [32.25MHz](https://github.com/juj/fbcp-ili9341/blob/master/waveshare35b.h#L12), i.e. 32.25mbit/second. That is, the Adafruit 2.8" PiTFT display can be driven 2.279x faster than the Waveshare35b one.

As result, twice the pixels and half the speed means that the Adafruit display is about 4 times faster than the Waveshare35b display, so it is quite constrained and Waveshare35b can get nowhere near 60fps on complex content. In simple NES emulators though 

I am running with the following `/boot/config.txt` file:

```
hdmi_group=2
hdmi_mode=87
hdmi_cvt=480 320 60 1 0 0 0
hdmi_force_hotplug=1

core_freq=258
force_turbo=1
boot_delay=1

dtparam=audio=on
audio_pwm_mode=2
```

That is, no dtoverlays for waveshare35b and no system spi driver enabled. Having `dtparam=audio=on` is important for some reason if you are running Retropie/Emulationstation, since otherwise emulators begin to periodically stutter, not sure what that is about. The `core_freq` line there is to tune the SPI bus speed to the maximum speed that the display can handle.

juj, 2018-05-26:
Published https://www.youtube.com/watch?v=dqOLIHOjLq4 as a demo video about ILI9486 support.

commit 842eceab (2018-05-15): Add support for ILI9486. Closes #1.