juj/fbcp-ili9341 #4: what about that DSI connector?

laurentopia, 2018-05-05:
it's got an awful lot of pins. is it fed by the GPU? if it'd work, then, no HDMI "hump"

juj, 2018-05-07:
The DSI connector, utilizing ["MIPI DSI"](https://www.mipi.org/specifications/dsi) that exists on the Pi is indeed fed directly by the GPU, via a closed source driver. The benefit of DSI is that it is direct from GPU, just like HDMI is, but with a limitation that there exists only one display product, [The Raspberry Pi Official 7" Touchscreen Display](https://thepihut.com/products/official-raspberry-pi-7-touchscreen-display) that it works with, i.e. [it is not possible to connect an arbitrary DSI display to that connector](https://www.raspberrypi.org/forums/viewtopic.php?t=188908). Some interesting notes from [a blog at launch time can be found here](http://raspi.tv/2015/new-official-raspberry-pi-dsi-screen-launches). Configuration info [here](https://www.raspberrypi.org/documentation/hardware/display/README.md).

PDFs of the MIPI DSI protocol have leaked online and [can be found in Chinese sites](http://bfiles.chinaaet.com/justlxy/blog/20171114/1000019445-6364627609238902374892404.pdf), but due to the closed source nature of the Pi GPU driver, I believe nobody has had success in making Pi interface with any other DSI displays.

So for your use case, if the official 7" display works out, definitely recommend using that, in which case the fbcp-ili9341 driver is not needed.

In addition to MIPI DSI, there does exist a third method, called the DPI connection/protocol, that can be used on the Pi. DSI and DPI are two completely different protocols, not to be confused with each other. DPI is kind of the same as SPI, but with the change of going wider in parallel with several GPIO pins, rather than just one serial pin that is used with SPI. Many controllers that support SPI also support wiring via DPI (although many are sold with just SPI pins exposed). The DPI connection is also directly managed by the GPU, and one utilizes the closed source driver blob to feed the display. There are good resources around for DPI, by both Ben Heck Show and Adafruit:
 - [Ben Heck Show: sudo Sergeant 17 - Raspberry Pi No HDMI Revisited GPIO](https://www.youtube.com/watch?v=tCHyiY23vPY)
 - [Ben Heck Show: Rasp Pi No HDMI - Portable MAME Part 1](https://www.youtube.com/watch?v=x0EGnydxxf4)
 - [Adafruit: Protos arrived! Testing out the new DPI Display Kippah prototype for the @Raspberry_Pi #raspberrypi](https://www.youtube.com/watch?v=91eVA3IsiD0)
 - [Adafruit: Testing raw DPI display capability on the Raspberry Pi 2, Model B and B+](https://www.youtube.com/watch?v=DTuvzxkVaKs)
 - [Raspberry Pi DPI wiring documentation](https://www.raspberrypi.org/documentation/hardware/raspberrypi/dpi/)
 - [Raspberry Pi forums on using DPI](https://www.raspberrypi.org/forums/viewtopic.php?t=133157)

A drawback with DPI is that it uses quite many of the GPIO pins, so there may be little room left to connect other peripherals. If that is not a problem, and the 7" official DSI display does not raise your fancy, then perhaps wiring in a DPI display will be the way to go. In DPI case as well, the fbcp-ili9341 driver will not play a role, so fbcp-ili9341 is only really needed for SPI displays.

juj, 2018-05-07:
Heh, yeah, I suppose it's always about finding the method where the downsides are the easiest ones to handle. Anyhow, let's mark this one resolved.

closed 2018-05-07 by juj