juj/fbcp-ili9341 #8: RGB to BGR

UtopiaEditorial, 2018-06-04:
Hi juj,
First, thank you for your hard work, I'm enjoying it very much.
Working with Pi Zero went flawless with waveshare 35b.
I have a 4" (a)     (https://www.waveshare.com/wiki/4inch_RPi_LCD_(A))
The fbcp can run the 4"(a) very well, just one problem...colors are inverted (not distorted).
Is there any way i can change something in code to swap bgr to rgb ?
Thanks in advance.
el.

juj, 2018-06-05:
Thanks, much appreciated, and great to hear that it works!

The display inversion aspect is curious. On my WaveShare 3.5" (B) display I needed to tell the ILI9486 display controller to invert colors with the `Display Inversion ON` command, or the colors would be inverted. You can remove that with

```diff
diff --git a/ili9486.cpp b/ili9486.cpp
index b87a768..e1b07f9 100644
--- a/ili9486.cpp
+++ b/ili9486.cpp
@@ -49,7 +49,7 @@ void InitILI9486()
     SPI_TRANSFER(0x11/*Sleep OUT*/);
     usleep(120*1000);
     SPI_TRANSFER(0x3A/*Interface Pixel Format*/, 0x00, 0x55/*DPI(RGB Interface)=16bits/pixel, DBI(CPU Interface)=16bits/pixel*/); // This can be switched from 0x55 to 0x66 for 18bits/pixel instead.
-    SPI_TRANSFER(0x21/*Display Inversion ON*/);
+//    SPI_TRANSFER(0x21/*Display Inversion ON*/);
     SPI_TRANSFER(0xC0/*Power Control 1*/, 0x00, 0x09, 0x00, 0x09);
     SPI_TRANSFER(0xC1/*Power Control 2*/, 0x00, 0x41, 0x00, 0x00);
     SPI_TRANSFER(0xC5/*VCOM Control*/, 0x00, 0x00, 0x00, 0x36);
```

or even with

```diff
diff --git a/ili9486.cpp b/ili9486.cpp
index b87a768..4f892e4 100644
--- a/ili9486.cpp
+++ b/ili9486.cpp
@@ -49,7 +49,7 @@ void InitILI9486()
     SPI_TRANSFER(0x11/*Sleep OUT*/);
     usleep(120*1000);
     SPI_TRANSFER(0x3A/*Interface Pixel Format*/, 0x00, 0x55/*DPI(RGB Interface)=16bits/pixel, DBI(CPU Interface)=16bits/pixel*/); // This can be switched from 0x55 to 0x66 for 18bits/pixel instead.
-    SPI_TRANSFER(0x21/*Display Inversion ON*/);
+    SPI_TRANSFER(0x20/*Display Inversion OFF*/);
     SPI_TRANSFER(0xC0/*Power Control 1*/, 0x00, 0x09, 0x00, 0x09);
     SPI_TRANSFER(0xC1/*Power Control 2*/, 0x00, 0x41, 0x00, 0x00);
     SPI_TRANSFER(0xC5/*VCOM Control*/, 0x00, 0x00, 0x00, 0x36);
```

this is from ILI9486 datasheet at http://www.datasheetspdf.com/datasheet/ILI9486.html.

If that helps, I wonder what & why the difference is between the WaveShare 4.0" (A) and WaveShare 3.5" (B) displays, and if they both have a ILI9486 (probably yes), and if display inversion aspect is a 4.0" vs 3.5" thing, or if it's rather a (A) vs (B) thing.

Let me know if the above works. If so, perhaps could make that a configuration option to allow inverting the display.

juj, 2018-06-05:
Rereading the title of the bug, if instead of display inversion this is a RGB vs BGR ordering issue, then it might be fixable with

```diff
diff --git a/ili9486.cpp b/ili9486.cpp
index b87a768..30c7c10 100644
--- a/ili9486.cpp
+++ b/ili9486.cpp
@@ -60,7 +60,7 @@ void InitILI9486()
 #define MADCTL_ROW_ADDRESS_ORDER_SWAP (1<<7)

 #define MADCTL_ROTATE_180_DEGREES 0xC0
-    uint8_t madctl = MADCTL_BGR_PIXEL_ORDER;
+    uint8_t madctl = 0;
 #ifdef DISPLAY_ROTATE_180_DEGREES
     madctl |= MADCTL_ROTATE_180_DEGREES;
 #endif
```

Perhaps that is the root cause instead?

UtopiaEditorial, 2018-06-05:
Hi juj,
Thank you so much for replying.
I’m not an expert…but i think the “problem” is the waveshare  displays - they state that the 35b and 4” (a) are hardware compatible, but if you use there’s dtoverlay -  for the 4” you need to use the 35a dtoverlay…and if you try to use the 35b dtoverlay, the “problem” popup again - inverted colors !
so that’s why i needed your help.
Thank you again.
Will try this fix tonight !
Bye’
el.

koreastardaily, 2018-06-06:
Hi,

I got a waveshare 3.5" (a) LCD and experienced the same problem.
Just want to let you know that by removing "SPI_TRANSFER(0x21/*Display Inversion ON*/)" does fix the problem, thanks.

UtopiaEditorial, 2018-06-06:
Fixed !

-    SPI_TRANSFER(0x21/*Display Inversion ON*/);
+    SPI_TRANSFER(0x20/*Display Inversion OFF*/);

Good to know that waveshare's Line A Lcd's is supported.
Thank you.

juj, 2018-06-11:
Added CMake options to toggle BGR vs RGB and Display Inversion and a FAQ item to Readme. These options should help more easily configure the output to look correct. Thanks for reporting!

closed 2018-06-11 by juj