juj/fbcp-ili9341 #9: Freeplay Use Full Viewable Area

TheFlav, 2018-06-07:
[image]

I feel like I might not be understanding this correctly, but I'd like to use the full "viewable" area of the Freeplay Zero (302x202).

I'd also like to set the HDMI resolution to 1024x768, because it works for most HDMI TVs/Monitors.  This way the HDMI output can be hot pluggable while the LCD remains fully usable.

Today we tried 1024x768 and 320x240 HDMI resolution on the Freeplay Zero.  As shown in the attached photo, the LCD gets sized to (I think) 269x202 which doesn't use the full viewable 302x202.

I understand that 269/202 = 1.3 just like 1024/768 (or 320/240), but I would prefer to stretch this output to 302x202 and allow the emulator to do the scaling.  That way the HDMI output and the LCD output will match and use the entire viewable area.

Can we attain this?  If so, what settings should I use to set it up?

juj, 2018-06-08:
> the LCD gets sized to (I think) 269x202 which doesn't use the full viewable 302x202.

The code in https://github.com/juj/fbcp-ili9341/blob/master/gpu.cpp#L396 performs an aspect ratio preserving scale, that does not stretch width/height disproportionately. The output in the photo looks correct and expected from that behavior viewpoint.

If you'd like to do an aspect ratio ignoring scale, you can try removing the computation of the single scaling factor that is applied equally to both width and height, and apply separate scaling factors to width and height. Something like following: (I'm on the road, so did not actually test this, off the top of my head)

```diff
diff --git a/gpu.cpp b/gpu.cpp
index 112fd52..d8b29e8 100644
--- a/gpu.cpp
+++ b/gpu.cpp
@@ -337,7 +337,8 @@ void InitGPU()
   displayYOffset = 0;
   int scaledWidth = DISPLAY_DRAWABLE_WIDTH;
   int scaledHeight = DISPLAY_DRAWABLE_HEIGHT;
-  double scalingFactor = 1.0;
+  double scalingFactorWidth = 1.0;
+  double scalingFactorHeight = 1.0;

   // Often it happens that the content that is being rendered already has black letterboxes/pillarboxes if it was produced for a different aspect ratio than
   // what the current HDMI resolution is. However the current HDMI resolution might not be in the same aspect ratio as DISPLAY_DRAWABLE_WIDTH x DISPLAY_DRAWABLE_HEIGHT.
@@ -393,25 +394,18 @@ void InitGPU()
   int relevantDisplayHeight = (int)(display_info.height * (1.0 - overscanTop - overscanBottom) + 0.5);
   printf("Relevant source display area size with overscan cropped away: %dx%d.\n", relevantDisplayWidth, relevantDisplayHeight);

-  if (DISPLAY_DRAWABLE_WIDTH * relevantDisplayHeight < DISPLAY_DRAWABLE_HEIGHT * relevantDisplayWidth)
-  {
-    scaledHeight = (int)((double)DISPLAY_DRAWABLE_WIDTH * relevantDisplayHeight / relevantDisplayWidth + 0.5);
-    scalingFactor = (double)DISPLAY_DRAWABLE_WIDTH/relevantDisplayWidth;
-    displayXOffset = DISPLAY_COVERED_LEFT_SIDE;
-    displayYOffset = DISPLAY_COVERED_TOP_SIDE + (DISPLAY_DRAWABLE_HEIGHT - scaledHeight) / 2;
-  }
-  else
-  {
-    scaledWidth = (int)((double)DISPLAY_DRAWABLE_HEIGHT * relevantDisplayWidth / relevantDisplayHeight + 0.5);
-    scalingFactor = (double)DISPLAY_DRAWABLE_HEIGHT/relevantDisplayHeight;
-    displayXOffset = DISPLAY_COVERED_LEFT_SIDE + (DISPLAY_DRAWABLE_WIDTH - scaledWidth) / 2;
-    displayYOffset = DISPLAY_COVERED_TOP_SIDE;
-  }
+  scaledWidth = DISPLAY_DRAWABLE_WIDTH;
+  scaledHeight = DISPLAY_DRAWABLE_HEIGHT;
+  displayXOffset = DISPLAY_COVERED_LEFT_SIDE;
+  displayYOffset = DISPLAY_COVERED_TOP_SIDE;
+
+  scalingFactorWidth = (double)scaledWidth / relevantDisplayWidth;
+  scalingFactorHeight = (double)scaledHeight / relevantDisplayHeight;

-  excessPixelsLeft = (int)(display_info.width * overscanLeft * scalingFactor + 0.5);
-  excessPixelsRight = (int)(display_info.width * overscanRight * scalingFactor + 0.5);
-  excessPixelsTop = (int)(display_info.height * overscanTop * scalingFactor + 0.5);
-  excessPixelsBottom = (int)(display_info.height * overscanBottom * scalingFactor + 0.5);
+  excessPixelsLeft = (int)(display_info.width * overscanLeft * scalingFactorWidth + 0.5);
+  excessPixelsRight = (int)(display_info.width * overscanRight * scalingFactorWidth + 0.5);
+  excessPixelsTop = (int)(display_info.height * overscanTop * scalingFactorHeight + 0.5);
+  excessPixelsBottom = (int)(display_info.height * overscanBottom * scalingFactorHeight + 0.5);

   gpuFrameWidth = scaledWidth;
   gpuFrameHeight = scaledHeight;
@@ -430,8 +424,8 @@ void InitGPU()
   videoCoreFramebuffer[0] += (gpuFramebufferSizeBytes>>1);
   videoCoreFramebuffer[1] += (gpuFramebufferSizeBytes>>1);

-  syslog(LOG_INFO, "GPU display is %dx%d. SPI display is %dx%d with drawable area of %dx%d. Applying scaling factor %.2fx, xOffset: %d, yOffset: %d, scaledWidth: %d, scaledHeight: %d", display_info.width, display_info.height, DISPLAY_WIDTH, DISPLAY_HEIGHT, DISPLAY_DRAWABLE_WIDTH, DISPLAY_DRAWABLE_HEIGHT, scalingFactor, displayXOffset, displayYOffset, scaledWidth, scaledHeight);
-  printf("Source GPU display is %dx%d. Output SPI display is %dx%d with a drawable area of %dx%d. Applying scaling factor %.2fx, xOffset: %d, yOffset: %d, scaledWidth: %d, scaledHeight: %d\n", display_info.width, display_info.height, DISPLAY_WIDTH, DISPLAY_HEIGHT, DISPLAY_DRAWABLE_WIDTH, DISPLAY_DRAWABLE_HEIGHT, scalingFactor, displayXOffset, displayYOffset, scaledWidth, scaledHeight);
+  syslog(LOG_INFO, "GPU display is %dx%d. SPI display is %dx%d with drawable area of %dx%d. Applying scaling factor %.2fx/%.2fx, xOffset: %d, yOffset: %d, scaledWidth: %d, scaledHeight: %d", display_info.width, display_info.height, DISPLAY_WIDTH, DISPLAY_HEIGHT, DISPLAY_DRAWABLE_WIDTH, DISPLAY_DRAWABLE_HEIGHT, scalingFactorWidth, scalingFactorHeight, displayXOffset, displayYOffset, scaledWidth, scaledHeight);
+  printf("Source GPU display is %dx%d. Output SPI display is %dx%d with a drawable area of %dx%d. Applying scaling factor %.2fx/%.2fx, xOffset: %d, yOffset: %d, scaledWidth: %d, scaledHeight: %d\n", display_info.width, display_info.height, DISPLAY_WIDTH, DISPLAY_HEIGHT, DISPLAY_DRAWABLE_WIDTH, DISPLAY_DRAWABLE_HEIGHT, scalingFactorWidth, scalingFactorHeight, displayXOffset, displayYOffset, scaledWidth, scaledHeight);

   uint32_t image_prt;
   printf("Creating dispmanX resource of size %dx%d (aspect ratio=%f).\n", scaledWidth + excessPixelsLeft + excessPixelsRight, scaledHeight + excessPixelsTop + excessPixelsBottom, (double)(scaledWidth + excessPixelsLeft + excessPixelsRight) / (scaledHeight + excessPixelsTop + excessPixelsBottom));
```

Mootikins, 2018-06-08:
This fix works perfectly as you wrote it, thanks a lot :D

juj, 2018-06-11:
Perfect! Added this as a `config.h` and CMake option in above commits.

closed 2018-06-11 by juj