juj/fbcp-ili9341 #12: integrating ili9342

32teeth, 2018-06-12:
Have you had a chance to do an ili9342 integration? looking to do 8-bit (DB0:7)

juj, 2018-06-12:
[ILI9342 spec sheet](http://www.ampdisplay.com/documents/pdf/ILI9342_DS_V008_20100331.pdf) seems to suggest the difference is that it is a landscape 320x240 display rather than [ILI9341](https://cdn-shop.adafruit.com/datasheets/ILI9341.pdf) which is portrait 240x320. Quick glance of the table of contents suggests they are otherwise the same. It might work by swapping the width and height fields around. I don't unfortunately have any ILI9342 displays to try though.

fbcp-ili9341 only does 4-wire SPI, which is a single serial data wire solution. 8-bit methods with multiple data lines DB0 through DB7 are DPI methods, and fbcp-ili9341 does not relate to those. Instead the Raspberry Pi GPU is directly configured to drive them, see e.g. The Ben Heck Show has a multipart series on DPI based display wiring starting from [Rasp Pi No HDMI - Portable MAME Part 1](https://www.youtube.com/watch?v=x0EGnydxxf4).

juj, 2018-07-03:
https://github.com/GameboyZero/ILI9342 reads like you were able to get ILI9342 working, so this issue can be now closed?

closed 2018-07-04 by 32teeth