juj/fbcp-ili9341 #14: Zero/Zero W inconsistent

Mootikins, 2018-06-22:
As the driver currently is, we couldn't get the Zero (or the W, could be one or the other) to work with DMA transfers, but the other worked just fine. This severely hampers our battery life and CPU usage and were wondering if you'd ever make a work around for the one that doesn't work or if we have to research and get it working ourselves.

At this point we think it's probably less an issue of hardware and more about some kind of driver that uses specific DMA channels (more than likely the WiFi/BT).

juj, 2018-06-22:
Try explicitly passing `-DPI_ZERO=ON` to CMake, does that help? CMake tries to autodetect the board, but perhaps the revision is different. On my Pi Zero W, I get

```
pi@raspberrypi_zero:~/code/fbcp-ili9341/build_hx8357d $ cmake ..
-- Doing a Release build
-- Board revision: 9000c1
-- Targeting Raspberry Pi Zero (W)
-- Scaling source image to view. If the HDMI resolution does not match the SPI display resolution, this will produce blurriness. Match the HDMI display resolution with the SPI resolution in /boot/config.txt to get crisp pixel perfect rendering, or alternatively pass -DDISPLAY_CROPPED_INSTEAD_OF_SCALING=ON to crop instead of scale if you want to view the center of the screen pixel perfect when HDMI and SPI resolutions do not match.
-- Preserving aspect ratio when scaling source image to the SPI display, introducing letterboxing/pillarboxing if HDMI and SPI aspect ratios are different (Pass -DDISPLAY_BREAK_ASPECT_RATIO_WHEN_SCALING=ON to stretch HDMI to cover full screen if you do not care about aspect ratio)
-- SPI_BUS_CLOCK_DIVISOR set to 8. Try setting this to a higher value (must be an even number) if this causes problems. Display update speed = core_freq/divisor. (on Pi3B, by default core_freq=400)
-- USE_DMA_TRANSFERS enabled, this improves performance. Try running CMake with -DUSE_DMA_TRANSFERS=OFF it this causes problems, or try adjusting the DMA channels to use with -DDMA_TX_CHANNEL=<num> -DDMA_RX_CHANNEL=<num>.
-- Targeting Adafruit 3.5 inch PiTFT with HX8357D
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/code/fbcp-ili9341/build_hx8357d
```

See "` Targeting Raspberry Pi Zero (W)`" above. Perhaps this line is not kicking in the autodetection? https://github.com/juj/fbcp-ili9341/blob/master/CMakeLists.txt#L24

If that line does get printed for both, then it's possible that there's a DMA channel conflict, in which case try changing the channel numbers.

Mootikins, 2018-06-22:
I was passing -DPI_ZERO explicitly every time I tried (since I thought it was always required). I will try a few times more with different Zeroes and see what happens.

Mootikins, 2018-06-22:
So I built twice, once with and once without -DPI_ZERO=ON, both with DMA on. They compiled fine but the Zero W returned the following:
`bcm_host_get_peripheral_address: 0x20000000, bcm_host_get_peripheral_size: 33554432, bcm_host_get_sdram_address: 0x40000000
BCM core speed: current: 400000000hz, max turbo: 400000000hz. SPI CDIV: 6, SPI max frequency: 66666667hz
Allocated DMA channel 5
Allocated DMA channel 1
Enabling DMA channels Tx:5 and Rx:1
DMA hardware register file is at ptr: 0xb4b29000, using DMA TX channel: 5 and DMA RX channel: 1
DMA hardware TX channel register file is at ptr: 0xb4b29500, DMA RX channel register file is at ptr: 0xb4b29100
DMA channel 0 has peripheral map 0 (is lite channel: 0, currently active: 0, current control block: (nil))
DMA channel 1 has peripheral map 0 (is lite channel: 0, currently active: 0, current control block: (nil))
DMA channel 2 has peripheral map 0 (is lite channel: 0, currently active: 0, current control block: (nil))
DMA channel 3 has peripheral map 0 (is lite channel: 0, currently active: 0, current control block: (nil))
DMA channel 4 has peripheral map 0 (is lite channel: 0, currently active: 0, current control block: (nil))
DMA channel 5 has peripheral map 11 (is lite channel: 0, currently active: 0, current control block: (nil))
DMA channel 6 has peripheral map 13 (is lite channel: 0, currently active: 1, current control block: (nil))
DMA channel 7 has peripheral map 0 (is lite channel: 1, currently active: 0, current control block: (nil))
DMA channel 8 has peripheral map 13 (is lite channel: 1, currently active: 0, current control block: (nil))
DMA channel 9 has peripheral map 6 (is lite channel: 1, currently active: 0, current control block: (nil))
DMA channel 10 has peripheral map 7 (is lite channel: 1, currently active: 0, current control block: (nil))
DMA channel 11 has peripheral map 0 (is lite channel: 1, currently active: 0, current control block: (nil))
DMA channel 12 has peripheral map 0 (is lite channel: 1, currently active: 0, current control block: (nil))
DMA channel 13 has peripheral map 0 (is lite channel: 1, currently active: 0, current control block: (nil))
DMA channel 14 has peripheral map 0 (is lite channel: 1, currently active: 0, current control block: (nil))
DMA TX channel 5 was assigned another peripheral map 11!
DMA TX channel was assigned another peripheral map!`
Luckily I was able to get it to work by changing the TX channel to 4 like so:
`cmake -DUSE_DMA_TRANSFERS=ON -DSPI_BUS_CLOCK_DIVISOR=6 -DDISPLAY_BREAK_ASPECT_RATIO_WHEN_SCALING=ON -DFREEPLAYTECH_WAVESHARE32B=ON -DSTATISTICS=0 -DDMA_TX_CHANNEL=4 ..`

closed 2018-06-22 by Mootikins