juj/fbcp-ili9341 #18: [Discussion] Reduce tearing in ILI9341V by ESP32 controller

rickyzhang82, 2018-07-02:
Thanks for sharing your project and the demo video in Youtube. I'm intrigued with your impressive work.

I want to reduce tearing effect in NES emulator under an odroid-go portable gadget that uses ILI9341V LCD running in ESP32 controller. Here is my proposal:
https://github.com/OtherCrashOverride/go-play/issues/5

The current owner didn't have a proper double buffering implemented. So I will first bring back double buffering. Then I want to experiment interlacing and progressive update as you suggested.

Knowing the fact that SPI doesn't allow using VSYNC, I want to know two things from you since you are more experienced on this topic. I swear I already RTFM on [ILI9341V specs](https://dn.odroid.com/source_peripherals/odroid_go/ILI9341V%20Data%20Sheet(V.100)_20130425.pdf). But I can't find the command in the specs:

1. Command to switch tearing style from diagonal tearing vs line tearing.
1. Command to do a progressive update.

BTW, your code is well written. It is just me being lazy or dumb to find it out from your source code. Thanks!

juj, 2018-07-02:
Thanks, much appreciated.

> Command to switch tearing style from diagonal tearing vs line tearing.

This is implemented by toggling `MADCTL_ROW_COLUMN_EXCHANGE` bit at https://github.com/juj/fbcp-ili9341/blob/master/ili9341.cpp#L58. If `MADCTL_ROW_COLUMN_EXCHANGE` is set, the display memory pixel addressing operates in landscape 320x240 and you get diagonal tearing, whereas if that is not passed at initialization, then the display memory pixel addressing operates in landscape 240x320 mode and you get straight line tearing.

The display panel always refreshes in portrait 240x320 direction from top to bottom, the panel cannot be changed to do its refresh sync in landscape, i.e. only the memory pixel addressing direction is changed by the MADCTL bit above.

When `MADCTL_ROW_COLUMN_EXCHANGE` is not specified, one has to then rotate the framebuffer from 320x240 to 240x320 in software before submitting the pixels to the display. That is achieved in https://github.com/juj/fbcp-ili9341/blob/master/gpu.cpp#L95

> Command to do a progressive update.

Do you mean you are looking for the code that decides whether to do half-field interlaced vs full frame updates? That is decided in https://github.com/juj/fbcp-ili9341/blob/master/fbcp-ili9341.cpp#L333.

rickyzhang82, 2018-07-02:
> The display panel always refreshes in portrait 240x320 direction from top to bottom, the panel cannot be changed to do its refresh sync in landscape, i.e. only the memory pixel addressing direction is changed by the MADCTL bit above.

I see. I see your code. You uses the command Memory Access Control `0x36` in specs which trigger different tearing mode.  How do you get that info from the specs? I can't find such helpful information from [mine here](https://dn.odroid.com/source_peripherals/odroid_go/ILI9341V%20Data%20Sheet(V.100)_20130425.pdf). Does it come with a user guide on this LCD controller as well?

> Do you mean you are looking for the code that decides whether to do half-field interlaced vs full frame updates? That is decided in https://github.com/juj/fbcp-ili9341/blob/master/fbcp-ili9341.cpp#L333.

Actually, I want to experiment two things: interlacing and finding a minimal rectangle change between two frame buffers. But I can't find the info in the specs how to transfer a partial update. What SPI command I should look into? I will implement interlacing first. It seems less computation intensive than finding a delta.

juj, 2018-07-02:
> How do you get that info from the specs? I can't find such helpful information from mine here.

The information is from `Memory Access Control (36h)` command in 8.2.29 on page 129. I don't have any other data sheet either.

> finding a minimal rectangle change between two frame buffers.

For this, see https://github.com/juj/fbcp-ili9341/blob/master/diff.cpp

> What SPI command I should look into?

Subrectangle updates are specified with SPI commands `8.2.20 Column Address Set (2Ah)` and `8.2.21 Page Address Set (2Bh)`.

rickyzhang82, 2018-07-02:
Thank you so much with your wisdom! I will borrow (steal) your good stuffs and port to ESP32.

closed 2018-07-02 by rickyzhang82