juj/fbcp-ili9341 #19: Backlight control on 9486L appears to do nothing.

Trekintosh, 2018-07-02:
I build with -DGPIO_TFT_BACKLIGHT=23 -DBACKLIGHT_CONTROL=ON and let the Pi sit for a while. After an hour (with the BL pin connected to GPIO 23, of course) nothing seems to have happened.

When connecting the BL pin on my screen to 3.3v, the screen gets slightly brighter. When connecting it to GND the screen goes black. Is this different behavior from other screens?

juj, 2018-07-03:
Thanks, the backlight control functionality was only wired for Adafruit's ILI9341. Added support now for ILI9486(L) as well in above commit.

closed 2018-07-03 by juj

Trekintosh, 2018-07-03:
New issue with backlight on same 9486L screen on a Raspberry pi zero w.

When backlight control turns off the screen, the screen goes black. However, when activity is detected (for reference, just pressing a key does NOT seem to count as detected, but hitting enter is. This is in the console immediately after logging in.) the screen flashes white then comes back to the previous screen, but stays frozen. Pressing keys or trying to interact does nothing.

Please let me know if you need more info.

juj, 2018-07-03:
Oh hmm, that happens because fbcp-ili9341 does not actually listen to keystrokes or mouse input, but activity to resume is detected from when content changes on screen. I suppose only pressing enter then resumes the display because that produces a newline in terminal that is a big visual change, whereas any other key only adds a single character that's only few pixels.

In above commit, now added reading of keyboard input to detect for activity as well. That should make any key trigger the backlight to come on immediately. You may need to configure this line https://github.com/juj/fbcp-ili9341/commit/24e97135fde4bb55fac2967904851216fe53cb8b#diff-9d960d517e61245605744031679a0fdeR22 , I'm not sure if that will correspond to the keyboard device on all systems, or if it is possible that e.g. `"/dev/input/event0"` can be the keyboard in some cases.

That commit adds new files to the build, so you'll need to rerun CMake or otherwise the build will finish with undefined symbols since CMake does not automatically detect new .cpp files in the build.

commit 5ea1eac4 (2018-07-03): Add BACKLIGHT_CONTROL support to ILI9486(L). Closes #19