juj/fbcp-ili9341 #20: Run on Power Up

32teeth, 2018-07-04:
Is there a way to add the driver to ```cmdline.txt``` to run on power up instead of having white screen for 15 seconds?

not really an issue, more a feature request

juj, 2018-07-04:
Hmm, what is `cmdline.txt`? I am not familiar with that feature. If it's https://elinux.org/RPi_cmdline.txt, then that reads like something to configure kernel parameters?

You can try making fbcp-ili9341 start up as part of Linux init.d routines, https://www.raspberrypi.org/forums/viewtopic.php?t=70520. I believe those should run much earlier than /etc/rc.local does, although I am not familiar with that mechanism more than knowing that it exists, so have not tried it out before. Googling for "fbcp init.d" might help, that finds something like this: https://gist.github.com/notro/eac0fec51cac67bb99c7

If I recall correctly, @Trekintosh was experimenting with something like above, though not sure how much init.d will improve timewise compared to /etc/rc.local.

Trekintosh, 2018-07-04:
I followed the init.d instructions in the thread @Juj linked and it worked perfectly. My screen starts up about 15-20% into the boot sequence. 

Cmdline.txt wouldn’t do anything for you in tho situation.

32teeth, 2018-07-10:
hmm, i did this and still getting only once everything is up and running.

@Trekintosh can you share your init.d that runs ili9341?

Trekintosh, 2018-07-10:
@32teeth No problem, here it is from my working system. Pop this file into /etc/init.d, make it executable, then do `sudo update-rc.d fbcp-ili9341init defaults`

[fbcp-ili9341init.zip](https://github.com/juj/fbcp-ili9341/files/2181041/fbcp-ili9341init.zip)

matemaciek, 2021-11-28:
Yet another version, so it kicks in even earlier:
- Put in `/etc/systemd/system/fbcp.service`:
```
 [Unit]
 Description=framebuffer copy to spi screen
 Before=basic.target
 After=local-fs.target
 DefaultDependencies=no

 [Service]
 Type=simple
 ExecStart=/home/pi/fbcp-ili9341/build/fbcp-ili9341
 [Install]
 WantedBy=basic.target
```
- `sudo systemctl daemon-reload`
- `sudo systemctl enable fbcp`

DansDesigns, 2023-07-18:
> Yet another version, so it kicks in even earlier:
>
> * Put in `/etc/systemd/system/fbcp.service`:
>
> ```
>  [Unit]
>  Description=framebuffer copy to spi screen
>  Before=basic.target
>  After=local-fs.target [...]

This works brilliantly except for the "DefaultDependencies", I had to set it to "yes" to work properly with my screen