juj/fbcp-ili9341 #23: keyboard event

am7619, 2018-07-08:
this driver is a great piece of work, thanks!!!
however, my keyboard was on another event (/dev/input) resulting in an error and i am planning to use no keyboard at all. is it possible to exclude the keyboard events e.g. as a compiler option?

juj, 2018-07-08:
Are you using the `BACKLIGHT_CONTROL` option? Currently keyboard control is hooked onto that option. Added `config.h` item `#define BACKLIGHT_CONTROL_FROM_KEYBOARD` that you can comment out to disable reading the keyboard altogether, even if using backlight control. That should do what you are looking for?

am7619, 2018-07-08:
indeed, i activated `BACKLIGHT_CONTROL` without fully understanding it. my fault. thank you very much for your help!

closed 2018-07-08 by am7619

juj, 2018-07-08:
Perfect, great to hear it works!