juj/fbcp-ili9341 #24: Setup freezes

EldritchIHC, 2018-07-09:
I  bought a 3.5 TFT LCD from Aliexpress (https://www.aliexpress.com/item/3-5-inch-26P-SPI-TFT-LCD-Display-Screen-with-Touch-Panel-320-480-for-RPi1/32746328272.html),it has an ILI9486 chip and it says that the SKU is MPI3501.Before finding your driver I have tried other methods to get it working but the refresh rate was very slow so I decided on your setup.
I have been trying to get it working with differny cmake options(waveshare,ili9486,specifing the pins),but it every time the setup gets stuck here:
[image].
The pinout for the display is:
[image]
DC:18,Reset:22
I would be very thankul if you could help me at least with some tips.

juj, 2018-07-09:
The log prints there indicate a successful startup, and at that point fbcp-ili9341 main loop should be running. Added a log message to clearly indicate that.

Not exactly sure what would be the issue here, try following the troubleshooting section starting from https://github.com/juj/fbcp-ili9341#i-am-running-fbcp-ili9341-on-a-display-that-was-listed-above-but-the-display-stays-white-after-startup. I have one display from Aliexpress that the manufacturers advertise to be ILI9486, but actually it is a ILI9486L instead, so you could try initializing the display as that. Also try using a higher cdiv setting, like `-SPI_BUS_CLOCK_DIVISOR=16` to see if that might help.

EmulatedBen, 2018-07-10:
Ok so it works. I had the same issue. Try doing it from putty. It has worked in your case.

juj, 2018-07-14:
Were you able to get this working @EldritchIHC ? Reading back the report, it is almost certain that the `-DSPI_BUS_CLOCK_DIVISOR=6` setting will be too fast for ILI9486/ILI9486L displays.

Some advanced options to try might be to build with

```
cmake -DILI9486=ON -DGPIO_TFT_DATA_CONTROL=18 -DGPIO_TFT_RESET_PIN=22 -DSPI_BUS_CLOCK_DIVISOR=20 ..
```

and remove line https://github.com/juj/fbcp-ili9341/blob/master/ili9486.h#L19 before building with `make -j`, or to build with

```
cmake -DILI9486L=ON -DGPIO_TFT_DATA_CONTROL=18 -DGPIO_TFT_RESET_PIN=22 -DSPI_BUS_CLOCK_DIVISOR=20 ..
```

and remove line https://github.com/juj/fbcp-ili9341/blob/master/ili9486l.h#L20 before building.

Also you could try not enabling reset pin at all:

```
cmake -DILI9486=ON -DGPIO_TFT_DATA_CONTROL=18 -DGPIO_TFT_RESET_PIN=0 -DSPI_BUS_CLOCK_DIVISOR=20 ..
```

If all of that fails, then the last resort is probably to use the official driver and reverse engineer the communication it sends to the display, and develop a new display backend .cpp file to do the same in fbcp-ili9341.

EldritchIHC, 2018-07-14:
Thank you for all the support! I have been busy since Ihave posted the issue,I will try all the suggestions monday and i will report.
I know that the display functions because I have tried with other drivers and it worked but  I wasn't pleased with the result.

EldritchIHC, 2018-07-17:
I have tried some of your suggestions and somtehing changed:
[image] it gets stuck at running main loop now(maybe it works but takes a long time?).

When I tried other drivers this two worked:
https://github.com/goodtft/LCD-show/blob/master/README.md   -this one stopped when my raspberry pi zero entered in emulstation

https://www.raspberrypi.org/forums/viewtopic.php?t=139917  -this one didnt always start ,made my installation require login every time and it was laggy

juj, 2018-07-22:
If you get that print, unless the Pi hardware physically freezes and you need to do a hard power off to reboot, I believe it is not actually "stuck", but fbcp-ili9341 is running its main loop happily, believing it is correctly outputting to the display. The root issue is likely that the ILI9486 controller code that fbcp-ili9341 has at https://github.com/juj/fbcp-ili9341/blob/master/ili9486.cpp is not somehow working/compatible for your display. Either the display is not a ILI9486, or it is a flavor of ILI9486 that is somehow different (different hardware revision?) than what fbcp-ili9341 code works with, or perhaps some of the pin configurations are not correct (worth to triple check for sanity).

At this point I unfortunately do not have anything more to suggest except to reverse engineer the working display driver to cross reference where fbcp-ili9341 is not supporting it. For good measure, you could try cmake configuring with `-DILI9486L=ON` instead of `-DILI9486=ON` (when changing that, delete the file `CMakeCache.txt` in the `build/` directory, or delete and recreate the whole `build/` directory), since some web sites have mixed up these two.

EldritchIHC, 2018-07-22:
Ok,thank you very much,you can close the issue now.I will try more settimg in the future.I thought about using the waveshare35 setting because te pinouts look similar.