juj/fbcp-ili9341 #25: Doesn't launch correctly without keyboard plugged in.

dmradford, 2018-07-22:
Summary: If there is no keyboard attached to the device on boot, the screen fails to load.
Plugging in a keyboard and rebooting restores screen functionality.

Expected behavior: Screen should load even if no keyboard is found.
Use Case: I am using this in a handheld game system that does not have a keyboard attached, it simply uses an arduino controller as a gamepad for HID input.

Issue occurs on Raspberry Pi 3, running Raspbian Scratch and Retropie 4.4

Built with the following configs:

cmake -DILI9341=ON -DGPIO_TFT_DATA_CONTROL=24 -DGPIO_TFT_BACKLIGHT=18 -DBACKLIGHT_CONTROL=ON -DGPIO_TFT_RESET_PIN=25 -DDISPLAY_ROTATE_180_DEGREES=ON -DSPI_BUS_CLOCK_DIVISOR=6 -DSTATISTICS=0 ..

juj, 2018-07-22:
Try removing the config line for BACKLIGHT_CONTROL_FROM_KEYBOARD in config.h if that works?

juj, 2018-07-22:
Reading the code in keyboard.cpp, lowered the failure message to a warning message in above commit to make the code ignore the failure to open keyboard, and continue on. That should help the driver to keep on trucking.