juj/fbcp-ili9341 #26: Can this be updated for ILI9488

Virgilius1995, 2018-08-01:
Any way to get this to be run with a ILI9488.

altaafsayed, 2018-08-02:
Yeah I would also like this to be updated for an ILI9488 please.

juj, 2018-08-04:
I am not currently familiar with ILI9488 and don't have an ILI9488 display unfortunately, so I am not able to do this. Perhaps ILI9488 is close enough to ILI9486 or ILI9486L so that one of those backends could be used as a basis to do a port, if you have got access to a ILI9488 display. If someone is able to get it working, patches are welcome.

Virgilius1995, 2018-08-05:
I've been modifying the ili9341 files since i read that its very similar to the ili9488. I've been able to get reset and back light control working but no luck on image yet. Any ideas are appreciated. I'm using the following sources but I've got little to no experience.

https://raspberrypi.stackexchange.com/questions/44179/rpi-2-and-3-95-tft-ili9488/44472#44472
https://www.raspberrypi.org/forums/viewtopic.php?t=180064
https://www.elecrow.com/download/ILI9488%20Data%20Sheet_100.pdf

Virgilius1995, 2018-08-06:
Well one massive issue stemming from me not reading I see that this driver is only for 4-wire not the dual line input header mine came with{;

juj, 2018-11-03:
fbcp-ili9341 now gained 3-wire SPI support (no Data/Control line) in the above commit. That can be activated in CMake by setting `-DGPIO_TFT_DATA_CONTROL=-1` to tell the build system that there is no D/C pin to connect. The support is experimental, and it currently does not support ILI9486 in 3-wire mode, but if ILI9488 works more like 9486L, then there is a chance that the new support might work for it too.

See the above commit for updates to Readme.md for the docs.

hakmo, 2020-06-12:
So I've gotten this to run on an ILI9488 display (specifically an [MSP3520 display](http://www.lcdwiki.com/3.5inch_SPI_Module_ILI9488_SKU:MSP3520)) on a Raspberry Pi 4 running Buster Light.  I just built for the ILI9486L, my specific options were:

`cmake -DILI9486L=ON -DGPIO_TFT_DATA_CONTROL=25 -DGPIO_TFT_RESET_PIN=7 -DGPIO_TFT_BACKLIGHT=24 -DSPI_BUS_CLOCK_DIVISOR=14 -DBACKLIGHT_CONTROL=ON`

Currently get ~19fps; I just have the display wired to the Pi with long dupont connectors right now so I haven't bothered to experiment with the clock divisor yet to see how fast I can get.

Also note that the '9488 supports 24-bit color, so a slight code change to the '9846L could improve display quality.  I'll look into making that change and submitting a pull req.

zarcade, 2020-06-15:
Also interested in ILI9488 support.

quezadaminter, 2020-06-18:
I purchased a couple of these displays [https://www.buydisplay.com/serial-spi-3-5-inch-tft-lcd-module-in-320x480-optl-touchscreen-ili9488](url). They have the ili9488 datasheet on their site [https://www.buydisplay.com/download/ic/ILI9488.pdf](url). It's a bit dense for me so I am struggling a bit to understand it. However I compared the registers listed in that datasheet with the ones used in the 9431 and 9486 and their purpose appears to line up well.

The supplier also has sample code for microcontrollers posted on the site and looking through those I wrote an ili9488 module using the 9486L as a template. However, I can see the screen reset and blank but I can't see anything on the screen.

I also found a project (for microcontroller) claiming that a slight change in the Adafruit 9341 driver allowed the 9488 to work: [https://www.instructables.com/id/Arduino-and-35-Inch-320x480-TFT-LCD-ILI9488-SPI-In/](url) with a slight modification:

>  Only, the Height and Width parameters at Adafruit_ILI9341.h file is modified to
>#define ILI9341_TFTWIDTH 320
>#define ILI9341_TFTHEIGHT 480
>And at the Adafruit_ILI9341.cpp file, at the function Adafruit_ILI9341::begin(void), the following lines are modified for adapting ILI9488 device (320x480 pixels).
>writecommand(ILI9341_DFUNCTR); // Display Function Control
>writedata(0x02);
>writedata(0x02);
>writedata(0x3B); 

I have not tried this yet.

There is also this project which added the ili9488 to the Adafruit display ecosystem: [https://github.com/jaretburkett/ILI9488](url). The code inside appears to be the same used in the code provided by the supplier I mentioned above. 

Would it be reasonable to use the initialization and control sequences used in the microcontroller code when writing the module for the Pi? I would expect this would be the case since the display does not care where the instructions come from, only that they match what it expects.

Also, I am doing this on a ZeroW. Not sure if that may be part of my struggle. I did get a Adafruit PiTFT working on it so I suspect the Zero is ok.

juj, 2020-06-19:
> Would it be reasonable to use the initialization and control sequences used in the microcontroller code when writing the module for the Pi?

It would - please make sure though that the licenses of the other repositories are compatible, I do not want to have other peoples' work reach this repository if their licensing requirements are different.

There may be a difference in color spaces: ILI9341 is RGB565, but iiuc ILI9488 like ILI9486L might be a R6X2G6X2B6X2 color format display? Though perhaps it does have different color formats configurable.

quezadaminter, 2020-06-19:
I agree, I did not mean to imply that the code should be copy/pasted but rather used as a basis. The only header with information in the code from the supplier is:

>//----------------------------------------------------------------------
>//EASTRISING TECHNOLOGY CO,.LTD.//
>// Module    : 3.5 INCH TFT LCD  320RGB*480 DOTS
>// Lanuage   : C51 Code
>// Create    : JAVEN LIU
>// Date      : 2014-02-08
>// Drive IC  : LCD:ILI9488  TP:TSC2046  
>// INTERFACE : LCD 4 WIRE SPI   TP:SPI  
>// MCU 		 : STC12LE5C60S2
>// VDD		 : 5V OR 3.3 V
>// MCU VDD   : 3.3V
>//----------------------------------------------------------------------

As far as I can tell it is provided on their website explicitly to be used and there is no mention of licensing conflicts. So I believe it should be ok.

The color space, according to the data sheet, varies depending on the interface method used. For SPI 3 wire there is a choice of 8 color (RGB 1, 1, 1) or 262K (RGB 6, 6, 6).

The 4 wire interface has 3 options: 8 color (RGB 1, 1, 1) or 65K (RGB 5, 6, 5) or 262K (RGB 6, 6, 6)
These options are selected by setting the DBI bits of the 3Ah register.

I got my 9488 to display content using @hakmo 's solution from above, but at least in my case the colors were off which I suspect is related to the details mentioned above, I will look into that as soon as I get a chance.

hakmo, 2020-06-19:
The datasheet indicates that the driver at least supports RGB888 space, but I've found that it'll run in R6X2G6X2B6X2, though it doesn't look great.  Colors are off like quezadaminter mentions (see attached images) but sometimes blink into looking good briefly.  I'm not sure if that's down to color space or some other settings.  I've started trying to implement RGB888 color space, but am having to learn dispmanx as I go.  Trying the settings quezadaminter mentioned first to see if that clears things up is a good idea.

For DSPI_BUS_CLOCK_DIVISOR, I was able to get that down to 8 before I got image corruption, still with long dupont wires.

[image]
[image]

quezadaminter, 2020-06-19:
Nice! I have not been able to set the clock divider lower than 12, but I am also on dupont wires for now. I do agree that it "works" in 6X2 but I have not checked what the 3Ah register is set to for the 9486L. You could try messing with that, the details I have from the datasheet I reference above are on pages 199 (3Ah register details) and 119-122 for the description of the color spaces and the expected bit packing order.

hakmo, 2020-06-19:
I can report that matching init settings from jaretburkett's repo fixes the color/display issues.  I'm otherwise running ILI9486L settings with clock divider of 8.
[image]

quezadaminter, 2020-06-20:
@hakmo That's great! Any chance you could share the code you set up? I tried doing the same thing but mine is locking up on startup, even at slow speeds. I clearly did something wrong with the settings.

Did you replace the code inside the BEGIN/END_SPI_COMMUNICATION() scope? I noticed that the END_SPI_ call is removed when building with DMA support. However jaretburkett's code calls the END, then BEGIN_, then sends another message, then calls END_ again. The datasheet makes reference to this steps as needed to reset the communication state with the 9488. Did you add these lines also?

quezadaminter, 2020-06-20:
@hakmo never mind, I found my typo. My display went from this:

[image]

To this:

[image]

@juj, I created a module for the 9488 in my fork of the repository. Would you be interested in a pull request to merge it into your base?

juj, 2020-06-20:
Sure, a pull request would be welcome!

le51, 2021-02-25:
Hi there,
would you mind helping me a bit.
I'm confused regarding the wiring of a raspberry pi3B+:

Panel   Panel        RPI       RPI pin
Pin       Label        func     number
1          VCC          3.3V        1
2          GND         GND       9
3         CS             not connected
4         RESET       GPIO7      26
5         DC/RS       GPIO25   22
6         MOSI        GPIO10   19  
7         SCK           GPIO11   23
8         LED           GPIO24   18

others pins are not connected

here the cmake command I've used:
cmake -DILI9488=ON -DGPIO_TFT_DATA_CONTROL=25 -DGPIO_TFT_RESET_PIN=7 -DGPIO_TFT_BACKLIGHT=24 -DSPI_BUS_CLOCK_DIVISOR=30 -DBACKLIGHT_CONTROL=ON -DARMV8A=ON .. 

at least display fired up when I launch "sudo ./fbcp-ili9341" but nothing is displayed.
kernel: Linux raspberrypi 5.10.11-v7+

Did I miss something ?

Thank for help.

quezadaminter, 2021-02-26:
@le51 , Can you post a picture of your setup? It may be helpful to see what your panel looks like. In the meantime, here are some flags you may consider adding to your build attempt:

`
-DBACKLIGHT_CONTROL=OFF -DSTATISTICS=1 -DUSE_DMA_TRANSFERS=ON
`
Remove the -DGPIO_TFT_BACKLIGHT flag from yours for now, we don't want the driver to blank the screen while you are testing. The STATISTICS flag will cause the driver to print some information directly from within itself. This should help confirm if the issue is with the driver or the OS. If you see the performance parameters at the top of the screen then the driver is ok.

Make sure you indeed have a 9488 display, the initialization sequences are not always interchangeable and will cause the panel to misbehave.

le51, 2021-02-26:
Hi, thank you very much for help !

This is the display I own:
http://www.lcdwiki.com/3.5inch_SPI_Module_ILI9488_SKU:MSP3520
From that page a link to the schematics is given.

I will investigate this tomorrow as it's getting late here.
Thanks again, Ill keep you in touch !

quezadaminter, 2021-02-27:
@le51 I noticed that your list of pin connections indicates that the CS pin is unconnected. This cannot be. The CS pin is a critical part of the SPI interface and if you don't connect it then the display won't know when the pi is talking to it. Connect the CS pin of the display to GPIO8 (CE0) on the pi and try your driver again.

le51, 2021-02-27:
Hi @quezadaminter , you make my day, thank you very much.
This was definitively a wiring issue.

For those who are here, this is my display: MSP3520 with ili9488 driver.
http://www.lcdwiki.com/3.5inch_SPI_Module_ILI9488_SKU:MSP3520
[image]

**Wiring I used:**

Panel--- Panel------BCM-------RPI pin
Pin------Label------Pin nb-----number
1--------VCC-------3.3V--------1
2 -------GND-------GND-------9
3 -------CS-----------8---------24
4 -------RESET-------7---------26
5--------DC/RS------25--------22
6--------MOSI------ 10--------19
7--------SCK--------11--- -----23
8--------LED--------3.3V-------18

**cmake command:**

 cmake -DILI9488=ON -DGPIO_TFT_DATA_CONTROL=25 -DGPIO_TFT_RESET_PIN=7 -DSPI_BUS_CLOCK_DIVISOR=10 -DBACKLIGHT_CONTROL=OFF -DARMV8A=ON -DSTATISTICS=0 -DUSE_DMA-TRANSFER=ON ..

Now, make the touch function to work !

Cheers

livep2000, 2023-12-30:
After a few weeks of fiddling, the driver seemed working, but got a "vc_dispmanx_display_open failed!" message, and the display showed no image.

the solution was here : [https://github.com/juj/fbcp-ili9341/issues/100]

(enabling tvservice)

sudo tvservice -p