juj/fbcp-ili9341 #28: "vc_dispmanx_snapshot() failed with return code -1!" when kodi launches

DSchndr, 2018-08-18:
When fbcp-ili9341 (also happens on fbcp-rpi) is running and kodi gets launched fbcp-ili9341 exits with "vc_dispmanx_snapshot() failed with return code -1!" on pi zero.
But when running fbcp-ili9341 after kodi was started everything works like it should.

Maybe retrying couple of times to make an dispmanx snapshot with some delay could fix this issue?

Also i have the issue with ^[B getting "pressed" after i kill fbcp-ili9341.

juj, 2018-08-19:
I believe this is the "Screen resize freezes DispmanX" issue mentioned in the readme. See https://github.com/raspberrypi/userland/issues/461 for more information.

(Although I do not have Kodi to verify)

xjamxx, 2018-10-19:
> 
> 
> I'll close this since there is no good course of action that I know that fbcp-ili9341 could take here. Updated the call site of where the print occurs to point back to bug trackers. It seems that what people are doing is they set up fbcp or fbcp-ili9341 as a system service that reboots itself when this occurs. That is the best way to recover here that I currently know.

I tried with systemd service unit, restart on-failure, but the screen freeze anyway, so not sure how to implement that?

juj, 2018-10-22:
I have not tried system services myself. In this thread https://github.com/juj/fbcp-ili9341/issues/20 @Trekintosh has a solution for that. It may be possible that it is some other issue.

You mentioned in first comment that _But when running fbcp-ili9341 after kodi was started everything works like it should._ , that makes me wonder: If you first have fbcp-ili9341 running, then launch Kodi to get the crash in fbcp-ili9341, and then manually restart fbcp-ili9341, does that work and enable fbcp-ili9341 to properly re-launch? Or does it still keep failing, and unable to recover even when manually restarted? (I wonder if the problem is in system service setup, or already in that if dispmanx crashes on fbcp-ili9341 that the dispmanx driver is then in bad state system wide)