juj/fbcp-ili9341 #29: waveshare 1.44in lcd hat & ST7735r driver , display works but various issues

chilvence, 2018-08-27:
Hi, I bought one of [these](https://www.waveshare.com/wiki/1.44inch_LCD_HAT) cheap displays ages ago to muck around with my pi zero, but they don't seem that widely supported so I am always having trouble getting it set up.

I found this driver by chance so decided to try it, because I saw it supports the ST7735r controller. This thing is actually an ST7735S, but I figured what the hell, and sure enough it worked. At first it was a bit garbled and the wrong resolution, but it was visible enough to make me keep going.

I tweaked the code to give it the correct resolution of 128x128 , and it seems to run stable enough with the spi clock divisor set at 12. It runs well enough, but the orientation is wrong (seems this can be the fault of the board layout, its mounted sideways to save space by the look of it), and also the left/right screen axis is flipped so everything reads backwards.

I managed to flip the horizontal axis. I have no idea how this is meant to be done properly, but I found a line that read 'madctl |= MADCTL_ROW_ADDRESS_ORDER_SWAP;', and commenting that out got me back to proper left and right orientation. Sorry if that seems like a kludge, but I am not really a programmer, my method is usually along the lines of 'hit it until it works', I'd appreciate some insight on how to properly do this.

I also don't know the best way to fix the rotation, it basically needs to be rotated 90 degrees, I know raspbian can do this, but I am typing this on my pi zero on a desktop monitor so I don't want to have to turn my head 90 degrees quite yet. I was also hoping there is a way to do this in the driver itself, because it would be nice to retain the ability to easily use the HDMI port, eg, to plug into a TV at someones house.

I'm really impressed by the performance, with fbtft+fbcp I only got terrible refresh rates and ended up sticking the thing in a drawer in disappointment, but with this I can play Doom with super smooth frame rates, and seemingly anti aliasing thrown in as a side effect of the screen scaling :) . It really is a nifty little gadget, a tiny joystick, 3 buttons and a screen on a board that perfectly matches the pi zero, so it's nice to finally get it working well, thanks for your hard work!

chilvence, 2018-08-28:
That's great info, I have managed to get it rotated correctly now, without having to mess with the hdmi output, which is great news. It seems that row/column exchange, combined with 180 degree rotation does the trick (in addition to the madctl |= MADCTL_ROW_ADDRESS_ORDER_SWAP; that was there already)

I guess this means that all the rotation is basically taken care of by the display controller itself?

Also, I have no deep understand of how these screens work, but is there enough here to add in support for this hat so that other people can use this easily?

juj, 2018-08-29:
To add support for an out of the box `-DST7735S=1` path or a `-DWAVESHARE_ST7735S_HAT=1` path, there are a few places where `ST7735S` vs `ST7735R` related changes will need to be done. These can be found by searching for `ST7735R` in the codebase. The items in particular for good implementation are to
 - audit through ST7735S spec sheet to verify that all the initialization commands that are shared from ST7735R at https://github.com/juj/fbcp-ili9341/blob/2e889803bfb8663eba08a92ffdd579241d9e1efc/st7735r.cpp#L28-L102 are actually valid ones also for ST7735S, so that it is not passing any silly ones there,
 - in particular, ensure that the refresh rate at https://github.com/juj/fbcp-ili9341/blob/2e889803bfb8663eba08a92ffdd579241d9e1efc/st7735r.cpp#L87-L88 is an appropriate one, or adapt it to compute a ST7735S specific rate,
 - add support for enabling/disabling `DISPLAY_OUTPUT_LANDSCAPE` in `config.h` (might not make sense for the hat, but for other ST7735Ses out there),
 - and, this is probably trickiest, find the "neutral" panel scan orientation to account for `DISPLAY_FLIP_ORIENTATION_IN_SOFTWARE` at https://github.com/juj/fbcp-ili9341/blob/2e889803bfb8663eba08a92ffdd579241d9e1efc/config.h#L180-L192 so that when `DISPLAY_FLIP_ORIENTATION_IN_SOFTWARE` is enabled, the driver should let the panel scan in its neutral orientation (`madctl = 0`), for least amount of tearing and latency, and when `DISPLAY_FLIP_ORIENTATION_IN_SOFTWARE` is not defined, update in `madctl` controlled way, like was already done here.

These are a bit tricky to do without a board, the semantics of these are laid out somewhat delicately.

Something to do to help is to run `git diff` in your working tree to post the changes you needed, along with the CMake command line you use to configure, and post the results here. That might help someone stumble to the instructions and is a good starting point.

juj, 2018-09-06:
Got access to a WaveShare 128x128 1.44" HAT display, and added support to it today in https://github.com/juj/fbcp-ili9341/commit/9170e872b67473f04719bbff3f8e5f35ac571f5f . It is enabled via CMake directive `-DWAVESHARE_ST7735S_HAT=ON`.

closed 2018-09-06 by juj

chilvence, 2018-09-06:
Oh wow, that is a result way quicker than I was expecting! I was almost about to give up and suggest I could send you mine, since I have no idea what I am doing...  Or at least, I trust you more than I trust me :O) Thank you very much for this though.

commit 9170e872 (2018-09-06): Add support for WaveShare ST7735S HAT. Closes #29