juj/fbcp-ili9341 #31: How do I remove  (CPU, GPU, SPI etc)refreshing

Qiushaobai, 2018-09-06:
Thanks，its very helpful for me. Im using a 2.2" ili9341 screen and got it working with a pi 3B plus. Just 1 issues i need help with. On the top part of the screen there is some stats permanently being displayed and constantly refreshing (CPU, GPU, SPI etc). How do I remove this? thank you

juj, 2018-09-06:
The prints come from the statistics overlay. Pass `-DSTATISTICS=0` to CMake config line to disable it. See here https://github.com/juj/fbcp-ili9341#specifying-other-build-options

Qiushaobai, 2018-09-07:
Thank you very much, you are better than some people .they don’t have the patience to teach newbies how to do this. They just want to make money from the fruits of others’ work.

juj, 2018-09-13:
Thanks! I'll close this as resolved.

closed 2018-09-13 by juj