juj/fbcp-ili9341 #32: GPU Memory leak?

smumryak, 2018-09-11:
After running for ~30 minutes on RPi 3B with 96MB GPU memory allocated i've started getting message "Failed to allocate GPU memory! Try increasing gpu_mem allocation in /boot/config.txt. See https://www.raspberrypi.org/documentation/configuration/config-txt/memory.md".
I don't actually think it's a memory leak, but maybe some issue in memory mapping calculation?

Video setup in config.txt:
`gpu_mem=96
hdmi_force_hotplug=1
hdmi_group=2
hdmi_mode=87
hdmi_cvt 480 320 60 6 0 0 0`

Built parameters:
`-DSPI_BUS_CLOCK_DIVISOR=12 -DWAVESHARE35B_ILI9486=ON -DARMV6Z=ON`

No overclocking/downclocking.

juj, 2018-09-13:
That memory allocation failure message comes from line 145 at

https://github.com/juj/fbcp-ili9341/blob/2393c11c31e151ee9f29af238eca753eab46923a/dma.cpp#L136-L153

The only place where that `AllocateUncachedGpuMemory()` is called is from `InitDMA()` in https://github.com/juj/fbcp-ili9341/blob/2393c11c31e151ee9f29af238eca753eab46923a/dma.cpp#L216-L217

which is called exactly once at application startup, but never at runtime. Exactly three calls to `AllocateUncachedGpuMemory()` are made.

It is odd if you are seeing this error appear during runtime. Are you shutting down and restarting fbcp-ili9341 multiple times while the Pi is running?

The deallocation function for those uncached GPU memory blocks is at

https://github.com/juj/fbcp-ili9341/blob/2393c11c31e151ee9f29af238eca753eab46923a/dma.cpp#L729-L735

where the corresponding three memory blocks are freed.

`DeinitDMA()` is called from `DeinitSPI()`, which is called at the end of the program at https://github.com/juj/fbcp-ili9341/blob/2393c11c31e151ee9f29af238eca753eab46923a/fbcp-ili9341.cpp#L562-L568.

If fbcp-ili9341 shuts down cleanly, it should free the GPU memory. There are a couple of places in the program where `exit(1)` is called directly without shutting down cleanly. Perhaps one of those is being called during execution that is causing GPU memory to be leaked behind? Try to debug if you are getting an unclean shutdown of the application.

juj, 2018-09-13:
Adjusted the code at https://github.com/juj/fbcp-ili9341/commit/83f1326cf05f3d88e5e14000e9b2798283395d36 to more gracefully shut down if DispmanX API fails. If that is the cause, then it is possible that the new code will properly free up GPU resources.

juj, 2018-10-01:
Reading the FATAL_ERROR does suggest like the above commit could fix the error. This line https://github.com/juj/fbcp-ili9341/commit/83f1326cf05f3d88e5e14000e9b2798283395d36#diff-097f3581ebc42989986bca651ddc68c4R126 is a known place where Raspberry Pi kernel code trips up due to this bug in system DispmanX: https://github.com/raspberrypi/userland/issues/461 that leaves no options to fbcp-ili9341 than to abort. The above commit changed it so that when that happens, it should cleanly free up system resources, so when running as a service, it should no longer leak.

I'll close this bug based on the assumption that the above machinery is working, though feel free to reopen if you still are encountering the issue.

closed 2018-10-01 by juj