juj/fbcp-ili9341 #35: Raspberry Pi Model B working

swerveshot, 2018-09-24:
Hi @juj!

This issue is to let you know that I got a working build on the Pi 1, model B.

I've got a cheap ili9341 screen like the one bamsegod used over at [sudomod](https://sudomod.com/forum/viewtopic.php?t=5854). Only I got the 2.4" version. I also had an old Model B Pi rev 000f lying around. So I thought to put it to good use by combining them and build a tiny retropie machine.

I used commit 83f1326 and started toying around with overclocking and divider settings. Currently I'm using the TURBO preset and using a CDIV value of 10. Using 8 does work, but I noticed some flickering when running ioquake3 which is almost completely absent when using 10 as a CDIV value.

Currently I'm testing Sonic because it has a nice attract mode which I can leave running for long periods of time just to test how stable the config is. Obviously I don't get a constant 60fps but there's no noticeable lag so that's awesome on such a humble machine. Even ioquake3 ran incredibly smooth as well.

So if there's anything I can do to test I'd be glad to help. It's the least I can do after the great work you provided.

juj, 2018-10-01:
Amazing! Thanks for the detailed writeup, nice to read a performance data point on the Pi 1 B.

CDIV=10 with BCM core frequency of 500MHz gives 50MHz SPI bus speed. CDIV=8 with BCM core frequency of 500MHz would give 62.5MHz, so the max refresh performance of the panel seems to be somewhere between 50-62.5MHz. You could try underclocking the BCM `core_freq` back to e.g. 440 from 500MHz, and see if that would unlock CDIV=8 for SPI bus speed of 440/8=55MHz.

Someone commented earlier that cable length could factor in to the signal quality and max refresh speed that can be achieved. I have not observed this first hand on any of my displays (that I have run both as hat-on-top or with cables) - though I only ever did a signal test on Pi3B and not on any others. If you are able to test out with shorter cables or stack on-top header connection, that may be interesting to compare if it can help at all.

Pi 1B, like Pi Zero, is particularly constrained on performance due to this limitation https://github.com/raspberrypi/userland/issues/440, if that gets picked up by the development team, it could make Pi 1 run even smoother, but great to hear that it is getting good use already!

closed 2018-10-01 by juj

sorriso93, 2020-04-18:
> Hi @juj!
> 
> This issue is to let you know that I got a working build on the Pi 1, model B.
> 
> I've got a cheap ili9341 screen like the one bamsegod used over at [sudomod](https://sudomod.com/forum/viewtopic.php?t=5854). Only I got the 2.4" version. I also had an old Model B Pi rev 000f lying around. So I thought it would be nice to put them to good use by combining them and build a tiny retropie machine.
> 
> [image]
> 
> I used commit [83f1326](https://github.com/juj/fbcp-ili9341/commit/83f1326cf05f3d88e5e14000e9b2798283395d36) and started toying around with overclocking and divider settings. Currently I'm using the TURBO preset and using a CDIV value of 10. Using 8 does work, but I noticed some flickering when running ioquake3 which is almost completely absent when using 10 as a CDIV value.
> 
> Currently I'm testing Sonic because it has a nice attract mode which I can leave running for long periods of time just to test how stable the config is. Obviously I don't get a constant 60fps but there's no noticeable lag so that's awesome on such a humble machine. Even ioquake3 ran incredibly smooth as well.
> 
> So if there's anything I can do to test I'd be glad to help. It's the least I can do after the great work you provided.

Hello I'm also trynng with exact hardware configuration but I get only white screen, probably I don't have correct cabling... can you share your cabling scheme?
I tried both with no success:
Display |Raspb1

MISO    | 21       
LED     | 11       
SCK     | 23       
MOSI    | 19       
DC      | 15       
RESET   | 13       
CS      | 24       
GND     | GND      
VCC     | 5V       

ILI9341__ Pi Zero pin
VCC______17 VCC
GND______09 GND
CS________24 CE0
RESET____16 GPIO23
D/C_______18 GPIO24
MOSI_____19 MOSI
SCK______23 SCLK
LED_______12 GPIO18
MISO_____--

thanks in advance

swerveshot, 2020-04-20:
I think I hooked it up like this.
[image]

sorriso93, 2020-05-26:
> I think I hooked it up like this.
> [image]

thanks I will try

sorriso93, 2020-05-27:
Hello. Same result.  The DC seems 24 but If I compile with DC 24 I  get nothing, black screen. Compiling with DC on 18 at least I get a white screen as soon as I launch the executable. Nothing else... 
sudo cmake -DILI9341=ON -DGPIO_TFT_DATA_CONTROL=18 -DSPI_BUS_CLOCK_DIVISOR=50 ..

Any other suggestion is welcome...

sorriso93, 2020-05-27:
... I moved the led cable to 3.3v pin just for curiosity and... wow it's running!!!
[image]