juj/fbcp-ili9341 #36: DMA channel conflict on Freeplay Zero?

OrdinaryVanity, 2018-09-25:
First, fantastic work! Seeing this in action is amazing, so kudos to you!

Second, the guys over at Freeplay Tech have been helping me out, but they referred me here for more help. I'm running a Freeplay Zero, and I have been using their binaries for this driver. They have a NoDMA option that works, but the CPU usage is too extreme to get any real performance from emulators. The other option they have is fbcpZero, which is supposed to work, and the guys at Freeplay say it works, but I seem to be getting DMA channel conflicts as soon as EmulationStation loads.

Here's the output after switching to the fbcpZero driver:

```
bcm_host_get_peripheral_address: 0x20000000, bcm_host_get_peripheral_size: 33554432, bcm_host_get_sdram_address: 0x40000000
BCM core speed: current: 500000000hz, max turbo: 500000000hz. SPI CDIV: 6, SPI max frequency: 83333333hz
Allocated DMA channel 5
Allocated DMA channel 1
Enabling DMA channels Tx:5 and Rx:1
DMA hardware register file is at ptr: 0xb4b9d000, using DMA TX channel: 5 and DMA RX channel: 1
DMA hardware TX channel register file is at ptr: 0xb4b9d500, DMA RX channel register file is at ptr: 0xb4b9d100
DMA RX channel 1 was assigned another peripheral map 17!
DMA channel 0 has peripheral map 0 (is lite channel: 0, currently active: 0, current control block: (nil))
DMA channel 1 has peripheral map 17 (is lite channel: 0, currently active: 1, current control block: (nil))
DMA channel 2 has peripheral map 0 (is lite channel: 0, currently active: 0, current control block: (nil))
DMA channel 3 has peripheral map 5 (is lite channel: 0, currently active: 1, current control block: 0xdf554620)
DMA channel 4 has peripheral map 13 (is lite channel: 0, currently active: 0, current control block: (nil))
DMA channel 5 has peripheral map 6 (is lite channel: 0, currently active: 0, current control block: (nil))
DMA channel 6 has peripheral map 13 (is lite channel: 0, currently active: 1, current control block: (nil))
DMA channel 7 has peripheral map 0 (is lite channel: 1, currently active: 0, current control block: (nil))
DMA channel 8 has peripheral map 7 (is lite channel: 1, currently active: 0, current control block: (nil))
DMA channel 9 has peripheral map 0 (is lite channel: 1, currently active: 0, current control block: (nil))
DMA channel 10 has peripheral map 0 (is lite channel: 1, currently active: 0, current control block: (nil))
DMA channel 11 has peripheral map 0 (is lite channel: 1, currently active: 0, current control block: (nil))
DMA channel 12 has peripheral map 0 (is lite channel: 1, currently active: 0, current control block: (nil))
DMA channel 13 has peripheral map 0 (is lite channel: 1, currently active: 0, current control block: (nil))
DMA channel 14 has peripheral map 0 (is lite channel: 1, currently active: 0, current control block: (nil))
DMA RX channel was assigned another peripheral map!
```
After that, the display just stops updating. I've made sure that I have no overclocking set in my /boot/config.txt. My next option was to do a new build and switch the TX and RX channels, but I'm not exactly sure how to go about this on the Freeplay Zero... any advice you have would be greatly appreciated!

OrdinaryVanity, 2018-09-26:
Actually, I figured it out.

It seems my video splashscreen was causing the display to stop updating. Disabling the splashscreen in RetroPie fixed my issue. Must be something wacky with OMX Player...

closed 2018-09-26 by OrdinaryVanity

juj, 2018-10-01:
What happened here was that fbcp-ili9341 was configured to use Raspberry Pi DMA channel number 5 to send data to SPI bus, and channel number 1 to receive data from SPI bus. During execution, fbcp-ili9341 discovered that some other process on the system stomped on this, and also claimed the same DMA channel number 1 for itself, and things get crazy from there, often leading to system crash.

This kind of DMA channel collision can occur because DMA channels can only be allocated by kernel space programs, but fbcp-ili9341 is a userland program, and hence it cannot allocate DMA channels. Therefore it just picks two that are free, and keeps fingers crossed that no other software on the system will ever do the same. When that assumption does not hold, it leads to the above crash.

The issue could be fixed if someone wrote a kernel space plugin that allowed user space programs to allocate DMA channels for themselves. If so, then fbcp-ili9341 could nicely ask the kernel for "please get me two DMA channels that are guaranteed to not be in use", and other programs could do the same, and when fbcp-ili9341 quits, it would give back the DMA channels to kernel at process exit. This kind of kernel module does not currently exist that I am aware of (though I might be wrong, I don't think I have a particularly wide view to all that's going on in kernel space), so fbcp-ili9341 asks you as the user to choose which DMA channels to use. This is done with the `-DDMA_TX_CHANNEL=<num>` and `-DDMA_RX_CHANNEL=<num>` CMake configuration items.

 [BCM2835 PDF](https://www.raspberrypi.org/app/uploads/2012/02/BCM2835-ARM-Peripherals.pdf) shows the different peripheral maps ("DREQs") listed in the above log print, pages 61-62:

```
0: none
1: DSI
2: PCM TX
3: PCM RX
4: SMI
5: PWM
6: SPI TX
7: SPI RX
8: BSC/SPI Slave TX
9: BSC/SPI Slave RX
10: unused
11: e.MMC
12: UART TX
13: SD HOST
14: UART RX.
15: DSI
16: SLIMBUS MCTX.
17: HDMI
18: SLIMBUS MCRX
19: SLIMBUS DC0
20: SLIMBUS DC1
21: SLIMBUS DC2
22: SLIMBUS DC3
23: SLIMBUS DC4
24: Scaler FIFO 0 & SMI
25: Scaler FIFO 1 & SMI
26: Scaler FIFO 2 & SMI
27: SLIMBUS DC5
28: SLIMBUS DC6
29: SLIMBUS DC7
30: SLIMBUS DC8
31: SLIMBUS DC9
```

fbcp-ili9341 wants to use peripherals 6 (SPI TX) and 7 (SPI RX).

So the error print
```
DMA RX channel 1 was assigned another peripheral map 17!
```
is saying that peripheral 17 (HDMI) was assigned to DMA channel 1 instead. Perhaps OMX Player communicates directly with HDMI, and is set to use channel 1, in which case fbcp-ili9341 should be configured to use some other channel instead.

If you want to configure another channel, you can try any of the printed ones above that are "free", i.e. has `peripheral map 0`, `currently active: 0` and `control block: (nil)`. In addition, DMA RX channel has a restriction that it must not be a lite channel, but DMA TX channel can be a lite or not.

So, from the above list, any of DMA channels 0, 2, 5, 7, (perhaps 8), 9, 10, 11, 12, 13 or 14 could be used for the CMake option `-DDMA_TX_CHANNEL=<num>`. Any of the non-lite DMA channels 0, 2 or 5 could be used for the CMake option `-DDMA_RX_CHANNEL=<num>`. For example, you could try rebuilding with

```
cmake -DDMA_TX_CHANNEL=5 -DDMA_RX_CHANNEL=2 <other options> ..
```

to find a non-conflicting channel assignment. One caveat is that the assignment table printed at the time of the error only pertains to the assignment shown at the time of when the error occurred. If something is using some of the DMA channels at another time that is not persistent, then it might not show up in the table.

OrdinaryVanity, 2018-10-01:
I knew there was a conflict, but I couldn't figure out what was
conflicting. I'm pretty sure it's OMX Player that emulationstation uses,
since this issue is not present when I don't have a video splashscreen.

I let the guys at Freeplay Tech know, so maybe they can release a binary
with different channels set. Thanks for the detailed response!