juj/fbcp-ili9341 #38: White screen only with custom resolution

BigJk, 2018-09-30:
**System:** Raspberry Pi 3
**Screen:** 3.2 Inch 320x240 TFT (ILI9341)
**Build Parameter:** ``-DILI9341=ON -DGPIO_TFT_DATA_CONTROL=24 -DGPIO_TFT_RESET_PIN=25 -DGPIO_TFT_BACKLIGHT=18 -DSPI_BUS_CLOCK_DIVISOR=6 -DDISPLAY_ROTATE_180_DEGREES=ON -DSTATISTICS=0``

If I don't use a custom resolution everything works. I can see the screen and scaling is applied, but if I set the HDMI resolution to the one of the screen it will just show a white screen. No error thrown. Is there something I could be doing wrong?

**/boot/config.txt**
```
hdmi_group=2
hdmi_mode=87
hdmi_drive=2
hdmi_cvt=320 240 60 1 0 0 0
hdmi_force_hotplug=1
```

**console output**
```
bcm_host_get_peripheral_address: 0x3f000000, bcm_host_get_peripheral_size: 16777216, bcm_host_get_sdram_address: 0xc0000000
BCM core speed: current: 250000000hz, max turbo: 400000000hz. SPI CDIV: 6, SPI max frequency: 66666667hz
Allocated DMA channel 7
Allocated DMA channel 1
Enabling DMA channels Tx:7 and Rx:1
DMA hardware register file is at ptr: 0x75ae6000, using DMA TX channel: 7 and DMA RX channel: 1
DMA hardware TX channel register file is at ptr: 0x75ae6700, DMA RX channel register file is at ptr: 0x75ae6100
Resetting DMA channels for use
DMA all set up
Initializing display
Resetting display at reset GPIO pin 25
Creating SPI task thread
InitSPI done
DISPLAY_FLIP_ORIENTATION_IN_SOFTWARE: Swapping width/height to update display in portrait mode to minimize tearing.
Relevant source display area size with overscan cropped away: 240x320.
Source GPU display is 240x320. Output SPI display is 240x320 with a drawable area of 240x320. Applying scaling factor horiz=1.00x & vert=1.00x, xOffset: 0, yOffset: 0,                                      scaledWidth: 240, scaledHeight: 320
Creating dispmanX resource of size 240x320 (aspect ratio=0.750000).
GPU grab rectangle is offset x=0,y=0, size w=240xh=320, aspect ratio=0.750000
All initialized, now running main loop...
```

juj, 2018-10-01:
Hmm, nothing here stands out. For cross-reference, my `/boot/config.txt` with ILI9341 reads

```
hdmi_group=2
hdmi_mode=87
hdmi_cvt=320 240 60 1 0 0 0
hdmi_force_hotplug=1
dtparam=audio=on
gpu_mem_256=128
gpu_mem_512=256
gpu_mem_1024=256
overscan_scale=1
avoid_warnings=2
```

I see I don't have `hdmi_drive=2` in particular, though cross-referencing at https://www.raspberrypi.org/documentation/configuration/config-txt/video.md , that should be fine.

Random things to try may be to
 - omit `-DDISPLAY_ROTATE_180_DEGREES=ON` to see if that behaves any different,
 - disable DMA, or try different DMA channels,
 - do a full power off of the Pi after changing /boot/config.txt, instead of a reboot (in case did not before),
 - try a higher SPI CDIV setting, e.g. 8 or 10,
 - comment out `DISPLAY_FLIP_ORIENTATION_IN_SOFTWARE`:

```diff
diff --git a/config.h b/config.h
index ced2b29..b448d6b 100644
--- a/config.h
+++ b/config.h
@@ -188,7 +188,7 @@
 // DISPLAY_FLIP_ORIENTATION_IN_SOFTWARE enabled: traditional no-vsync tearing (tear line runs in portrait
 // i.e. narrow direction)
 #if !defined(SINGLE_CORE_BOARD)
-#define DISPLAY_FLIP_ORIENTATION_IN_SOFTWARE
+//#define DISPLAY_FLIP_ORIENTATION_IN_SOFTWARE
 #endif
```
 - (last resort, try an earlier git version of fbcp-ili9341 in case there may have been a regression)

BigJk, 2018-10-21:
So in the end commenting out ``DISPLAY_FLIP_ORIENTATION_IN_SOFTWARE`` made it work like expected. Thanks for helping out!

closed 2018-10-21 by BigJk