juj/fbcp-ili9341 #39: Is this comatible with Driver: ILI9486 Dots: 320x480 SKU: MPI3501?

xjamxx, 2018-10-17:
I tried with dwaverhare35b_ili9486 and div 8, and got half screen in black and the other random horizontal lines that update, with more random...

I know this driver requires 4-wire spi, sadly i don't even know the pin out of this display it came with no manual.
It is probably a 3 wire, but who knows may some one knows something about this display.

juj, 2018-10-17:
Fbcp-ili9341 should work with ILI9486 and ILI9486L displays. Try increasing the SPI bus clock divisor to a safe higher value to start with, e.g. 20 or 30.

If there are pixels rendered on the screen, the driver is getting something right at least, and there may be a chance. One possibility is that the Data/Control line is not

xjamxx, 2018-10-17:
ok, i put 20 and it worked, got flip up down (need to change config) image, but its working quite smooth even with 20 (tryied xmen clone wars gens and ff6 snes they look amazing vs the rpifbcp), should i lower it?

Thanks, you did a lot already.

closed 2018-10-17 by xjamxx

juj, 2018-10-22:
Great to hear! You can definitely try to find a good middle ground between CDIV of 20 and 6, to find where the limit goes for your display. See also section https://github.com/juj/fbcp-ili9341#tuning-performance if you want to fine tune even further to find refresh rates between of what the possible CDIV values can give you.

CDIV 16 on Pi 3B+ will be 400MHz/16 = 25MHz, whereas CDIV 14 on Pi 3B+ will be 28.57MHz. You can try underclocking to e.g. 380MHz and use CDIV 14 for 380/14=27.14MHz to see if it gives better performance. Or set 390 MHz and CDIV 14 for 27.86MHz. (27.86MHz would be +11.4% more bandwidth over 25MHz, so perhaps a frame or two better fps). 25MHz does sound great already for ILI9486, not bad at all for those controllers.