juj/fbcp-ili9341 #41: having problems building for pi2b

LordNuke, 2018-11-03:
adafruit 2.8" 320x240 tft
pi 2b
raspbian stretch fresh instal+updates

everything worked fine until i  get hung up on the make -j line.
getting these compile errors:

`pi@raspberrypi:~/fbcp-ili9341/build $ sudo make -j
[ 11%] Building CXX object CMakeFiles/fbcp-ili9341.dir/ili9341.cpp.o
[ 11%] Building CXX object CMakeFiles/fbcp-ili9341.dir/dma.cpp.o
[ 16%] Building CXX object CMakeFiles/fbcp-ili9341.dir/display.cpp.o
[ 22%] Building CXX object CMakeFiles/fbcp-ili9341.dir/diff.cpp.o
[ 27%] Building CXX object CMakeFiles/fbcp-ili9341.dir/statistics.cpp.o
[ 33%] Building CXX object CMakeFiles/fbcp-ili9341.dir/spi.cpp.o
In file included from /home/pi/fbcp-ili9341/display.cpp:3:0:
/home/pi/fbcp-ili9341/spi.h: In member function ‘uint8_t* SPITask::PayloadStart()’:
/home/pi/fbcp-ili9341/spi.h:121:58: error: ‘size9BitTaskWithPadding’ was not declared in this scope
   inline uint8_t *PayloadStart() { return data + (size - size9BitTaskWithPadding); }
                                                          ^~~~~~~~~~~~~~~~~~~~~~~
/home/pi/fbcp-ili9341/spi.h: In member function ‘uint32_t SPITask::PayloadSize() const’:
/home/pi/fbcp-ili9341/spi.h:123:48: error: ‘size9BitTaskWithPadding’ was not declared in this scope
   inline uint32_t PayloadSize() const { return size9BitTaskWithPadding - SPI_9BIT_TASK_PADDING_BYTES; }
                                                ^~~~~~~~~~~~~~~~~~~~~~~
CMakeFiles/fbcp-ili9341.dir/build.make:86: recipe for target 'CMakeFiles/fbcp-ili9341.dir/display.cpp.o' failed
make[2]: *** [CMakeFiles/fbcp-ili9341.dir/display.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /home/pi/fbcp-ili9341/diff.cpp:6:0:
/home/pi/fbcp-ili9341/spi.h: In member function ‘uint8_t* SPITask::PayloadStart()’:
/home/pi/fbcp-ili9341/spi.h:121:58: error: ‘size9BitTaskWithPadding’ was not declared in this scope
   inline uint8_t *PayloadStart() { return data + (size - size9BitTaskWithPadding); }
                                                          ^~~~~~~~~~~~~~~~~~~~~~~
/home/pi/fbcp-ili9341/spi.h: In member function ‘uint32_t SPITask::PayloadSize() const’:
/home/pi/fbcp-ili9341/spi.h:123:48: error: ‘size9BitTaskWithPadding’ was not declared in this scope
   inline uint32_t PayloadSize() const { return size9BitTaskWithPadding - SPI_9BIT_TASK_PADDING_BYTES; }
                                                ^~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/pi/fbcp-ili9341/ili9341.cpp:5:0:
/home/pi/fbcp-ili9341/spi.h: In member function ‘uint8_t* SPITask::PayloadStart()’:
/home/pi/fbcp-ili9341/spi.h:121:58: error: ‘size9BitTaskWithPadding’ was not declared in this scope
   inline uint8_t *PayloadStart() { return data + (size - size9BitTaskWithPadding); }
                                                          ^~~~~~~~~~~~~~~~~~~~~~~
CMakeFiles/fbcp-ili9341.dir/build.make:62: recipe for target 'CMakeFiles/fbcp-ili9341.dir/diff.cpp.o' failed
make[2]: *** [CMakeFiles/fbcp-ili9341.dir/diff.cpp.o] Error 1
/home/pi/fbcp-ili9341/spi.h: In member function ‘uint32_t SPITask::PayloadSize() const’:
/home/pi/fbcp-ili9341/spi.h:123:48: error: ‘size9BitTaskWithPadding’ was not declared in this scope
   inline uint32_t PayloadSize() const { return size9BitTaskWithPadding - SPI_9BIT_TASK_PADDING_BYTES; }
                                                ^~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/pi/fbcp-ili9341/statistics.cpp:15:0:
/home/pi/fbcp-ili9341/spi.h: In member function ‘uint8_t* SPITask::PayloadStart()’:
/home/pi/fbcp-ili9341/spi.h:121:58: error: ‘size9BitTaskWithPadding’ was not declared in this scope
   inline uint8_t *PayloadStart() { return data + (size - size9BitTaskWithPadding); }
                                                          ^~~~~~~~~~~~~~~~~~~~~~~
/home/pi/fbcp-ili9341/spi.h: In member function ‘uint32_t SPITask::PayloadSize() const’:
/home/pi/fbcp-ili9341/spi.h:123:48: error: ‘size9BitTaskWithPadding’ was not declared in this scope
   inline uint32_t PayloadSize() const { return size9BitTaskWithPadding - SPI_9BIT_TASK_PADDING_BYTES; }
                                                ^~~~~~~~~~~~~~~~~~~~~~~
CMakeFiles/fbcp-ili9341.dir/build.make:422: recipe for target 'CMakeFiles/fbcp-ili9341.dir/statistics.cpp.o' failed
make[2]: *** [CMakeFiles/fbcp-ili9341.dir/statistics.cpp.o] Error 1
CMakeFiles/fbcp-ili9341.dir/build.make:206: recipe for target 'CMakeFiles/fbcp-ili9341.dir/ili9341.cpp.o' failed
make[2]: *** [CMakeFiles/fbcp-ili9341.dir/ili9341.cpp.o] Error 1
In file included from /home/pi/fbcp-ili9341/dma.cpp:17:0:
/home/pi/fbcp-ili9341/spi.h: In member function ‘uint8_t* SPITask::PayloadStart()’:
/home/pi/fbcp-ili9341/spi.h:121:58: error: ‘size9BitTaskWithPadding’ was not declared in this scope
   inline uint8_t *PayloadStart() { return data + (size - size9BitTaskWithPadding); }
                                                          ^~~~~~~~~~~~~~~~~~~~~~~
/home/pi/fbcp-ili9341/spi.h: In member function ‘uint32_t SPITask::PayloadSize() const’:
/home/pi/fbcp-ili9341/spi.h:123:48: error: ‘size9BitTaskWithPadding’ was not declared in this scope
   inline uint32_t PayloadSize() const { return size9BitTaskWithPadding - SPI_9BIT_TASK_PADDING_BYTES; }
                                                ^~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/pi/fbcp-ili9341/spi.cpp:16:0:
/home/pi/fbcp-ili9341/spi.h: In member function ‘uint8_t* SPITask::PayloadStart()’:
/home/pi/fbcp-ili9341/spi.h:121:58: error: ‘size9BitTaskWithPadding’ was not declared in this scope
   inline uint8_t *PayloadStart() { return data + (size - size9BitTaskWithPadding); }
                                                          ^~~~~~~~~~~~~~~~~~~~~~~
/home/pi/fbcp-ili9341/spi.h: In member function ‘uint32_t SPITask::PayloadSize() const’:
/home/pi/fbcp-ili9341/spi.h:123:48: error: ‘size9BitTaskWithPadding’ was not declared in this scope
   inline uint32_t PayloadSize() const { return size9BitTaskWithPadding - SPI_9BIT_TASK_PADDING_BYTES; }
                                                ^~~~~~~~~~~~~~~~~~~~~~~
CMakeFiles/fbcp-ili9341.dir/build.make:350: recipe for target 'CMakeFiles/fbcp-ili9341.dir/spi.cpp.o' failed
make[2]: *** [CMakeFiles/fbcp-ili9341.dir/spi.cpp.o] Error 1
CMakeFiles/fbcp-ili9341.dir/build.make:110: recipe for target 'CMakeFiles/fbcp-ili9341.dir/dma.cpp.o' failed
make[2]: *** [CMakeFiles/fbcp-ili9341.dir/dma.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/fbcp-ili9341.dir/all' failed
make[1]: *** [CMakeFiles/fbcp-ili9341.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
`

juj, 2018-11-03:
Try doing a `git pull`, there were changes that went in just recently related to here. Does that help?

LordNuke, 2018-11-04:
ah thanks that worked.

juj, 2018-11-04:
Great!

closed 2018-11-04 by juj