juj/fbcp-ili9341 #44: fbcp-ili9341 does not initialize on boot

donutkeyboards, 2018-11-15:
Hello there,

I installed and compiled the driver for my Raspberry Pi Zero W running Retropie over Raspbian Stretch, and it seems that it won't start the driver after a reboot. The log says that it can't find the directory specified in rc.local (which I definitely copied and pasted there). Any advice? 

Thanks!

donutkeyboards, 2018-11-15:
More specifically, it says "command not found" on boot

donutkeyboards, 2018-11-15:
FIXED! I was simply not inputting the correct path. It might be worth mentioning in the documentation that the command is more like sudo /home/pi/fbcp-ili9341/build/fbcp-ili9341 & for the novices like myself.

juj, 2018-11-19:
Thanks! This is a good suggestion, added an example in above commit to illustrate the actual line to add in the `/etc/rc.local` file.

closed 2018-11-19 by juj