juj/fbcp-ili9341 #46: Image smaller than screen - STM7789

RmigD, 2018-11-24:
Hi. The screen is already small, so I'd love to make the image take up the whole pixel area. This is a GameBoy Colour ROM.
[image]

Any idea?

juj, 2018-11-24:
Thanks, much appreciated!

Make sure /boot/config.txt is configured to have a framebuffer that is the proper size for the display (240x240 iirc?). See [this section](https://github.com/juj/fbcp-ili9341#configuring-hdmi-and-tft-display-sizes). When fbcp-ili9341 boots up, it prints out the sizes of the input and output framebuffers it sees, and the scaling factor that it performs. To get full 1:1 output, check that those prints do not output any scaling to appear (input and output should show 240x240).

If that happens, and scaling is 100%, then it is up to the application in question to expand its rendering to the full framebuffer size. The way that is done depends on the specific application. If this is RetroPie/RetroArch setup, there is some documentation starting at e.g. https://github.com/RetroPie/RetroPie-Setup/wiki/Smaller-RetroArch-Screen .

juj, 2018-12-10:
Marking this as closed. If this is still bugging, check out the startup log of fbcp-ili9341, that should print the detected input and output framebuffer sizes, that should help investigating where the black borders come from.

closed 2018-12-10 by juj