juj/fbcp-ili9341 #48: Driver don't work with official Waveshare 3.5inch LCD (B)

juanjointera5, 2018-11-30:
Hello!

(sorry for my bad english)

First, thanks for this awesome driver!

I'm trying it with an official Waveshare 3.5 inch LCD IPS (B) display (not a clone), which is supported out of the box. But, I only got black screen with sometimes flash in white.

I used this cmake:

    pi@raspberrypi:~/fbcp-ili9341/build $ cmake -DSPI_BUS_CLOCK_DIVISOR=8 -DWAVESHARE35B_ILI9486=ON -DSTATISTICS=1 ..

The output:

    -- Doing a Release build
    -- Board revision: a020d3
    -- Detected this Pi to be one of: Pi 2B rev. 1.2, 3B, 3B+, CM3 or CM3 lite, with 4 hardware cores and ARMv8-A instruction set CPU.
    -- Enabling optimization flags that target ARMv8-A instruction set (Pi 2B >= rev. 1.2, 3B, 3B+, CM3 or CM3 lite)
    -- Using 4-wire SPI mode of communication, with GPIO pin 10 for Data/Control line
    -- Using GPIO pin 25 for Reset line
    -- Scaling source image to view. If the HDMI resolution does not match the SPI display resolution, this will produce blurriness. Match the HDMI display resolution with the SPI resolution in /boot/config.txt to get crisp pixel perfect rendering, or alternatively pass -DDISPLAY_CROPPED_INSTEAD_OF_SCALING=ON to crop instead of scale if you want to view the center of the screen pixel perfect when HDMI and SPI resolutions do not match.
    -- Preserving aspect ratio when scaling source image to the SPI display, introducing letterboxing/pillarboxing if HDMI and SPI aspect ratios are different (Pass -DDISPLAY_BREAK_ASPECT_RATIO_WHEN_SCALING=ON to stretch HDMI to cover full screen if you do not care about aspect ratio)
    -- SPI_BUS_CLOCK_DIVISOR set to 8. Try setting this to a higher value (must be an even number) if this causes problems. Display update speed = core_freq/divisor. (on Pi3B, by default core_freq=400)
    -- USE_DMA_TRANSFERS enabled, this improves performance. Try running CMake with -DUSE_DMA_TRANSFERS=OFF it this causes problems, or try adjusting the DMA channels to use with -DDMA_TX_CHANNEL=<num> -DDMA_RX_CHANNEL=<num>.
    -- Targeting Adafruit PiTFT with ILI9340/ILI9341
    -- Configuring done
    -- Generating done
    -- Build files have been written to: /home/pi/fbcp-ili9341/build

The first problem I see is `-- Targeting Adafruit PiTFT with ILI9340/ILI9341`. But I continue.

    pi@raspberrypi:~/fbcp-ili9341/build $ make -j
    [  5%] Building CXX object CMakeFiles/fbcp-ili9341.dir/fbcp-ili9341.cpp.o
    [ 11%] Building CXX object CMakeFiles/fbcp-ili9341.dir/diff.cpp.o
    [ 16%] Building CXX object CMakeFiles/fbcp-ili9341.dir/display.cpp.o
    [ 22%] Building CXX object CMakeFiles/fbcp-ili9341.dir/dma.cpp.o
    [ 27%] Building CXX object CMakeFiles/fbcp-ili9341.dir/mem_alloc.cpp.o
    [ 33%] Building CXX object CMakeFiles/fbcp-ili9341.dir/ili9341.cpp.o
    [ 38%] Building CXX object CMakeFiles/fbcp-ili9341.dir/gpu.cpp.o
    [ 44%] Building CXX object CMakeFiles/fbcp-ili9341.dir/hx8357d.cpp.o
    [ 50%] Building CXX object CMakeFiles/fbcp-ili9341.dir/text.cpp.o
    [ 55%] Building CXX object CMakeFiles/fbcp-ili9341.dir/ili9486.cpp.o
    [ 61%] Building CXX object CMakeFiles/fbcp-ili9341.dir/spi.cpp.o
    [ 66%] Building CXX object CMakeFiles/fbcp-ili9341.dir/st7735r.cpp.o
    [ 72%] Building CXX object CMakeFiles/fbcp-ili9341.dir/mz61581.cpp.o
    [ 77%] Building CXX object CMakeFiles/fbcp-ili9341.dir/statistics.cpp.o
    [ 83%] Building CXX object CMakeFiles/fbcp-ili9341.dir/mailbox.cpp.o
    [ 88%] Building CXX object CMakeFiles/fbcp-ili9341.dir/keyboard.cpp.o
    [ 94%] Building CXX object CMakeFiles/fbcp-ili9341.dir/ssd1351.cpp.o
    [100%] Linking CXX executable fbcp-ili9341
    [100%] Built target fbcp-ili9341

Then run driver:

    pi@raspberrypi:~/fbcp-ili9341/build $ sudo ./fbcp-ili9341
    bcm_host_get_peripheral_address: 0x3f000000, bcm_host_get_peripheral_size: 16777216, bcm_host_get_sdram_address: 0xc0000000
    BCM core speed: current: 250000000hz, max turbo: 400000000hz. SPI CDIV: 8, SPI max frequency: 50000000hz
    Allocated DMA channel 7
    Allocated DMA channel 1
    Enabling DMA channels Tx:7 and Rx:1
    DMA hardware register file is at ptr: 0x75b66000, using DMA TX channel: 7 and DMA RX channel: 1
    DMA hardware TX channel register file is at ptr: 0x75b66700, DMA RX channel register file is at ptr: 0x75b66100
    Resetting DMA channels for use
    DMA all set up
    Initializing display
    Resetting display at reset GPIO pin 25
    Creating SPI task thread
    InitSPI done
    DISPLAY_FLIP_ORIENTATION_IN_SOFTWARE: Swapping width/height to update display in portrait mode to minimize tearing.
    Relevant source display area size with overscan cropped away: 480x320.
    Source GPU display is 480x320. Output SPI display is 240x320 with a drawable area of 240x320. Applying scaling factor horiz=0.50x & vert=0.50x, xOffset: 0, yOffset: 80, scaledWidth: 240, scaledHeight: 160
    Creating dispmanX resource of size 240x160 (aspect ratio=1.500000).
    GPU grab rectangle is offset x=0,y=0, size w=240xh=160, aspect ratio=1.500000
    All initialized, now running main loop...

Here, I see the problem `Output SPI display is 240x320 with a drawable area of 240x320`. Display resolution is 480x320, not 240x320. I have forced 480x320 HDMI resolution in `/boot/config.txt`.

The display turns black, and if I move the mouse the screen sometimes flash in white.

I'm using a Raspberry Pi 3 B+ with latest Raspbian. No legacy fbcp driver is loaded and SPI is disabled in `/boot/config.txt`.

Maybe I'm missing something ?

Thanks!!!

juanjointera5, 2018-12-01:
Hello again:

Just learned that I must do a `rm -r *` in the build directory before every `cmake`. This correctly detect the board setting. But still don't work.

After a lot of trial and error, just discovered that my display is a 'Rev2.0' from Waveshare. Look at: https://www.waveshare.com/wiki/3.5inch_RPi_LCD_(B)

So, parameters for the previous version don't work. Data pin has changed, now is GPIO 24. So, I just finally got the display working with this cmake:

    cmake -DSPI_BUS_CLOCK_DIVISOR=18 -DILI9486=ON -DGPIO_TFT_DATA_CONTROL=24 -DGPIO_TFT_RESET_PIN=25 -DSTATISTICS=1 ..

Less values for -DSPI_BUS_CLOCK_DIVISOR= make display full of dots.

But, this get a weird color display configuration:

[image]

I already tried -DDISPLAY_SWAP_BGR=ON, -DDISPLAY_INVERT_COLORS=ON and ever both but don't work. So, maybe this display have different colors channels.

Thanks.

juj, 2018-12-10:
Marking this as closed. If this is still an issue, try poking the above invert command line code to toggle the inversion parameter, testing on my ILI9486, adjusting that does invert the display colors between negative and normal.

closed 2018-12-10 by juj

thepatchwerkboy, 2019-05-15:
Getting the exact same problem as juanjointera5 with the Waveshare 3.5(B) IPS rev 2... 
[image]

juj, 2019-05-17:
Noted - looks like something relevant has changed in revision 2 compared to the original rev 1. More conversation in #83.