juj/fbcp-ili9341 #49: Adafruit 3.5 TFT

wstevens0n, 2018-12-05:
Hello juj,

I'm having issues with the pitft_35r_hx8357d.

I am utilizing the following cmake flags and the screen is flipped horizontally.

cmake -DSPI_BUS_CLOCK_DIVISOR=6 -DADAFRUIT_HX8357D_PITFT=ON -DDISPLAY_SWAP_BGR=ON -DSTATISTICS=0 ..

Thanks.

wstevens0n, 2018-12-05:
I got it working, it required a full reboot oddly.

Here were my final flags.

cmake -DSPI_BUS_CLOCK_DIVISOR=8 -DADAFRUIT_HX8357D_PITFT=ON -DDISPLAY_SWAP_BGR=OFF -DSTATISTICS=0 -DDISPLAY_CROPPED_INSTEAD_OF_SCALING=OFF -DDISPLAY_BREAK_ASPECT_RATIO_WHEN_SCALING=ON -DSINGLE_CORE_BOARD=ON -DHX8357D=ON ..

/boot/config.txt
disable_overscan=1
framebuffer_width=480
framebuffer_height=320
#hdmi_force_hotplug=1
hdmi_group=2
hdmi_mode=87
hdmi_cvt=480 320 60 1 0 0 0
overscan_scale=1

juj, 2018-12-10:
Great to hear!

Sometimes I do find that displays get in an odd state after e.g. running them with too high bus speed, and full power off seems to be needed to recover.

closed 2018-12-10 by juj