juj/fbcp-ili9341 #51: I am trying 128x128 SSD1351 display SPI 7 contacts

Jabberwacky, 2018-12-14:
The system is: 
Machine model: Raspberry Pi Zero Rev 1.3
Linux version 4.14.83+ (dom@dom-XPS-13-9370) - some Raspbian.

Display started to work in a 128x98 SSD1351 configuration. I even changed the resolution to 128x128 in ssd1351.h file. BUT it still "GPU grab rectangle is offset x=0,y=0, size w=128xh=96, aspect ratio=1.333333" So the screen shows me very distorted pictures. Where i can tune it up? I can't find.

Thanx in advance!

P.S.
"source display area size with overscan cropped away: 640x480"

Jabberwacky, 2018-12-14:
Thank you. /boot/config.txt I had already tested. It can set just several predefined resolutions. And will reject something else. Even when I am using hdmi_cvt commands. This string DDISPLAY_BREAK_ASPECT_RATIO_WHEN_SCALING=ON  did not changed nothing. Searching solutions for aspect ratio. Will report

Jabberwacky, 2018-12-14:
"-DDISPLAY_CROPPED_INSTEAD_OF_SCALING=ON .." This shows me cropped but correct fonts. Will try to change HDMI somehow

Jabberwacky, 2018-12-14:
This issue was solved with help of your suggestion. I have some other questions. I got some troubles with starting driver in autostart mode. And I still unable to show video. I can see the mess of dots, or a regular white screen instead of picture, when starting to use omxplayer. There is some DMA access troubles. But this is out from theme of this issue. Thank you for your great work!!!

juj, 2018-12-14:
Omxplayer seems to use DMA transfers on its own, which collides with fbcp-ili9341s use of DMA. You can try finding other DMA channel numbers for fbcp-ili9341 to use to see if the channel collision can be avoided. Be careful though, SD card also uses DMA, so if you use its DMA channel, it may corrupt the contents of the card, so backup is advised.

Jabberwacky, 2018-12-14:
Thank you very much! I'll try to use raspivid instead. This is a goal of my device - to show the camera output. :)))  And some additional info as a text