juj/fbcp-ili9341 #53: Can't compile for Kuman 3.5'' TFT

spaghiajoeojo, 2018-12-17:
Hi, my display is a kuman and I can't compile the driver to speed up its framerate. I found this in official driver cd:

[3.5 inch touch screen datasheet  V1.0.pdf](https://github.com/juj/fbcp-ili9341/files/2687014/3.5.inch.touch.screen.datasheet.V1.0.pdf)

I tried ILI9486 and ILI9486L but I can't really say what pin is Data Control. For what I see it can also be an ILI9488L... can you help me?

juj, 2018-12-17:
Unfortunately I have not tested Kuman displays. If it is this one http://www.kumantech.com/kuman-35quot-inch-480x320-rgb-tft-lcd-pixels-touch-screen-with-touch-pen-for-raspberry-pi-3-2-model-b-b-a-a-module-spi-interface-sc06_p0014.html , then it resembles the KeDei display a lot, which was discussed in #40.

To try the KeDei driver in hope the Kuman display and KeDei displays might be clones of each other, run CMake with `-DKEDEI_V63_MPI3501=ON` setting, and a relatively safe SPI speed to start with, e.g. `-DSPI_BUS_CLOCK_DIVISOR=30`.

spaghiajoeojo, 2018-12-17:
Sadly it doesn't work with Kedei preset...

In /boot/config.txt I found
```
# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on
dtoverlay=waveshare35a
#dtoverlay=ads7846,cs=1,penirq=17,penirq_pull=2,speed=1000000,keep_vref_on=1,sw$
hdmi_force_hotplug=1
#max_usb_current=1
hdmi_group=2
hdmi_mode=1
hdmi_mode=87
hdmi_cvt 480 320 60 6 0 0 0
hdmi_drive=2
```
So I tried `-DWAVESHARE35B_ILI9486=ON` but nothing changed :(

juj, 2018-12-17:
Then it looks like Kuman displays fall outside fbcp-ili9341 support. If you are able to reverse engineer the protocol, you can try cross-referencing the support against existing drivers, e.g. the KeDei or the ILI9486/ILI9486L ones, perhaps one of those can be modded to work also on Kuman displays.

juj, 2018-12-20:
I'll mark this retired, but keep it open, so that other people will see that Kuman displays have been observed to not work well. Thanks for reporting!

juj, 2018-12-21:
Youtube user haveallgames documented that the current KeDei backend does not seem to be able to support all KeDei displays, his results are at https://www.youtube.com/watch?v=fzNl4dB_Vis and https://www.youtube.com/watch?v=FrldurB37QM . It is hence possible that even if the Kuman display you have is same or similar to the KeDei display, that there is some notable difference that fbcp-ili9341 is not currently able to take into account, that is causing an issue here as well.

WiredLife, 2019-11-15:
cmake -DWAVESHARE35B_ILI9486=ON -DSPI_BUS_CLOCK_DIVISOR=25 -DDMA_TX_CHANNEL=9 -DDMA_RX_CHANNEL=6 -DSTATISTICS=0 ..

works for my Kuman 3.5'' TFT @ Raspberry Pi 4B 1G

spaghiajoeojo, 2019-11-16:
Thanks to @WiredLife works for me too:

cmake -DWAVESHARE35B_ILI9486=ON -DSPI_BUS_CLOCK_DIVISOR=25 -DDMA_TX_CHANNEL=9 -DDMA_RX_CHANNEL=5 -DSTATISTICS=0 ..

on raspi 3 B and raspbian

closed 2019-11-16 by spaghiajoeojo

spaghiajoeojo, 2019-11-16:
Can you unset this as "retired" please? @juj 
It is working on Kuman 3.5" 😃

juj, 2019-11-17:
Nice! Good info and great testing here!

WiredLife, 2019-11-17:
here some tips
use #define ALL_TASKS_SHOULD_DMA to safe cpu
and #define NO_INTERLACING to get a good picture on videos

btw you can go up to -DSPI_BUS_CLOCK_DIVISOR=24 but NOT -DSPI_BUS_CLOCK_DIVISOR=23

@spaghiajoeojo do you get more as 10 fps on video?
maybe someone has a tip to get higher fps?

spaghiajoeojo, 2019-11-18:
According to the guide of @juj DSPI_BUS_CLOCK_DIVISOR should be an even number, I managed to set its value to 16 and core_freq to 280 in boot/config.txt. Since my pi is used mainly to stream from my main pc it runs smoothly at 30-40 fps interlaced. I can play GTA5 running with steam link on a marvelous 480x320 display!

WiredLife, 2019-11-18:
@spaghiajoeojo are you sure you have the kuman display?
i get disturbed picture when i go under 24
the value of 16 is impossible here
btw i have a raspberry pi 4 with core_freq of 500