juj/fbcp-ili9341 #54: over exposure on watterott rpi-display (based on ili9341 controller)

normanzb, 2019-01-02:
Watterott rpi-display b+ is a 2.8'' inch display based on ili9341 controller, when it is enabled with fbcp-ili9341. the graphical looks over exposed by comparing to the origin fbcp.

I tried to compile without gamma correction at https://github.com/juj/fbcp-ili9341/blob/master/ili9341.cpp#L94, the result looks better and quite close to the original fbcp result:

screenshot of retropie with gamma correction, you can barely see the menu text:
https://photos.app.goo.gl/DUqaJamDo2hWJEe16

without gamma correction:
https://photos.app.goo.gl/5UnFw7suYmZjB2JD7

Here is the configuration for the build:
```
cmake -DILI9341=ON -DGPIO_TFT_DATA_CONTROL=24 -DSPI_BUS_CLOCK_DIVISOR=6 -DGPIO_TFT_BACKLIGHT=18 -DGPIO_TFT_RESET_PIN=23 -DSTATISTICS=0 -DBACKLIGHT_CONT ROL=ON ..
```

Here is the docs for the display:
https://learn.watterott.com/rpi-display/

juj, 2019-01-04:
Thanks for reporting, very interesting to know. Probably the best way will be to allow to customize the gamma curve, or to disable it. I'll make a note to add this at some point - looks like you already found the right place to go and disable it manually.

normanzb, 2019-01-04:
thanks for reply @juj , yes for me I managed compiled it without gamma correction and it works fabulous, thanks again for the work. 
And I agree it would be great if we can make it customisable.

ChristopherRush, 2020-02-07:
@normanzb How did you get this display to work. This is all I see:
[image]

juj, 2020-12-08:
Closing out old issues, sorry I missed the last comment in this thread. Feel free to open a new issue if still actively struggling on a problem.

closed 2020-12-08 by juj