juj/fbcp-ili9341 #56: Added ability to rotate 90 and 270 degrees

nhenry1647, 2019-01-07:


juj, 2019-01-09:
Thanks, much appreciated.

It should be possible to rotate 90 and 270 degrees, though fbcp-ili9341 does not use degree numbers to specify these. Comment out https://github.com/juj/fbcp-ili9341/blob/ccebad5362623c435dcb5be11cc891f246ec5c50/config.h#L162 to do a 90 degrees rotation, and combine that with uncommenting https://github.com/juj/fbcp-ili9341/blob/ccebad5362623c435dcb5be11cc891f246ec5c50/config.h#L158 to rotate the 90 degrees upside down to a 270 degrees.

nhenry1647, 2019-01-09:
Wow, thanks for the quick response.  I thought I combed that file but missed that.  I did what you suggested and it did work (sort of).  I now see the display as a mirror image.  I tried a few things and tried to figure that out on my own but got no where and ran out of time.  I will dig some more later.  

Would it be a simple edit to get the image to flip to non-mirrored?  I've added a picture of the project I am working on to show the mirror image.  I'm using the HDMI out port of the pi as well as the TFT which is why the config.txt needs to have the display_rotate as 0.  Thanks again!!!

[image]

juj, 2019-01-09:
Nice project, great photo :)

Which display controller is that in there? If the image shows up flipped, then probably one of the MADCTL (Memory Addressing Control) registers are not getting specified appropriately for the display controller with the given options. For the display controller code code in question, try adding a column or row flip, e.g. if it's an ILI9341, then right before this line:

https://github.com/juj/fbcp-ili9341/blob/ccebad5362623c435dcb5be11cc891f246ec5c50/ili9341.cpp#L64

add

```c++
madctl ^= MADCTL_ROW_ADDRESS_ORDER_SWAP;
```

or 

```c++
madctl ^= MADCTL_COLUMN_ADDRESS_ORDER_SWAP;
```

one of those will mirror the output horizontally, another one vertically.

juj, 2019-01-10:
For ST7789VW you can try adding one of the swaps before this line

https://github.com/juj/fbcp-ili9341/blob/ccebad5362623c435dcb5be11cc891f246ec5c50/st7735r.cpp#L69

nhenry1647, 2019-01-15:
Thanks for all your time.  Unfortunately those procedures did not work
either, even after removing build and starting fresh.  No worries though, i
just took everything appart and phyisically rotated the screen.

I just wanted to say thanks for the help and I really appreciate the code
you have here.  Everything works great now.

juj, 2019-01-16:
Hmm, perhaps ST7789VW interprets the MADCTL register differently and does not have the same address swapping capability.

Great to hear you got it resolved. Rotating the screen upright also has an added benefit that it will retain the internal panel scan order of the display to match with the memory order, so that diagonal tearing will not occur.

closed 2019-01-16 by juj

froggestspirit, 2019-01-19:
I had the same issue, but for anyone reading, doing the column flip in the st7735r.cpp fixed it