juj/fbcp-ili9341 #57: Cannot run "make -j"

C112358, 2019-01-09:
Hi!
i'm trying to get my ili9241 working on a Raspberry Pi 3A+ with Retropie. 
For further clarification, I run:
cmake -DSPI_BUS_CLOCK_DIVISOR=6 -DARMV8A=ON DILI9341=ON ..
before attempting to run "make -j"
While the "cmake" command appears to complete without issues, "make -j"  gives me problems and prevents the above code from working. 
[image]
Thank you
P.S. Thanks for everything you've done with ILI displays on the Raspberry Pi :)

juj, 2019-01-09:
Thanks, much appreciated!

This reads as if there is something wrong in the way you obtained the code. I.e. it is saying there is a directory `~/fbcp-ili9341/build/fbcp-ili9341/` that exists (which should not be there).  Try deleting the `build` directory and re-creating it and reissuing CMake.

C112358, 2019-01-10:
Hello,
Just tried the above steps. I now have a successfully built target at fbcp-ili9341! Unfortinately, I have encountered another issue. Now, the next step; "sudo ./fbcp-ili9341", is giving me trouble. After running this, I get "all intialized, now running main loop..." and it continues this seemingly indefinitely (after waiting 30 minutes to no avail). There are no errors thrown before this. Thanks again!

juj, 2019-01-12:
Try running `cmake -DSTATISTICS=0 ..` in the `build/` directory, and after that, re-issue `make -j`.

If you have one good working build, then you can also create another build directory e.g. `build2/`, and use that to perform `cmake` + `make` build. That way you save the first build and can experiment on another one.

C112358, 2019-01-17:
Great! Thanks for the timely feedback. Will have to try this later.

closed 2019-01-19 by C112358