juj/fbcp-ili9341 #58: Does not work with raspberry pi

vsmelov, 2019-01-14:
I tried to run the library with my ST7789 display and Raspberry Pi 3 B+

I connected display to the board this way
```
GND -> Ground
VCC -> 3v3 power
SCL -> BCM11
SDA -> BCM10
RES -> BCM4
DC -> BCM3
BLK -> BCM2
```

I configured lib with

```
cmake -DST7789=ON -DGPIO_TFT_DATA_CONTROL=3 -DGPIO_TFT_RESET_PIN=4 -DGPIO_TFT_BACKLIGHT=2 -DSPI_BUS_CLOCK_DIVISOR=6 ..
```

and build/run

```
make -j
sudo ./fbcp-ili9341
```

But saw just a black screen.

I tried to use other pins

```
cmake -DST7789=ON -DGPIO_TFT_DATA_CONTROL=27 -DGPIO_TFT_RESET_PIN=22 -DGPIO_TFT_BACKLIGHT=17 -DSPI_BUS_CLOCK_DIVISOR=6 ..
```

But without success

[image]
[image]

juj, 2019-01-14:
The command line looks good, not sure what is wrong here. There is another variant of ST7789 called ST7789VW, see https://github.com/juj/fbcp-ili9341#if-you-wired-the-display-to-the-pi-yourself , that you could try. Also you could try a higher SPI_BUS_CLOCK_DIVISOR setting, and disabling DMA, to see if either of those would have an effect.

facelessloser, 2019-01-14:
I was having this trouble, The screen wont work on the breakout board that it comes on. You need to be able to control the C/S pin. I made a new breakout board that you can order from oshpark [here](https://oshpark.com/shared_projects/OWRdBmLR) You will need a hot air solder station to de-solder the screen from the old breakout board

Hope this helps

Diehard4077, 2019-01-14:
im using the same driver on the screen from adafruit https://www.adafruit.com/product/3787  and im reciving the same issue not only that but i am extreamly new to creating things 

and i have tossed it in a box so many times over the past year out of shear frustration it is my first project with spi (without having a wiring guide and step by step) 

i may have wired it wrong (i used eagle cad and the snake eyes bonnet to visualy trace the pins to the locations on the zero) i may have screwed up the code (forgot to take a picture) but the screen stays black aswell

[image]
[image]

facelessloser, 2019-01-14:
> im using the same driver on the screen from adafruit https://www.adafruit.com/product/3787 and im reciving the same issue not only that but i am extreamly new to creating things
>
> and i have tossed [...]

Ive seen these screens but not used one. One thing that stuck out was that the BL (back light) pin isn't connected to anything, Normally on these board the BL pin has a transistor thats controlled by one of the pi's pins. This could be your problem

Hope this helps

vsmelov, 2019-01-16:
Why does manufacturer supply these screens without C/S pin? I believe it is possible to run it without that one. May be I can change your library code somehow. 

I connected BL, I can see display highlights but no pixel is active so it's black)

vsmelov, 2019-01-16:
This crazy archive I received from the manufacturer when asked him to share workable example with me :-) 
https://wetransfer.com/downloads/a85e6317dc2b004be1292d1af4e49d1c20190115060936/cd931e471898cd06625b9ddeba929ef920190115060936/19261e

vsmelov, 2019-01-16:
And they also sent me sitronix manual for st7789 :-)

juj, 2019-01-16:
Some displays do not implement a C/S line, which means that the SPI slave device on the display does not have a support of being programmatically turned on or off, but instead it will continuously listen to data transferred on the bus. This is the same as wiring the C/S pin to ground on displays that do have a C/S pin, to make it behave "always on".

If a display does not have a C/S pin, fbcp-ili9341 does not need to know or be configured any differently. It will keep toggling the BCM8 pin (CS0 line) by default, but if that line is not even connected, that will then not matter. (If you'd like to reuse BCM8 pin for something else, then may want to disable it, but that's probably another matter)

The driver support for ST7789 was written against that Adafruit display (https://www.adafruit.com/product/3787), so it has been successfully used at least once in the past.

You could try using another driver for the display to cross-reference and rule out if it's software or hardware, and if it turns out to be software, find the git commit in history in time when ST7789 support was originally added, to see if something may have regressed.

Diehard4077, 2019-02-24:
according to the commit history c/s pin is needed for the display im using i think i now have it wired right but i am still having issues i moved it to a new issue section with more detail i home u dont mind juj

juj, 2019-07-23:
Closing stale issue.

closed 2019-07-23 by juj