juj/fbcp-ili9341 #59: Two TFT screens

mpuzar, 2019-01-15:
First of all, thanks for this wonderful project, it worked right out of the box (Pi Zero W)!

I was wondering, would it be possible (and if so, how) to connect two screens, sharing the same image?

juj, 2019-01-16:
It is not possible with current code to drive two displays that are different, or two of the same displays that need to have different content displayed on them. There is no hard limitation for this though to say it could not be done, so you may be able to improve the driver to support this kind of scenario by implementing sharing of the SPI bus across two devices into the driver.

If the two displays are identical and the image to be shown on them is also identical, you may be able to solve it on hardware wiring level by connecting both displays to the same MOSI, SCLK and CS pins, so that they receive the same signal from the Pi. That will practically do a hardware level mirroring of the output, and will have the benefit of keeping full bandwidth (as opposed to having to cut bandwidth to half for each display).

mpuzar, 2019-01-16:
My scenario is indeed with identical image on both of them, so basically it might just work by connecting them in parallel? I wasn't sure whether the displays are also sending signals back which might make it problematic.

juj, 2019-01-16:
The displays are sending signals back, which occurs on the MISO line, but fbcp-ili9341 does not actually care about that line, and it never needs to read back any bytes from the displays. So practically fbcp-ili9341 does not even know if it is connected to any display or not, but the communication is one-way. The MOSI, SCLK and CS pins are one-way driven by the Pi as a SPI master, so multiple displays connected to them in parallel should be ok (modulo possible signal quality issues that may e.g. require transmitting at a lower max speed or similar)

mpuzar, 2019-01-16:
Ok, thanks for the clarification, I'll try it.

closed 2019-01-16 by mpuzar

mpuzar, 2019-02-07:
Just to confirm - it works like a charm!

juj, 2019-02-07:
Great to hear!