juj/fbcp-ili9341 #60: screen reversed when run with ssd1351 1.5inch oled

xlla, 2019-01-23:
I have compiled this library with my SSD1351 oled screen, it has 128x128 resolution, so I modify the macro DISPLAY_NATIVE_HEIGHT, and change this command according to my vendor's sample code
SPI_TRANSFER(0xCA/*Set Multiplex Ratio*/, 95); 

it was work, but whole screen is reversed, it's not rotate ,  if I watch it through a mirror, then every chars is correct.

I have read over the readme.md and not found any parameter to help with this problem.
[image]

juj, 2019-01-23:
That's interesting. The memory addressing mode is not matching up it seems like.

Try editing this line

https://github.com/juj/fbcp-ili9341/blob/ccebad5362623c435dcb5be11cc891f246ec5c50/ssd1351.cpp#L36

from `0x34` to `0x24` (so that it removes the `0x10` vertical swap addressing mode). Does that work?

juj, 2019-01-23:
According to [SSD1351 spec sheet, page 39](https://www.newhavendisplay.com/app_notes/SSD1351.pdf), the "10.1.5 Set Re-map & Dual COM Line Mode (A0h)" command also has a horizontal swap addressing mode in bit index 1 (hex `0x02`). So instead of `0x34`, you can also try value `0x36` to swap horizontally instead of vertically.

xlla, 2019-01-24:
I am glad to got your help quickly.
Before see your replay, I think it was caused by DISPLAY_FLIP_ORIENTATION_IN_SOFTWARE, so I pass -DSINGLE_CORE_BOARD=ON and recompiled, it work.

but when I try play a video on it, it was glitch and flicker;
I am running on raspberry pi 3B+, maybe SINGLE_CORE_BOARD does not suit it,
then I comment DISPLAY_FLIP_ORIENTATION_IN_SOFTWARE simple, it works but in landscape mode, see picture below. it is smooth when play video, but the statistics number just 10p. 

I will try your advice later.
[image]

xlla, 2019-01-27:
I finally comment DISPLAY_OUTPUT_LANDSCAPE, neither CPU nor chip SSD1351 will need to flip picture, so this can improve performance, is it right?

juj, 2019-01-27:
The performance difference of rotating the display is managed by

https://github.com/juj/fbcp-ili9341/blob/ccebad5362623c435dcb5be11cc891f246ec5c50/config.h#L180-L192

If DISPLAY_FLIP_ORIENTATION_IN_SOFTWARE is set, then CPU flips the screen with performance impact to CPU.

If DISPLAY_FLIP_ORIENTATION_IN_SOFTWARE is not set, then SSD1351 flips the screen. This does not have a performance impact to either SSD1351 or the CPU, but it is free.

xlla, 2019-01-28:
all right, I got it, thanks!

closed 2019-01-28 by xlla