juj/fbcp-ili9341 #63: Have someone tested with waveshare 3.5 (C)?

wolfallein, 2019-02-05:
I'm looking to buy a new LCD for my raspberry zero, and I would like to use a 3.5 inches screen. Waveshare has a ["High speed SPI 125MHz" 3.5 screen 320x480 screen](https://www.waveshare.com/wiki/3.5inch_RPi_LCD_(C)) but don't specify the LCD controller utilized, neither how much FPS we can get.

Have someone tried to use this screen?

juj, 2019-02-05:
I emailed WaveShare directly about this display December last year, asking them what controller that display uses, and where the data sheet of the display could be found. They replied back asking about my order, and when I said I have not purchased one, they said back that there does not exist a data sheet for the display, and would not tell me what controller it uses. (I doubt they would even if I had bought one)

They do supply a precompiled binary driver for the display, so it should be possible to reverse engineer it from there, but I kind of lost interest after such stereotypically secretive business practice from WaveShare's behalf. Seems to be the norm in Asian markets, perhaps due to copying being such a norm there, but that was quite silly.

I have only ever come across the MZ61581 display controller that supports 125MHz speed (the Tontec display in Readme). That one was possible to be clocked up to 140MHz. No data sheet for the MZ61581 either. The string "MZ61581" is something that I saw written out in an eBay listing of the display iirc, no markings on the display itself. It is possible that WaveShare uses the same controller (or could also be something else altogether)

If someone contributes support, would be happy to merge it in. fbcp-ili9341 on a Pi 3B is able to push pixels at about 85% bandwidth saturation if running a display at 140MHz, so it should definitely be able to feed the WaveShare display fast enough to get good refresh rates. (unless it turns out that the display protocol is gimped somehow compared to the MZ61581)

wolfallein, 2019-02-06:
Thanks for your repply.

I ended up ordering one. First I will try with their driver and see how it works, later I'm going to use the fbcp-ili9341 using the MZ61581 driver and see if works. I suspect that they used this controller as well.

I had one SPI LCD in the past but never worked well, the low FPS make it unusable for almost everything, that time I couldn't even use the fbcp. I bought a 3.5 HDMI LCD thinking that it would solve all my problems, but now that I want to make a battery powered device I found that this LCD is not power efficient the old integrated circuit and not power optimized design on the board, besides that our raspberry is not designed to run with batteries (lack of deep sleep support) ...

The fbcp-ili9341 project gave me a light in the tunnel for my project :)

It is quite possible that they used fbcp-ili9341 code on their binary, as they didn't have it before.

Agree with you about these companies. They copy everything, but nobody can copy them... it is "piracy" against the pirate.

BR

closed 2019-02-06 by wolfallein

wolfallein, 2019-02-08:
Hi, just to update

The screen arrived and I tested with their driver. I didn't try to measure the fps but apparently was good. I used `lsmod` and I saw that the module for the ILI9486 was loaded, the same used in the 35b (B) version, I looked in the pin configuration and the pins are the same.

So, I just compiled with the option `-DWAVESHARE35B_ILI9486=ON`, and worked really nice. Then I tried to decrease the clock divisor to the minimum possible, and I could get `-DSPI_BUS_CLOCK_DIVISOR=2` without any problems :)

I'm using a raspberry Pi zero W verion, and the SPI speed is 200 MHz, look at the picture below:
 
[image]

The black sky in the game is a problem of the emulator....

I haven't try to optimize anything else.

juj, 2019-02-08:
Just wow! This looks like a really efficient display! I am curious what is the maximum mbps  upload number you are able to see on the stats display? How close to 200mbps can it reach?

wolfallein, 2019-02-08:
I can see 70mbps in SNES emulation, but maybe the limitation here is the CPU. Can you suggest me a game or application that works smooth at higher fps on the Rpi zero to test?

juj, 2019-02-08:
To maximize bandwidth usage, you can disable pixel diffing altogether with the UPDATE_FRAMES_WITHOUT_DIFFING option. That should stress against max bandwidth cap. Alternatively, playing a game with full screen scrolling or watching a video might do the same trick.

wolfallein, 2019-02-09:
ok.. I tried Sonic with picodrive, I can see 145mbps. No alterations in the compilation.

[image]

wolfallein, 2019-02-09:
I tried the modifications that you suggested. I have now a fixed upload of ~145mbps, with less variance in fps. Fantastic! Looks like waveshare is fulfilling the promise of 125 MHZ SPI.

[image]

juj, 2019-02-09:
Lovely.. 480*320*60*16 = 147456000 = 147mbps. Cool stuff!