juj/fbcp-ili9341 #64: cmake error

C112358, 2019-02-09:
Hello,
After running

"cmake -DCMAKE_BUILD_TYPE=Release -DSPI_BUS_CLOCK_DIVISOR=8 -DADAFRUIT_ILI9341_PITFT=ON .." 

I am given:

-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
CMake Error in CMakeLists.txt:
  No CMAKE_C_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.

CMake Error in CMakeLists.txt:
  No CMAKE_CXX_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

-- Configuring incomplete, errors occurred!
See also "/home/osmc/fbcp-ili9341/build/CMakeFiles/CMakeOutput.log".
See also "/home/osmc/fbcp-ili9341/build/CMakeFiles/CMakeError.log".
osmc@osmc:~/fbcp-ili9341/build$

What does this mean, and how can it be fixed? Thank you!

juj, 2019-02-09:
The error here is that there is no C/C++ compiler installed on the system you are running on. You will need to install one, e.g. with apt-get or whatever package manager your distro is using? (`sudo apt-get install c++` or `g++` or `gcc` or something along those lines) I presume this is not Raspbian?

C112358, 2019-02-10:
Hi,
Thanks, juj! That worked and let me run the cmake correctly. After this, "make -j" points out a missing file (bcm_host.h) that causes a fatal error. How can I source this file with an install of OSMC? Thank you again for the great work with ili9341 displays. I greatly look forward to using this screen in my project.

juj, 2019-02-10:
I am not familiar with OSMC enough to say. See this thread about bcm_host.h not existing in the Linux distribution: https://github.com/juj/fbcp-ili9341/issues/43

If OSMC does not use DispmanX, then it will be hard to get fbcp-ili9341 working there. If it does use DispmanX, but just does not supply the development headers, then you can try compiling fbcp-ili9341 on e.g. a Raspberry Pi 3B, and copy the resulting binary fbcp-ili9341 executable over to OSMC. That is how I was able to get fbcp-ili9341 run on Kodi some months back when testing.

C112358, 2019-02-10:
I'm back,
OSMC does have DispmanX. I was able to add the "include" folder, containing "bcm_host.h" to OSMC, which allowed "make -j" to run. After this "sudo ./fbcp-ili9341" does initialize the display, but the main loop never ends.

juj, 2019-02-11:
The main loop is where the driver should be running and outputting an image to the display. If the display is not reacting, something is wrong. Try debugging through different  options and double check the configuration. It could also be some kind of incompatibility with OSMC.

C112358, 2019-02-11:
After running "sudo ./fbcp-ili9341", the display does work and ends up being fully usable within OSMC. Despite this, the command never finishes running from the view of SSH. As a result, I am unable to apply necessary changes to rc.local. I have also tried to alter the "DSPI_BUS_CLOCK_DIVISOR" to lower as well as higher values. Are there any other potential solutions?

juj, 2019-02-11:
Oh, you mean it does not give back control to the terminal window? Try starting it in background mode with

sudo ./fbcp-ili9341 &

The & sign at the end makes the process run in background

C112358, 2019-02-11:
Worked like a charm! Assuming the development headers are manually added, the fbcp-ili9341 driver works perfectly. Thanks again!

DRAgon734465502, 2019-02-12:
hello, I get the other cmake error,

pi@raspberrypi:~/fbcp-ili9341/build $ sudo cmake  -DILI9341=ON ..
CMake Error at CMakeLists.txt:130 (message):
  Please define -DSPI_BUS_CLOCK_DIVISOR=<some even number> on the CMake
  command line! (see files ili9341.h/waveshare35b.h for details) This
  parameter along with core_freq=xxx in /boot/config.txt defines the SPI
  display speed.  Smaller divisor number=faster speed, higher number=slower.
-- Configuring incomplete, errors occurred!
See also "/home/pi/fbcp-ili9341/build/CMakeFiles/CMakeOutput.log".

DRAgon734465502, 2019-02-12:
please tell me what's  wrong

C112358, 2019-02-12:
Hi DRAgon,
You need to define the DSPI Bus Clock Divisor. With that in mind, acceptable numbers differ from display to display; finding the right value is trial and error. Just type "-DSPI_BUS_CLOCK_DIVISOR=X" where "x" is an even number. As stated in the output above, lower numbers allow the display to run faster.

DRAgon734465502, 2019-02-12:
hello, I have tried that.  Now no error .

But just HDMI disply , the spi LCD can not enter the os desktop.  just disply to stop at  " _ "

DRAgon734465502, 2019-02-12:
sudo /home/pi/fbcp-ili9341/build/fbcp-ili9341
bcm_host_get_peripheral_address: 0x3f000000, bcm_host_get_peripheral_size: 16777216, bcm_host_get_sdram_address: 0xc0000000
BCM core speed: current: 250000000hz, max turbo: 250000000hz. SPI CDIV: 6, SPI max frequency: 41666667hz
Allocated DMA channel 7
Allocated DMA channel 1
Enabling DMA channels Tx:7 and Rx:1
DMA hardware register file is at ptr: 0x75b9d000, using DMA TX channel: 7 and DMA RX channel: 1
DMA hardware TX channel register file is at ptr: 0x75b9d700, DMA RX channel register file is at ptr: 0x75b9d100
Resetting DMA channels for use
DMA all set up
Initializing display
Creating SPI task thread
InitSPI done
DISPLAY_FLIP_ORIENTATION_IN_SOFTWARE: Swapping width/height to update display in portrait mode to minimize tearing.
Relevant source display area size with overscan cropped away: 320x480.
Source GPU display is 320x480. Output SPI display is 240x320 with a drawable area of 240x320. Applying scaling factor horiz=0.67x & vert=0.67x, xOffset: 13, yOffset: 0, scaledWidth: 213, scaledHeight: 320
Creating dispmanX resource of size 213x320 (aspect ratio=0.665625).
GPU grab rectangle is offset x=0,y=0, size w=213xh=320, aspect ratio=0.665625
All initialized, now running main loop...

DRAgon734465502, 2019-02-12:
then the LCD go to show "_"  and can not go to the os desktop

juj, 2019-02-12:
What do you mean by `"LCD go to show "_"`? Can you take a photo to illustrate?

DRAgon734465502, 2019-02-13:
the command seems do not to work
 sudo /home/pi/fbcp-ili9341/build/fbcp-ili9341
http://ae01.alicdn.com/kf/HTB1Fx0.bcfrK1RkSmLyq6xGApXau.jpg

juj, 2019-02-13:
The good bit is that it looks like fbcp-ili9341 is working properly, and it is outputting first TTY output to the SPI display.

That TTY seems to have a command line terminal active on it, without an active login to Linux, so it is showing an empty caret. You may be able to physically connect keyboard and mouse to the Pi, and then switch to TTY1, see https://unix.stackexchange.com/questions/167386/how-to-switch-between-tty-and-xorg-session , perhaps `Ctrl-Alt-F1` or similar? And then you may be able to log in in that TTY, and the console text would show up on the screen.

Most likely X is rendering to another TTY output. You'll need to configure the X rendering output to occur to the first TTY, I think. See this section, https://github.com/juj/fbcp-ili9341#does-fbcp-ili9341-work-with-linux-command-line-terminal-or-x-windowing-system , perhaps that can be of help.

TTY configuration is not super-familiar to me in Linux, so I am not 100% sure if the above is correct, but that is a bit of guess about the scenario.

DRAgon734465502, 2019-02-14:
hello , I have  tried Ctrl-Alt-F1 or F2  ,   which do not work

this is my config.txt add code
dtparam=spi=on
hdmi_group=2
hdmi_mode=87
hdmi_cvt=480 320 60 1 0 0 0
hdmi_force_hotplug=1

this my /etc/rc.lcaol add code
sudo modprobe fbtft_device name=flexfb gpios=dc:22,reset:27 speed=32000000
sudo modprobe flexfb width=320 height=240 buswidth=8 init=-1,0xCB,0x39,0x2C,0x00,0x34,0x02,-1,0xCF,0x00,0XC1,0X30,-1,0xE8,0x85,0x00,0x78,-1,0xEA,0x00,0x00,-1,0xED,0x64,0x03,0X12,0X81,-1,0xF7,0x20,-1,0xC0,0x23,-1,0xC1,0x10,-1,0xC5,0x3e,0x28,-1,0xC7,0x86,-1,0x36,0x28,-1,0x3A,0x55,-1,0xB1,0x00,0x18,-1,0xB6,0x08,0x82,0x27,-1,0xF2,0x00,-1,0x26,0x01,-1,0xE0,0x0F,0x31,0x2B,0x0C,0x0E,0x08,0x4E,0xF1,0x37,0x07,0x10,0x03,0x0E,0x09,0x00,-1,0XE1,0x00,0x0E,0x14,0x03,0x11,0x07,0x31,0xC1,0x48,0x08,0x0F,0x0C,0x31,0x36,0x0F,-1,0x11,-2,120,-1,0x29,-1,0x2c,-3

con2fbmap 1 0
sleep 7
FRAMEBUFFER=/dev/fb0 startx
sudo /home/pi/fbcp-ili9341/build/fbcp-ili9341 &

DRAgon734465502, 2019-02-14:
this my/boot/cmdline.txt  code
dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait fbcon=map:10 fbcon=font:ProFont6x11 logo.nologo

wolfallein, 2019-02-14:
I think you shouldn't have any framebuffer, and SPI driver enabled, as it is on the installation instructions in the main page, before thecompilation process:

>Installation
>Check the following sections to set up the driver.
>Boot configuration
> This driver does not utilize the notro/fbtft framebuffer driver, so that needs to be disabled if active.  That is, if your /boot/config.txt file has lines that look something like dtoverlay=pitft28r, ..., dtoverlay=waveshare32b, ... or dtoverlay=flexfb, ..., those should be removed.

> This program neither utilizes the default SPI driver, so a line such as dtparam=spi=on in /boot/config.txt should also be removed so that it will not cause conflicts.

> Likewise, if you have any touch controller related dtoverlays active, such as dtoverlay=ads7846,... or anything that has a penirq= directive, those should be removed as well to avoid conflicts. It would be possible to add touch support to fbcp-ili9341 if someone wants to take a stab at it.

So,.. I would try removing the lines:

dtparam=spi=on

from your /boot/config.txt

and:

sudo modprobe fbtft_device name=flexfb gpios=dc:22,reset:27 speed=32000000
sudo modprobe flexfb width=320 height=240 buswidth=8 init=-1,0xCB,0x39,0x2C,0x00,0x34,0x02,-1,0xCF,0x00,0XC1,0X30,-1,0xE8,0x85,0x00,0x78,-1,0xEA,0x00,0x00,-1,0xED,0x64,0x03,0X12,0X81,-1,0xF7,0x20,-1,0xC0,0x23,-1,0xC1,0x10,-1,0xC5,0x3e,0x28,-1,0xC7,0x86,-1,0x36,0x28,-1,0x3A,0x55,-1,0xB1,0x00,0x18,-1,0xB6,0x08,0x82,0x27,-1,0xF2,0x00,-1,0x26,0x01,-1,0xE0,0x0F,0x31,0x2B,0x0C,0x0E,0x08,0x4E,0xF1,0x37,0x07,0x10,0x03,0x0E,0x09,0x00,-1,0XE1,0x00,0x0E,0x14,0x03,0x11,0x07,0x31,0xC1,0x48,0x08,0x0F,0x0C,0x31,0x36,0x0F,-1,0x11,-2,120,-1,0x29,-1,0x2c,-3

con2fbmap 1 0
sleep 7
FRAMEBUFFER=/dev/fb0 startx

from your /etc/rc.local

keeping the: sudo /home/pi/fbcp-ili9341/build/fbcp-ili9341 &

make a backup copy of your files if you want to use later. 

but I would try without to make sure that it will work, this line is only for starting the driver during the boot process, fbcp-ili9341 doesn't need any SPI/fbcpdriver loaded to work.

DRAgon734465502, 2019-02-18:
I use waveshare 3.2inch LCD

if I do as your methion,  remove my code form  /boot/config.txt  and /etc/rc.local

The LCD can not be driver .  the LCD is always keep black

wolfallein, 2019-02-18:
When you ran the comand:

`sudo /home/pi/fbcp-ili9341/build/fbcp-ili9341`

what you have? Also, can you run the command `lsmod`, and post what you get?

DRAgon734465502, 2019-02-24:
hello  , I have find my error . it is caused  by my  error  cmake code. I lost some parameter

Now it can display,  but come new problem. 
1. it can not display at the top and bottom of the LCD
2. the  parameter  display at the top of the os desktop  (I have see your introduction  , can change config.h to remove the parameter, but the parameter is so much and my english is bad, and is hard to find  which parameter I should to remove at the config.h)

see picture
http://ae01.alicdn.com/kf/HTB1oEdjbELrK1Rjy0Fjq6zYXFXaw.jpg

please  help me

juj, 2019-02-24:
>     1. it can not display at the top and bottom of the LCD

This is due to Raspberry Pi desktop resolution not matching the display resolution. Try setting in /boot/config.txt the display resolution to 320x240, 480x320, 640x480, or some other size that is in the aspect ratio of the display (4:3).

>     2\. the  parameter  display at the top of the os desktop  (I have see your introduction  , can change config.h to remove the parameter, but the parameter is so much and my english is bad, and is hard to find  which parameter I should to remove at the config.h)

See https://github.com/juj/fbcp-ili9341/#statistics-overlay , the option is `-DSTATISTICS=0`.

DRAgon734465502, 2019-02-25:
thank you ,  all OK 

can you update the software to use touchsreen ?

Do you have the plan ?

juj, 2019-07-23:
Closing resolved issues. Kpisphere has been working on touch support recently, available at https://github.com/kpishere/fbcp-ili9341.

closed 2019-07-23 by juj