juj/fbcp-ili9341 #65: Open Loop Stuck

bassmatt23, 2019-02-10:
Hello,

I am a beginner with all of this stuff, and I am currently trying to create a retro pie system with a 3.5 inch RPi LCD (B) WaveShare SpotPear. After seeing your video on youtube, I I thought your driver would be great to use because I already had the same model LCD you were using. I followed all the installation until I got to the "sudo ./fbcp-ili9341" step. The command executed and then got stuck at a part that said "all intialized, now running main loop..." I have seen several other people post about this, but no solution. 

Any ideas?

Thank You

juj, 2019-02-11:
The main loop is where the driver should be outputting an image to the display. If nothing is showing up, you will need to troubleshoot it through. The way that fbcp-ili9341 communicates with the display is unidirectional, so it is unable to know if there is a display connected back even, so double checking the wiring is also good to do.

C112358, 2019-02-11:
Just had a possibly similar issue resolved. Some of the later points may apply to the issue you've experienced. 

https://github.com/juj/fbcp-ili9341/issues/64

bassmatt23, 2019-02-11:
Ok, I will play around with it when I Hager’s home today and see what I can do.

Thanks!

bassmatt23, 2019-02-11:
“Sudo ./fbcp-ili9341 &” did not work. How do I go about troubleshooting my issue?

bassmatt23, 2019-02-11:
Ok, I raised the DSPI_BUS_CLOCK_DIVISOR to 20. Now the display shows what’s on my TV while the main loop command is running, but it is still stuck. When I use control C, the screen goes white.

juj, 2019-02-11:
Double check the ampersand directive and spaces, i.e.

`sudo ./fbcp-ili9341 &`
that is,
`sudo<space>./fbcp-ili9341<space>&`, with no spaces after `&`

For more info, see https://kb.iu.edu/d/afnz

If that is not working (perhaps some different shell than sh/bash?), then another method on Linux is to start the program with

`sudo ./fbcp-ili9341`
then tap Ctrl-Z, this suspends the program to background, and returns to terminal,
then type the command `bg` to resume the program to run on the background.

Check out https://www.thegeekdiary.com/understanding-the-job-control-commands-in-linux-bg-fg-and-ctrlz/

bassmatt23, 2019-02-11:
Well that seemed to work, but it still won’t display after a reboot. I typed “sudo /home/pi/fbcp-ili9341/build/fbcp-ili9341 &” in /etc/rc.local but the display won’t turn upon a reboot.

juj, 2019-02-12:
Hmm, hard to say here. Is this on Raspbian or some other OS? Some ideas that come to mind:
 - Perhaps /etc/rc.local is not getting executed at launch on your system, if it's not Raspbian
 - Or possibly /etc/rc.local requires that Pi is set to auto-boot to console or desktop. Not quite sure about this one, I think I have all my Pis set to auto-login without asking for a password. If this is the case, there may be some other better place to put the initialization.
 - Perhaps there is something in /etc/rc.local file before fbcp-ili9341 launch that makes the script abort in the middle. You can try adding a line `echo Current time is $(date) > /home/pi/started.txt` right after the fbcp-ili9341 initialization line, and then reboot, and check the contents of the file to verify that it got executed (then you'll know fbcp-ili9341 command also got executed)
 - Perhaps there is some other driver initializing at startup time that conflicts, which is why only initializing fbcp-ili9341 later works

Try checking that fbcp-ili9341 is running after the boot, via `ps`, `ps -A`, `top` or `htop` command.

bassmatt23, 2019-02-12:
It is on Retropi, not raspbian. I will try a couple of these out

Thanks!

VeteranGamer, 2019-02-16:
Hi...
Im also trying to enable this on a RPi3 running Retropie
its a Generic (bare LCD) ILI9341

heres how I had cmake

-DSPI_BUS_CLOCK_DIVISOR=6 -DILI9341=ON -DARMV8A=ON -DGPIO_TFT_DATA_CONTROL=24 -DGPIO_TFT_RESET_PIN=25 -DSTATISTICS=0 ..

I also tried...

sudo ./fbcp-ili9341

and 

sudo ./fbcp-ili9341 &

but hit a brick wall at the loop

please advise

Thanks..

.

bassmatt23, 2019-03-02:
Since I am utilizing this driver on RetroPi, where should I add the line of code to run the driver at launch

VeteranGamer, 2019-03-03:
> 
> 
> Since I am utilizing this driver on RetroPi, where should I add the line of code to run the driver at launch

I've managed to get this up and running on RetroPie (finally)....

if you have a look here....

https://sudomod.com/forum/viewtopic.php?f=42&t=7389#p70512

I've posted how I went about it....

[image]

.

bassmatt23, 2019-03-05:
Still can’t get it to work. It still gets stuck at main loop every time and I can’t find any other way around it except CtrlZ and then “bg” . The display goes white when I run the “sudo ./fbcp-ili9341 &” and then just gets stuck at “all initialized, now running main loop...” I feel like when I do the CtrlZ then “bg” that the command still isn’t finishing. I think this is the source of the issue.

Here’s what I typed in to CMake 

cmake -DSPI_BUS_CLOCK_DIVISOR=6 -DWAVESHARE35B_ILI9486=ON -DARMV8A=ON -DGPIO_TFT_DATA_CONTROL=24 -DGPIO_TFT_RESET_PIN=25 -DSTATISTICS=0 -DDISPLAY_ROTATE_180_DEGREES=ON ..

I’m using waveshare spotpear 3.5 in LCD (B)

Bunanaben, 2019-03-05:
Hey man.I just had a look on that problems you listed above,you said that you were successfully lighted up the screen,but just can't use it when you reboot your Pi,I recommended you have a check on your file named rc.local to make sure you get the right path to your fbcp-ili9341 ( complied ) folder,such as my path is /home/porkwotonlee/fbcp-ili9431/build/...etc.and another problem is your screen have a strange display when you use your mouse,you can try to set a higher value of BUS CLOCK DIVISOR,mine is 24，for more safer,I set it to 30 at first,and slowly reduce the value down to  24，22Will have broken screen。。。I'm a Chinese,please ignore my 'not so good' English.

bassmatt23, 2019-03-06:
Ok- will do

Thanks!

juj, 2019-07-23:
Closing old issues.

closed 2019-07-23 by juj