juj/fbcp-ili9341 #66: Screen going white after some time

mpuzar, 2019-02-16:
After running nicely for a while (sometimes a minute, sometimes many, the screen goes white and stays like that (sometimes it never initializes properly, so it just remains white on boot). Restarting fbcp-ili9341 tends to help but not always.

At first I thought it might be some bad contact/soldering, which I tried to do heavy debugging on without finding any problems, but as I see it happening with different combinations of pi0s and screens, I'm beginning to wonder whether it might be a software issue. Any suggestions on how to approach the problem to determine what is in fact happening? Any parameter I should try playing with when compiling fbcp-ili9341?

Of course, it still might be some bad contact that's causing it, but at this moment I'd like to start investigating other possibilities as well.

juj, 2019-02-16:
It could certainly be a software bug, as synchronization of Data/Control pin and MOSI pin is really tight in the driver. Some ideas to try:
 - use a slower CDIV setting (perhaps considerably slower to ensure safest speed/signal limits)
 - try disabling DMA - it uses a completely different kind of synchronization scheme than no-DMA does
 - add slowdowns around code that clears/sets the Data/Control pin: https://github.com/juj/fbcp-ili9341/blob/ccebad5362623c435dcb5be11cc891f246ec5c50/spi.cpp#L300 (assuming this is running in 4-wire SPI mode? 3-wire SPI doesn't use the D/C pin)
 - try another type of display, in case it might be display controller code specific. I have one ILI9486L that was donated to me, that is observed to occassionally go white after running for a while. But that display also has hardware damage to its connector and a quarter of the screen pixels are corrupted due to bad contacts, so I have excluded that from consideration. On other display controllers, I have not run into issues. ILI9341 and MZ61581 are the displays that I run regularly, other display controllers have not been tested that  much beyond the first implementation and some minutes of use to superficially verify it works.

juj, 2019-07-23:
Closing old issues.

closed 2019-07-23 by juj