juj/fbcp-ili9341 #67: Blinking prompt overlays image after copying to framebuffer

freerko, 2019-02-21:
Hi,

I am building a camera with a Raspberry Pi Zero with a PiTFT 2.2" 320x240 (pitft22) as a display. The picamera FAQ [claims that the preview system doesn't support PiTFT or other GPIO/SPI displays](https://picamera.readthedocs.io/en/latest/faq.html#the-preview-doesn-t-work-on-my-pitft-screen) , but thanks to fbcp-ili9341 the picamera.start_preview() works just fine!

But if I copy a still image via `convert -resize 320x240 -background black -gravity center -extent 320x240 '+basedir+'/'+img+' bgra:/dev/fb0` to the framebuffer the image is shown with a green blinking shell prompt in the button left corner. Is there a way to prevent that the image is overwritten with the blinking prompt character? Can I somehow set a priority of the display overlays? Or do I have to disable the prompt blinking in Raspbian? I'd like to keep the possibility to exit the python script and using the shell with the PiTFT display.

juj, 2019-02-22:
I suspect what happens there is that the `convert` command copies the image to fb0, but then since there is no program to persist an overlay element of the image on the DispmanX graphics driver processing stack, the command line prompt will then overwrite it with whatever contents is rendered next.

Using `convert` command (repeatedly?) to show a video feed on the Pi is probably not efficient, but will be very CPU consuming.

If you have a static image, or a video feed on the CPU side, then a more proper solution may be to create a C program that uses the DispmanX API to create a DispmanX overlay (a rectangular image area on the GPU), and then repeatedly submit new frames to it from C code.

Raspberry Pi userland repository has a small example of how to do this: https://github.com/raspberrypi/userland/blob/master/host_applications/linux/apps/hello_pi/hello_dispmanx/dispmanx.c

The benefit of that approach is that it will avoid high CPU overhead, but also that the GPU will then persistently keep track that there is supposed to be a image on the screen (until the app shuts down)

Another approach, that will be even faster would be to rewrite fbcp-ili9341 to decouple itself from DispmanX altogether, and instead make it a CPU side library for pushing pixels to a display. The #define RANDOM_TEST_PATTERN at https://github.com/juj/fbcp-ili9341/blob/ccebad5362623c435dcb5be11cc891f246ec5c50/gpu.cpp#L21 shows a bit of direction for this kind of approach. (this kind of development will need more advanced C programming)

freerko, 2019-02-23:
Thank you very much for the explanation! I didn't think it would be that complicated. I just disabled the blinking prompt with the line
`@reboot echo -e '\033[?17;0;0c' > /dev/tty1`
in the root crontab.
That workaround works fine for me :)

juj, 2019-07-23:
Closing as resolved.

closed 2019-07-23 by juj