juj/fbcp-ili9341 #68: Added basic touchscreen support for WaveShare 35B controller

dmitryponv, 2019-02-21:
Added a hack to WaveShare35b 480x320 display touchscreen XPT2046 
Reads touchscreen 60 times a second and dumps coordinates to /tmp/TCfifo
All credit to Original XPT2046 library by Markus Sattler https://github.com/Links2004/XPT2046 under GNU Lesser General Public License
For simplicity, #define USE_SPI_THREAD is commented out, and touchscreen is read synchronously before each frame
The hack involves changing SPI0 CLK and CS0 and CS1 pins temporarily and changing it back after reading TC. 
Datasheet https://www.raspberrypi.org/app/uploads/2012/02/BCM2835-ARM-Peripherals.pdf on page 148

Here is the Forked repository for review https://github.com/dmitryponv/fbcp-ili9341

F.Y.I. I spent more time on figuring out how to use github and contribute to projects with limited information for beginners. I don't fully understand the process by which contributions are reviewed. Please advise if I need to make any additional steps

kpishere, 2019-03-13:
I gave it a try in my branch.  Didn't go well.  screen went white as soon as you access tmp file.  I don't get how you could use SPI drive along with this code at the same time as one uses the SPI driver and the other accesses the hardware directly.

dmitryponv, 2019-03-13:
The **/tmp/TCfifo** file is just how I decided to read out the data, you can run a "cat /tmp/TCfifo" command in console, click somewhere on the screen and it will display the coordinates. 

I use SPI the same way the official drive would, I drive CS pin for display low, and CS pin for touchscreen high, then read back data from touchscreen, and revert CS pin state to normal. I also had to change SPI configuration a bit, the clock rate was too high for touchscreen. Hence why its a hack. 

If your screen is going blank when you read the file, I assume with another C++ program, the issue is in the Inter Process communication. 

Try this code, it works for me

```
#include <stdio.h> 
#include <string.h> 
#include <fcntl.h> 
#include <sys/stat.h> 
#include <sys/types.h> 
#include <unistd.h>   
int main() 
{ 
	int fd;   
	// FIFO file path 
	char * myfifo = "/tmp/TCfifo";   
	// Creating the named file(FIFO) 
	// mkfifo(<pathname>, <permission>) 
	mkfifo(myfifo, 0666);   
	char arr1[80], arr2[80]; 
	while (1) 
	{ 
  		// Open FIFO for Read only 
		fd = open(myfifo, O_RDONLY); 
  		// Read from FIFO 
		read(fd, arr1, sizeof(arr1)); 
  		// Print the read message 
		printf("Data: %s\n", arr1); 
		close(fd); 
	} 
}
```

dmitryponv, 2019-03-14:
I dont know how this happened, looks like it committed unmodified files, I will update

kpishere, 2019-05-30:
Jui, et. al., I'm so close to getting this driver working fully with touch-screen on the Kedei type devices!  Since kernel module was out of commission, I took liberty of re-writing it with the sole function of exposing interrupt.  (there are other GPIO pin drivers already but they have been changing lots from kernel to kernel .. simpler and more portable to write a new one).  I have remaining issue of needing to blank interrupt whilst sampling and maybe play with limiting sample rates so it does not interfere with refresh too much.  I have noticed screen must be undergoing redraw in order to perform touch-sampling.  Not sure how to tackle that one.  Open to suggestions on that front.

kpishere, 2019-05-31:
it now works nicely in my branch