juj/fbcp-ili9341 #69: Pi zero with adafruit 240x240 1.54" not working

Diehard4077, 2019-02-24:
The driver I have is (st7789)

I have wired this up twice and I still can not get the screen to display even with multiple different  cmake changes

nhenry1647, 2019-02-26:
Try ST7789VW.  I have this running on a 240x240 screen as well and it does
work.  Also try deleting the build folder to start a fresh compile.

Diehard4077, 2019-02-26:
I was deleting the build file each time as well as restarting and removing the power from the pi I guess I could try the VW variant

Crackedconsole, 2019-03-20:
I am working with a raspberrypi zero w as well trying to get a similar 1.54inch LCD to work with this driver.
I've tried both ST7789 and ST7789VW
https://www.aliexpress.com/item/Raspberry-Pi-Game-zero-W-2B-3B-1-54inch-mini-LCD-touchscreen/32970035492.html?spm=a2g0s.12269583.0.0.159b43fcoQ6btZ

I reached out to the manufacture and they verified the display driver IC is a ST7789 and provided a spec sheet. It showed datachannel as BCM pin 22 and RST as 27.

I've tried several fresh makes using both ST7789 and ST7789VW, with DMA transfers on and off, between 60-4 clock divisor, 2 different DMA channel settings all to no avail and just a blank screen.

THOUGH, oddly, when I set the DC pin to 27 during the make process and use the ST7789VW option, for just 1-2 frames I get a flash of the console boot when the driver is initialized.

Has anyone had success with this driver on these 1.54inch screens and a PiZero, or even specifically the device I'm working with?

Thank you

Crackedconsole, 2019-03-20:
I found this below works for my display (1.54 inch with ST7789 driver), yet last night I had the -DTS7789VW=on at the end of the string and it never worked oddly. Moving it to the front and it now works. Though I did also add explicitly "DUSE_DMA_TRANSFERS=ON"

 cmake -DST7789VW=ON -DGPIO_TFT_DATA_CONTROL=22 -DGPIO_TFT_RESET_PIN=27 -DUSE_DMA_TRANSFERS=ON -DSPI_BUS_CLOCK_DIVISOR=4 -DDISPLAY_ROTATE_180_DEGREES=ON -DSTATISTICS=0 ..

winterism, 2019-06-03:
> I found this below works for my display (1.54 inch with ST7789 driver), yet last night I had the -DTS7789VW=on at the end of the string and it never worked oddly. Moving it to the front and it now [...]

Thanks for sharing your config. 
I have more question. What is your rasbian img? Is it official or custom?

juj, 2019-07-23:
Closing old issues.

closed 2019-07-23 by juj