juj/fbcp-ili9341 #70: 3.5" Waveshare Spotpear (A) V3

VeteranGamer, 2019-02-28:
Hi,
I've got a 3.5" Waveshare Spotpear (A) V3
I've tried to enable it using
`-DILI9486L=ON`

but no success

my cmake is
`cmake -DSPI_BUS_CLOCK_DIVISOR=12 -DILI9486L=ON -DARMV8A=ON -DGPIO_TFT_DATA_CONTROL=24 -DGPIO_TFT_RESET_PIN=25 -DSTATISTICS=0 -DDISPLAY_ROTATE_180_DEGREES=ON ..`

Pi3A+

any tips on how I should proceed

Thank you

VeteranGamer, 2019-02-28:
when I do 

`cmake -DSPI_BUS_CLOCK_DIVISOR=12 -DILI9486L=ON -DARMV8A=ON -DGPIO_TFT_DATA_CONTROL=24 -DGPIO_TFT_RESET_PIN=25 -DSTATISTICS=0 -DDISPLAY_ROTATE_180_DEGREES=ON ..`

I get nothing except a black screen....

I've also tried this...

`-DARMV8A=ON -DWAVESHARE35B_ILI9486=ON -DSPI_BUS_CLOCK_DIVISOR=12 -DGPIO_TFT_RESET_PIN=25 -DGPIO_TFT_DATA_CONTROL=24 -DSTATISTICS=0 -DDISPLAY_ROTATE_180_DEGREES=ON ..`

which initially felt like it worked (as this is what was displayed)

[img]https://i.imgur.com/6mTU8qe.jpg[/img]

but as soon as you press any buttons, the screen starts to break up

[img]https://i.imgur.com/WIScipt.jpg[/img]

[img]https://i.imgur.com/kysY6QA.jpg[/img]

the screen used in the 
3.5" Waveshare Spotpear (A) V3
is this one

[img]https://i.imgur.com/ZfEb8G9.jpg[/img]

which comes up as a ili9486L

I'm sure i'm not doing something right....
just need a point in the right direction....

Thanks...

.

Bunanaben, 2019-03-03:
Thank you Bro ！ I made it on my waveshare 3.5inch LCD (A) screen(Raspbian)which similar to yours with the code you finally uploaded。During I working on this，I also found the same tearing screen likes yours，it work nice at first ，but after I set the resolution in the raspi-config from default to DMT:mod 640*480...… tearing screen……but why not use the ssh terminal to set screen parameter in the raspi-config，maybe you will make it。My English is not so good，sorry。

Bunanaben, 2019-03-04:
you can also try to set a higher value of ”-DSPI_BUS_CLOCK_DIVISOR=“，up to 30 or higher。

juj, 2019-07-23:
Sorry, missed this issue when it was raised. The corruption is likely due to too fast SPI speed, Bunanaben's suggestion should be the appropriate fix.

I'll close this as old, feel free to reopen if this still is unresolved.

closed 2019-07-23 by juj