juj/fbcp-ili9341 #75: is it possible to run this driver together with VC4 GL driver

krzysztofkuczek, 2019-03-13:
Hi,
it works really fine! Thank you.
One question, is it possible to run it when experimental hardware OpenGL driver is enabled by adding line below to config.txt
`dtoverlay=vc4-kms-v3d`

In this case i get 
`vc_dispmanx_display_open failed!`

Thank you !
Kris

juj, 2019-03-26:
Glad the driver works out for you!

Unfortunately I am not familiar with the vc4-kms-v3d driver. Googling around, it looks like popcornmix commented in https://github.com/raspberrypi/firmware/issues/556#issuecomment-191843362 that vc4-kms-v3d does not support dispmanx. Fbcp-ili9341 is based on top of dispmanx, which it uses to snapshot/acquire the rendered frame stream from the GPU.

Perhaps it may be possible to develop a corresponding rendered frame stream capture mechanism on top of the vc4-kms-v3d driver to use as a replacement for dispmanx in such scenarios.

I'll leave this entry open to highlight the unsupported configuration, in case someone knowledgeable might be able to pick this up.

Kneesnap, 2019-07-04:
I would like to advocate for full KMS support too. I'm going to try my hand at it, but I make no guarantees. I have no idea what I'm doing.

juj, 2019-08-25:
Testing on Raspberry Pi 4 with Raspbian Buster, with /boot/config.txt

```
dtparam=audio=on

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
hdmi_group=2
hdmi_mode=87
hdmi_cvt=480 320 60 1 0 0 0
hdmi_force_hotplug=1
```
and
```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.19.66-v7l+ #1253 SMP Thu Aug 15 12:02:08 BST 2019 armv7l GNU/Linux
```
I notice that works with fbcp-ili9341 well. I am not familiar with the different display drivers, in particular about what the difference is between`vc4-fkms-v3d` vs `vc4-kms-v3d`, however I wonder if this observation about `vc4-fkms-v3d` would help anything for users here?

It is notable that `hdmi_force_hotplug=1` is required for fbcp-ili9341 to work, updated the repo to add a note about that.