juj/fbcp-ili9341 #77: strange numbers in top left corner for waveshare 1.44in lcd

krzysztofkuczek, 2019-03-13:
Hi,
Everything works fine, driver compiled for pi zero board with following flags
`cmake -DWAVESHARE_ST7735S_HAT=ON -DSPI_BUS_CLOCK_DIVISOR=12 ..`
The only issue is that strange numbers/letters (maybe debug info?) id displayed in top left corner.

[image]

Thank you in advance
Kris

krzysztofkuczek, 2019-03-13:
solved by adding -DSTATISTICS=0 :)
Maybe it should be default...

closed 2019-03-13 by krzysztofkuczek

Merith-TK, 2020-09-15:
Did you ever fix the issue where the display signal is larger than the actual physical display, so fbcp scales it to the screen leaving you with a squished, unusable screen?

juj, 2020-09-20:
You can adjust the video resolution in hdmi_mode directive in /boot/config.txt. If you make it match the native resolution of the display, then there will be no scaling.