juj/fbcp-ili9341 #78: Waveshare 3.2 C

VanDraven, 2019-04-02:
Hi, first, I would like to thank you for your amazing work.
I used a Waveshare 3.5 C with a pi zero, I have a great image and I'm able to push it to 125MHz. But I have drops in framerate when I play some demanding games like Sonic on Megadrive. It seems that 480 x 360 is a little too much for such a limited CPU. So I ordered one of the new 3.2 C but no matter what I do, I can't get it to work. It stays white no matter what. Any Idea?

VanDraven, 2019-04-10:
I asked engineers at Waveshare.
3.2C is ili9341V
4C is ili9488

juj, 2019-04-18:
Hmm, not sure what would be different with WaveShare 3.2C and ILI9341V compared to the other ILI9341 displays. You could try driving it via the ILI9340 support that was contributed with #73. (One gotcha is that when changing the controller to target from command line, create a new build directory or delete the old CMakeCache.txt file in the old build directory, otherwise CMake might still pick up the old one)

juj, 2019-07-23:
Closing old issues.

closed 2019-07-23 by juj

nsilveri, 2020-01-21:
news about this display?

VanDraven, 2020-01-21:
I've tried to make it work but I wasn't successful.  I hope someone will be able to solve this.

nsilveri, 2020-01-21:
> I've tried to make it work but I wasn't successful. I hope someone will be able to solve this.

me neither, i must use official driver that are very heavy in my opinion