juj/fbcp-ili9341 #79: Building on arm64?

satmandu, 2019-04-13:
I'm building for waveshare 3.5inch LCD (A) for 3b+.

and this is what I get with compilation (after commenting out the set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} in CMakeLists.txt )

```
 cmake -DARMV8A=ON -DWAVESHARE35B_ILI9486=ON -DSPI_BUS_CLOCK_DIVISOR=30 -DGPIO_TFT_RESET_PIN=25 -DGPIO_TFT_DATA_CONTROL=24 -DSTATISTICS=0 -DDISPLAY_ROTATE_180_DEGREES=ON ..
-- Doing a Release build
-- Board revision: 
CMake Warning at CMakeLists.txt:40 (message):
  The board revision of this hardware is not known.  Please add detection to
  this board in CMakeLists.txt.  (proceeding to compile against a generic
  multicore CPU)

-- Enabling optimization flags that target ARMv8-A instruction set (Pi 2B >= rev. 1.2, 3B, 3B+, CM3 or CM3 lite)
-- Using 4-wire SPI mode of communication, with GPIO pin 24 for Data/Control line
-- Using GPIO pin 25 for Reset line
-- Scaling source image to view. If the HDMI resolution does not match the SPI display resolution, this will produce blurriness. Match the HDMI display resolution with the SPI resolution in /boot/config.txt to get crisp pixel perfect rendering, or alternatively pass -DDISPLAY_CROPPED_INSTEAD_OF_SCALING=ON to crop instead of scale if you want to view the center of the screen pixel perfect when HDMI and SPI resolutions do not match.
-- Preserving aspect ratio when scaling source image to the SPI display, introducing letterboxing/pillarboxing if HDMI and SPI aspect ratios are different (Pass -DDISPLAY_BREAK_ASPECT_RATIO_WHEN_SCALING=ON to stretch HDMI to cover full screen if you do not care about aspect ratio)
-- SPI_BUS_CLOCK_DIVISOR set to 30. Try setting this to a higher value (must be an even number) if this causes problems. Display update speed = core_freq/divisor. (on Pi3B, by default core_freq=400). A safe starting default value may be -DSPI_BUS_CLOCK_DIVISOR=40
-- Rotating display output by 180 degrees
-- USE_DMA_TRANSFERS enabled, this improves performance. Try running CMake with -DUSE_DMA_TRANSFERS=OFF it this causes problems, or try adjusting the DMA channels to use with -DDMA_TX_CHANNEL=<num> -DDMA_RX_CHANNEL=<num>.
-- Targeting WaveShare 3.5 inch (B) display with ILI9486
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/src/fbcp-ili9341/build
~/src/fbcp-ili9341/build$ make
[  5%] Building CXX object CMakeFiles/fbcp-ili9341.dir/diff.cpp.o
[ 10%] Building CXX object CMakeFiles/fbcp-ili9341.dir/display.cpp.o
[ 15%] Building CXX object CMakeFiles/fbcp-ili9341.dir/dma.cpp.o
/home/pi/src/fbcp-ili9341/dma.cpp: In function ‘void DumpDMAPeripheralMap(’:
/home/pi/src/fbcp-ili9341/dma.cpp:173:12: warning: format ‘%p’ expects argument of type ‘void*’, but argument 6 has type ‘uint32_t’ {aka ‘unsigned int’} [-Wformat=]
     printf("DMA channel %d has peripheral map %d (is lite channel: %d, currently active: %d, current control block: %p)\n", i, (channel->cb.ti & BCM2835_DMA_TI_PERMAP_MASK) >> BCM2835_DMA_TI_PERMAP_SHIFT, (channel->cb.debug & BCM2835_DMA_DEBUG_LITE) ? 1 : 0, (channel->cs & BCM2835_DMA_CS_ACTIVE) ? 1 : 0, channel->cbAddr);
            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~                                                                                                                                                                                        ~~~~~~~~~~~~~~~
/home/pi/src/fbcp-ili9341/dma.cpp: In function ‘void CheckDMAChannelNotStolen(int, int)’:
/home/pi/src/fbcp-ili9341/dma.cpp:192:12: warning: format ‘%p’ expects argument of type ‘void*’, but argument 3 has type ‘uint32_t’ {aka ‘unsigned int’} [-Wformat=]
     printf("DMA channel collision! Some other program has submitted a DMA task to our DMA channel %d! (DMA task at unknown control block address %p)\n", channelNumber, cbAddr);
            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~                 ~~~~~~
/home/pi/src/fbcp-ili9341/dma.cpp: In function ‘void DumpDMAState()’:
/home/pi/src/fbcp-ili9341/dma.cpp:357:10: warning: format ‘%p’ expects argument of type ‘void*’, but argument 2 has type ‘uint32_t’ {aka ‘unsigned int’} [-Wformat=]
   printf("****** DMATX cbAddr: %p\n", dmaTx->cbAddr);
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~  ~~~~~~~~~~~~~
/home/pi/src/fbcp-ili9341/dma.cpp:365:10: warning: format ‘%p’ expects argument of type ‘void*’, but argument 2 has type ‘uint32_t’ {aka ‘unsigned int’} [-Wformat=]
   printf("****** DMARX cbAddr: %p\n", dmaRx->cbAddr);
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~  ~~~~~~~~~~~~~
[ 21%] Building CXX object CMakeFiles/fbcp-ili9341.dir/fbcp-ili9341.cpp.o
[ 26%] Building CXX object CMakeFiles/fbcp-ili9341.dir/gpu.cpp.o
[ 31%] Building CXX object CMakeFiles/fbcp-ili9341.dir/hx8357d.cpp.o
[ 36%] Building CXX object CMakeFiles/fbcp-ili9341.dir/ili9341.cpp.o
[ 42%] Building CXX object CMakeFiles/fbcp-ili9341.dir/ili9486.cpp.o
[ 47%] Building CXX object CMakeFiles/fbcp-ili9341.dir/keyboard.cpp.o
[ 52%] Building CXX object CMakeFiles/fbcp-ili9341.dir/mailbox.cpp.o
[ 57%] Building CXX object CMakeFiles/fbcp-ili9341.dir/mem_alloc.cpp.o
[ 63%] Building CXX object CMakeFiles/fbcp-ili9341.dir/mpi3501.cpp.o
[ 68%] Building CXX object CMakeFiles/fbcp-ili9341.dir/mz61581.cpp.o
[ 73%] Building CXX object CMakeFiles/fbcp-ili9341.dir/spi.cpp.o
/home/pi/src/fbcp-ili9341/spi.cpp: In function ‘void* spi_thread(void*)’:
/home/pi/src/fbcp-ili9341/spi.cpp:487:1: warning: no return statement in function returning non-void [-Wreturn-type]
 }
 ^
/home/pi/src/fbcp-ili9341/spi.cpp: In function ‘int InitSPI()’:
/home/pi/src/fbcp-ili9341/spi.cpp:508:10: warning: format ‘%p’ expects argument of type ‘void*’, but argument 2 has type ‘unsigned int’ [-Wformat=]
   printf("bcm_host_get_peripheral_address: %p, bcm_host_get_peripheral_size: %u, bcm_host_get_sdram_address: %p\n", bcm_host_get_peripheral_address(), bcm_host_get_peripheral_size(), bcm_host_get_sdram_address());
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~  ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/pi/src/fbcp-ili9341/spi.cpp:508:10: warning: format ‘%p’ expects argument of type ‘void*’, but argument 4 has type ‘unsigned int’ [-Wformat=]
[ 78%] Building CXX object CMakeFiles/fbcp-ili9341.dir/ssd1351.cpp.o
[ 84%] Building CXX object CMakeFiles/fbcp-ili9341.dir/st7735r.cpp.o
[ 89%] Building CXX object CMakeFiles/fbcp-ili9341.dir/statistics.cpp.o
/home/pi/src/fbcp-ili9341/statistics.cpp: In function ‘int InitStatistics(’:
/home/pi/src/fbcp-ili9341/statistics.cpp:290:23: warning: no return statement in function returning non-void [-Wreturn-type]
 int InitStatistics() {}
                       ^
[ 94%] Building CXX object CMakeFiles/fbcp-ili9341.dir/text.cpp.o
[100%] Linking CXX executable fbcp-ili9341
[100%] Built target fbcp-ili9341
```

So this is really close, but I get a bus error when running the app.

Any chance of fixing this for arm64?

Plain fbcp runs fine.

satmandu, 2019-04-13:
(Also btw the differences between waveshare35a and waveshare35b seem minimal from looking at the dts files in https://github.com/swkim01/waveshare-dtoverlays )

juj, 2019-04-18:
Running on actual ARM64 OS / the CPU/process running in 64-bit mode is actually a very different beast compared to the current 32-bit support. I am not familiar with how the 64-bit address spacing works for Pi. The BCM2835 controller itself is 32-bit hardware, so there will need to exist some kind of 64-bit virtual pages -> 32-bit physical pages translation for BCM2835 SPI and DMA controllers.

I'll retire this for the backlog, perhaps someone with intimate knowledge of Pi's 64-bit mode may be able to pick this one up in the future.