juj/fbcp-ili9341 #80: linux/gpio.h

kpishere, 2019-04-21:
juj :  Try as I might, there is something I'm just not getting about kernal mode programming.  When I add linux/gpio.h (and the necessary kernal headers) .. all goes to pot with how this project is configured.  there is some incompatibility with calling kernal headers and /opt/vc headers.

Any ideas?  I there a concept document that you're aware of that may shed some light?  My goal is to capture an upper trigger on the IRQ pin for the touch screen and then flag a need to read the SPI port in the main run-loop.

juj, 2019-04-21:
Fbcp-ili9341 assumes exclusive control to the SPI pins and the displays Data/Control pin. It is not possible to cooperatively access those pins from any other source. Fbcp-ili9341 does not do anything with the irq pin, so you can control that at will, but accessing the SPI bus is not possible unless you implement the SPI directly inside fbcp-ili9341 SPI main loop to interleave with rest of the SPI communication that fbcp-ili9341 drives.

kpishere, 2019-04-23:
Yes, that is right.  This is why I want to configure an IRQ handler within this code.  Then, change the SPI behaviour within this driver.

closed 2019-04-27 by kpishere