juj/fbcp-ili9341 #81: Kernel not compiling

kpishere, 2019-04-28:
In trying the compiler version without any changes, I'm getting the following errors.  Any pointers?

---

pi@raspberrypi:~/fbcp-ili9341/kernel $ cat Makefile 
obj-m := bcm2835_spi_display.o

CFLAGS_bcm2835_spi_display.o := -O3 -std=gnu99 -Wno-declaration-after-statement -DKERNEL_MODULE -DMPI3501=1 -DSPI_BUS_CLOCK_DIVISOR=48 -DKEDEI_V63_MPI3501=ON -DDISPLAY_ROTATE_180_DEGREES=ON  
pi@raspberrypi:~/fbcp-ili9341/kernel $ 
pi@raspberrypi:~/fbcp-ili9341/kernel $ 
pi@raspberrypi:~/fbcp-ili9341/kernel $ ./build_kernel_module.sh 
Killing existing instances of user space driver program fbcp-ili9341
Stopping kernel module bcm2835_spi_display.ko
rmmod: ERROR: Module bcm2835_spi_display is not currently loaded
make: Entering directory '/usr/src/linux-headers-4.9.35+'
  CC [M]  /home/pi/fbcp-ili9341/kernel/bcm2835_spi_display.o
In file included from ./include/linux/compiler.h:58:0,
                 from ./include/uapi/linux/stddef.h:1,
                 from ./include/linux/stddef.h:4,
                 from ./include/uapi/linux/posix_types.h:4,
                 from ./include/uapi/linux/types.h:13,
                 from ./include/linux/types.h:5,
                 from ./include/linux/buffer_head.h:10,
                 from /home/pi/fbcp-ili9341/kernel/bcm2835_spi_display.c:1:
./include/linux/compiler-gcc.h:73:17: error: expected specifier-qualifier-list before ‘inline’
 #define inline  inline  __attribute__((always_inline)) notrace
                 ^
/home/pi/fbcp-ili9341/kernel/../spi.h:124:3: note: in expansion of macro ‘inline’
   inline uint8_t *PayloadStart() { return data + (size - sizeExpandedTaskWithPadding); }
   ^
In file included from /home/pi/fbcp-ili9341/kernel/bcm2835_spi_display.c:28:0:
/home/pi/fbcp-ili9341/kernel/../spi.h: In function ‘AllocTask’:
/home/pi/fbcp-ili9341/kernel/../spi.h:342:5: error: implicit declaration of function ‘usleep’ [-Werror=implicit-function-declaration]
     usleep(100); // Since the SPI queue is full, we can afford to sleep a bit on the main thread without introducing lag.
     ^
/home/pi/fbcp-ili9341/kernel/../spi.h: In function ‘CommitTask’:
/home/pi/fbcp-ili9341/kernel/../spi.h:372:58: error: ‘SPITask’ has no member named ‘PayloadSize’
   __atomic_fetch_add(&spiTaskMemory->spiBytesQueued, task->PayloadSize()+1, __ATOMIC_RELAXED);
                                                          ^
In file included from /home/pi/fbcp-ili9341/kernel/../spi.cpp:14:0,
                 from /home/pi/fbcp-ili9341/kernel/bcm2835_spi_display.c:39:
/home/pi/fbcp-ili9341/kernel/../mailbox.h: At top level:
/home/pi/fbcp-ili9341/kernel/../mailbox.h:3:22: fatal error: inttypes.h: No such file or directory
 #include <inttypes.h>
                      ^
cc1: some warnings being treated as errors
compilation terminated.
scripts/Makefile.build:299: recipe for target '/home/pi/fbcp-ili9341/kernel/bcm2835_spi_display.o' failed
make[1]: *** [/home/pi/fbcp-ili9341/kernel/bcm2835_spi_display.o] Error 1
Makefile:1490: recipe for target '_module_/home/pi/fbcp-ili9341/kernel' failed
make: *** [_module_/home/pi/fbcp-ili9341/kernel] Error 2
make: Leaving directory '/usr/src/linux-headers-4.9.35+'
pi@raspberrypi:~/fbcp-ili9341/kernel $

juj, 2019-04-29:
Oh, the `kernel/` subdirectory contains remains of a different architectured version of the driver I was working on a longer time ago. The idea was to experiment how the driver would work if I used a kernel side module + a user-space library for the driver. Main benefit of that was that a) I'd be able to properly allocate DMA channels, and b) register and use interrupts for SPI completion (for both DMA-based SPI transfers and non-DMA based transfers). Userland side of the driver would then talk to the DispmanX API, grab GPU frames, diff them, and push the generated SPI tasks over to the kernel side driver module to drive to SPI bus.

I got as far as to get the kernel side driver + userland code running, but was not using DMA there, so got b) part done, but not a). However, the learning there was that IRQs had too much latency that the SPI bus would starve for too long time. Without using DMAs, the IRQs would also fire way too often that it would kill kernel process scheduling, and audio processing would starve. Overall performance was considerably slower than what the full userland driver was able to achieve.

When I later got DMA enabled in user space with DMA chaining, I abandoned the use of kernel side approach altogether, since using IRQs became disinteresting due to their high latency. After that change, the further work has bitrotted the code in `kernel/` subdirectory and it's not currently used. I should probably delete the code from there, but kept it around in case it'd be useful.

If you want to experiment with the kernel+user space approach, try checking out old commits with git log, to find a version where it still built. E.g. https://github.com/juj/fbcp-ili9341/commit/59b4bc51c7dad787f46f8ff5e6f6ba74f8a6aede might be a working version, though hard to say, it's been quite some time since I worked on the kernel side module.

kpishere, 2019-04-29:
Ah.  OK.  Thanks for explanation.  Hmm. Yes, don't want than many IRQs.  Really just want the touch sense IRQ to set a flag for the SPI task processor to perform a chip select and then GPIO read for the touch screen.  Way less IRQs. Will investigate.  Perhaps this will be more of a touch screen driver in kernel that just puts flag of irq to file system device, then other code checks in user land entirely.

closed 2019-05-31 by kpishere