juj/fbcp-ili9341 #82: Driver when using Freeplaytech Waveshare 32B Will Not Start or Exit

Mootikins, 2019-05-15:
Been trying to build and update our Freeplay driver, but been having two main issues.

Firstly, the driver will not run at all on current master branch unless an older version of the driver (most recent that worked correctly was ~October '18) has been run and quit before it. Possibly some initialization issues, but also started happening after we ran some updates. Will update with the packages that were updated.

Driver also will not exit cleanly at all, which makes the system shutdown slow since it will wait for the stop job to time out.

Mootikins, 2019-05-20:
Ran a git bisect, looks like commit d2f69320fb165d2ec95fa33ea4d06d865c49f56e is to blame for the broken Freeplay Tech flag. Will investigate further.

juj, 2019-05-20:
Thanks for the report. Glancing through the regressing commit, not immediately obvious what would be the bug. Unfortunately don't have the chance to debug on device at the moment. Let me know if you find something relevant.

Mootikins, 2019-07-18:
I have found some time to delve a little deeper; current master will work as expected if you checkout `spi.cpp`, `spi.h`, and `display.h` to edcdad7639faf1d6a48abcdb9a517ea319b0d2f9.

If compiled without this checkout, the driver doesn't start correctly unless you have run an older version (from the aforementioned commit) before it.

juj, 2019-07-19:
Thanks for the info. Sounds indeed like something important has regressed. Unfortunately I do not have any Raspberry Pi setups at hand atm - I am away from home for a couple of months, so unable to dig into this to verify.

Went through commit https://github.com/juj/fbcp-ili9341/commit/d2f69320fb165d2ec95fa33ea4d06d865c49f56e in more detail, and there is one change that did stand out, reverted it at https://github.com/juj/fbcp-ili9341/commit/b2eeec1de91dc6967e3331c37dcec0fcecb1b8b0 . Can you double check if that helps the issue? (I authored that change blind without being able to test on device, so not sure if that really resolves the issue)

Mootikins, 2019-07-19:
Well that fixed it, appears to be working on boot now. Have only tested on a CM3 but will test on Zero and let you know.

closed 2019-07-19 by Mootikins

juj, 2019-07-20:
Perfect, great to know that was the issue!