juj/fbcp-ili9341 #83: Issue with Waveshare 3.5(B) IPS v2

thepatchwerkboy, 2019-05-15:
Performance is divine, except that the colors are horribly incorrect, and not in a way that can be resolved by any build options found. 

Hardware:
- Raspberry Pi 3B+
- Waveshare 3.5(B) IPS screen v2 (480x360)
OS/Platform:
- RetroPie v4.4 direct image download, unmodified.
CMake Commandline:
cmake -DSPI_BUS_CLOCK_DIVISOR=10 -DWAVESHARE35B_ILI9486=ON ..

[image]

leshiy200, 2019-05-16:
Faced with the same issue, cane anyone help asap?)

leshiy200, 2019-05-16:
> Faced with the same issue, cane anyone help need asap?)

thepatchwerkboy, 2019-05-16:
Addendum - that 10 divisor is with the clock set to 300. I've tried it with the standard clock and divisors all the way up to 32. Only change is framerate - color issues remain the same throughout.

juj, 2019-05-17:
> Hardware:
> Raspberry Pi 3B+
> Waveshare 3.5(B) IPS screen v2 (480x360)

Is the display this one? https://www.waveshare.com/wiki/3.5inch_RPi_LCD_(B) . From that page, it looks like the original Rev 1.0 display that I did the initial port on is no longer sold; but not sure what the differences are between the two revisions.

The display I have was [this one](https://www.waveshare.com/wiki/3.5inch_RPi_LCD_(A)) (I think), a picture can be seen in [this video](https://www.youtube.com/watch?v=dqOLIHOjLq4) you probably saw.

I believe a divisor 32 should be very safe to avoid color issues (300/32=9.375MHz), so too high bus speed is probably not the issue here. You can try something like 50 or 100 as divisor to be extra sure - all displays should definitely manage that one. Also might be worth double checking with stock Pi clock speeds if you didn't already - to rule out any issues that clocking might have.

I wonder what controller the Waveshare 3.5(B) IPS screen v2 uses. Could not find info in Waveshare site unfortunately, though that is not that unexpected.

Any chance that you can make the display show a test image, something like https://blenderartists.org/uploads/default/original/4X/6/8/0/6802c142294826056a853cb43b4b505fe2171c25.png ? First suspect is color space: fbcp-ili9341 currently operates on RGB565 color data (conversion from Linux RGB888 framebuffer to RGB565 is done by the GPU on hardware at https://github.com/juj/fbcp-ili9341/blob/master/gpu.cpp#L464 ). Perhaps the new revision of the display is a RGB555 or a BGR555 display, hence the odd mixing of the color spaces. Rendering a test gradient of R, G, B primaries should prove or disprove that suspicion.

Controllers like ILI9486 do have an options register to control the color space: for ILI9486 that is controlled here: https://github.com/juj/fbcp-ili9341/blob/0b46b04f1d05a5f5088e750bfde28b2099b8da7d/ili9486.cpp#L41 . You can try commenting out that line, or blind trying different values to see if you might get lucky.

Ultimately if controller cannot be identified and spec sheet is not available, the only option is then to reverse engineer the original LCD-Show driver that WaveShare provides for download.

Btw, whenever changing controller in CMake (e.g. between ILI9486 and ILI9486L if trying out different options), delete CMakeCache.txt file manually (or create a different build dir altogether) - otherwise CMake does a bit bad and can still keep reusing the previous controller option.

juj, 2019-05-17:
If it turns out to be RGB555 instead of RGB565, then looking at https://github.com/raspberrypi/userland/blob/master/interface/vctypes/vc_image_types.h suggests it may be possible to substitute `VC_IMAGE_RGB565` with `VC_IMAGE_TF_RGBA5551` to have the Pi GPU generate images as 555.

Another thing is that it could be a byte order difference, i.e. the image is still RGB565, but the byte order is reversed from revision 1 (most significant byte vs least significant byte first difference). I do recall that there was a command to change the byte order in some controllers. Glancing at [ILI9486 spec](https://datasheetspdf.com/pdf-file/945603/ILITEK/ILI9486/1) did not find a command from quick read, though it might have one.

To rule out if it might be a MSB vs LSB issue, try replacing this line

https://github.com/juj/fbcp-ili9341/blob/master/fbcp-ili9341.cpp#L507

with
```c
while(x < endX) *data++ = scanline[x++];
```
i.e. take out the byte swap operation.

thepatchwerkboy, 2019-05-17:
Thanks for the tips, but no joy - I tried flipping MSB/LSB, but there was no apparent change on rebuild. Also tried about a dozen blind values for the pixelFormat constant, all with disastrous results. Interestingly enough, after setting everything back to normal, and then leaving things alone for a bit, when the software "dimmed" the screen, many of the colors went normal (see attached), so it's definitely a colorspace issue - almost as if it's running a lower bitcount per pixel, and there's an overflow.

I'm afraid that, while I am a programmer, my C++ days are twenty years behind me, and changing that VC_IMAGE_RGB556 constant is a bit lower-level than I'm familiar with in terms of where to find it and  what I'd need to rebuild.

[image]

thepatchwerkboy, 2019-05-17:
Okay, here's that image you wanted to see... it's definitely a colorspace issue...
[image]

thepatchwerkboy, 2019-05-17:
Unrelated, I just received a pair of generic 2.4" LED displays from RobotShop.com, rebuilt the driver with the -DILI9341 option, and both work phenomenally, so partial win.

juj, 2019-05-18:
The fact that the vertical dark band occurs on all color channels in the exact same spot suggests it is a gamma/color LUT table related issue. If it was a RGB565 vs RGB555 (vs BGR or hi-lo byte or RGB888 issue), any band that occurs would be very striking/harsh one pixel line full intensity<->black transition; but instead here the band gradually tapers down from full intensity to black and then back to high intensity. That is a strong hint that the color decoding LUT or gamma correction (or in a fringe case, some voltage control in init sequence) is off for revision 2 of the display.

Try to comment out these two lines:
https://github.com/juj/fbcp-ili9341/blob/0b46b04f1d05a5f5088e750bfde28b2099b8da7d/ili9486.cpp#L93-L94
or try to manually edit them to see if the artifact reacts to changes to that.

Another option may be to try to edit/comment out any other initialization sequence lines to try to see if that might get lucky and find one that causes a reaction to the colors. Unfortunately I do not have any other idea besides that except to find a logic analyzer and snoop the initialization sequence off a proper driver and reproduce that then in fbcp-ili9341.

sergiopoverony, 2019-06-10:
@juj How remove vertial lines? this is as vertical strokes

juj, 2019-06-11:
Not sure what you mean by vertical lines. Can you post a photo of the artifact?

sergiopoverony, 2019-06-11:
@juj  Static pictures - perfect, if any moving on screen... 
[image]
[image]
[image]

juj, 2019-06-11:
Oh, that is the interlacing feature that is used to improve frame rate. You can disable it by removing the comments `// ` on line

https://github.com/juj/fbcp-ili9341/blob/0b46b04f1d05a5f5088e750bfde28b2099b8da7d/config.h#L67

replace with

`#define NO_INTERLACING`

and then rebuild with `make -j`.

sergiopoverony, 2019-06-11:
@juj 
Let's sum up for others user:
**For correct color** i comment 88 and 89 lines in fbcp-ili9341/ili9486.cpp
**Help for remove lines,** #define NO_INTERLACING  in fbcp-ili9341/config.h

In 2d games. I test in Batman. (nes) i move to up in tower. Fps is down.

sergiopoverony, 2019-06-11:
@juj When you change the color, as in the video, you can see the gaps, then you can see the tearing on the ray of light - call Batman. 
Video link https://youtu.be/NawEdsVOsqs
Pictures
[image]
[image]
[image]

juj, 2019-06-11:
The tearing is due to lack of vsync support in SPI displays. It looks like the dimming of the ray of light is done by drawing the light at 100% and 0% intensity at alternating frames, resulting in average 50% brightness when vsync is present, but flickers when no vsync synchronization occurs.

You can try toggling the `DISPLAY_FLIP_ORIENTATION_IN_SOFTWARE` option, which can be found in config.h. See

https://github.com/juj/fbcp-ili9341/blob/0b46b04f1d05a5f5088e750bfde28b2099b8da7d/config.h#L180-L192

Toggling that option changes vertical tearing to diagonal tearing that looks different. In this case diagonal tearing might help smooth out the artifact a little bit. Whichever looks better is a bit of a subjective call.

sergiopoverony, 2019-06-11:
@juj  Yes, This is subjective. I think this maximum fro this scrren. You Great man. Thank you. I hope peoples can read this issue and can fix trouble.

sergiopoverony, 2019-06-12:
@juj i test stock drivers. if add speed 98000000 - work perfwctly - but color is littlebit yellow 
dtoverlay=waveshare35b:speed=98000000,fps=50

juj, 2019-06-16:
On my Waveshare 3.5 inch display I also observed that if I set too fast CDIV speed, then the display goes out of sync first via losing correct colors, and only when pushing speed even faster it loses sync completely. I believe that is due to the shift register based word reading mechanism employed by the hardware on that display. Though if I remember correctly, my Waveshare went all green colors when it was going too fast, so yellowish shade could be caused by something else as well.

One thing particularly noteworthy with the speed=xxx values with dtoverlay is that those speeds are not actual bus speeds, but the dtoverlay-based driver uses a heuristic based on that value to compute the final Clock Divisor (CDIV) register value that it will then use ([[1]](https://github.com/torvalds/linux/blob/e01e060fe00dca46959bbf055b75d9f57ba6e7be/drivers/spi/spi-bcm2835.c#L756)). The CDIV register is what specifies the bus speed in hardware. In fbcp-ili9341 I decided to just expose the CDIV register value directly and let the user choose the exact value for that, instead of relying on heuristic transformations. In dtoverlay it means that if user specifies for example 94000000, 96000000 or 98000000, they will all result in the exact same bus speed since they all round to the same CDIV value. I believe in this case, if `clk_get_rate()`([[2]](https://github.com/torvalds/linux/blob/e01e060fe00dca46959bbf055b75d9f57ba6e7be/drivers/spi/spi-bcm2835.c#L749)) returns `1200000000` (which I am not 100% sure), then speed=98000000 would end up as 1200000000/98000000 = 12.24..., which would round up to CDIV=14 getting used by the dtoverlay driver.

The reason I write "heuristic" is that the clock speed is not actually static, but varies dynamically at runtime between two power saving/speed modes of the Pi - there is idle speed at 800MHz, and turbo speed at 1200MHz (which you can see toggling if you run fbcp-ili9341 with statistics enabled). `clk_get_rate()` probably computes CDIV via the safer turbo speed to avoid overshooting the actual bus speed.

unclemarde, 2019-06-22:
Did you manage to solve it? I having the same issues with waveshare 3.5" display....

sergiopoverony, 2019-06-22:
> Did you manage to solve it? I having the same issues with waveshare 3.5" display....

Yes.

unclemarde, 2019-06-23:
i having the inverted color isseus with my waveshare 3.5" spi display and set the divisor to 12, anything below that get white screen. What should i do next? im a lil bit confuse here with all the solution.
But i did try stock driver and change the fps and speed. no joy.

sergiopoverony, 2019-06-23:
> 3.5" spi
3.5" spi IPS? WaveShares make 3.5 two models, simple TFT TN and  IPS. Here we taslk about IPS model/

mailmartinviljoen, 2019-06-29:
Hi Thanks for this thread, It solved almost all of my issues. One more thing which bothers a bit.
even after un-commenting #define NO_INTERLACING , I still seem to get interlacing, Exactly as  sergiopoverony I also used Mario64 but it happens even in the Retro pi Menu , When standing still the picture is perfect, but when moving around it seems to interlace.

PS: This is excellent work so far well done and respect!

juj, 2019-06-30:
Thanks!

Regarding the interlacing issue, I think it should definitely be handled by enabling the `#define NO_INTERLACING` option. Try double-checking that it built correctly with `make -j` after editing the change, and that you are running the right built executable, and not e.g. and earlier version or similar?

The code path for interlacing is in https://github.com/juj/fbcp-ili9341/blob/0b46b04f1d05a5f5088e750bfde28b2099b8da7d/fbcp-ili9341.cpp#L336 . You can try introducing an error to the other code paths, something like

```diff
--- a/fbcp-ili9341.cpp
+++ b/fbcp-ili9341.cpp
@@ -334,11 +334,14 @@ int main()
 
 #ifdef NO_INTERLACING
     interlacedUpdate = false;
+    #warning yay, building with no_interlacing set like I wanted
 #elif defined(ALWAYS_INTERLACING)
     interlacedUpdate = (numChangedPixels > 0);
+    #error I did not want to build with always interlacing enabled
 #else
     uint32_t bytesToSend = numChangedPixels * SPI_BYTESPERPIXEL + (DISPLAY_DRAWABLE_HEIGHT<<1);
     interlacedUpdate = ((bytesToSend + spiTaskMemory->spiBytesQueued) * spiUsecsPerByte > tooMuchToUpdateUsecs); // Decide whether to do interlacedUpdate - only updates half of the screen
+    #error I did not want to build with interlacing enabled
 #endif
 
     if (interlacedUpdate) frameParity = 1-frameParity; // Swap even-odd fields every second time we do an interlaced update (progressive updates ignore field order)
```
then reissuing `make -j` should pring the diagnostic message note (as a compiler-issued warning print), or error out if one of the undesired code paths is incorrectly being reached.

mailmartinviljoen, 2019-07-15:
Maybe my Idea of interlacing is not interlacing. :-)
Standing still it's perfectly fine
[image]

But moving around sort of creates a ghost effect. As if the pixels only gets updated afterwards the new ones came in. 
Look at the below. The guy at the bottom looks like multiple sprites. 

[image]

juj, 2019-07-16:
That does look like interlacing. Did the above not give you a non-interlaced build? Try adding also `-DSTATISTICS=1` to get the onscreen statistics - that will tell you if it was able to rebuild correctly.

mailmartinviljoen, 2019-07-16:
Unfortunately for me I'm coming right with this, Also what is a bit confusing. Further up in this thread you mentioned that // #define NO_INTERLACING  should be replaced with #define NO_INTERLACING , But by default it's already disabled. 

I always clean the build directory before rebuilding using `sudo rm -rf build` I also tried adding -NO_INTERLACE when configuring the build.
`
sudo cmake -DSPI_BUS_CLOCK_DIVISOR=14 -DWAVESHARE35B_ILI9486=ON -DDISPLAY_INVERT_COLORS=ON -DUSE_DMA_TRANSFERS=ON -DSTATISTICS=1 -DARMV8A=ON  -DDMA_TX_CHANNEL=4 -DDMA_RX_CHANNEL=5 -NO_INTERLACING ..`

adding -NO_INTERLACING to the en does not do anything. 
With DSTATISTICS=1 it doesnt have an indicator that interlacing is on or off unless you meant that I should enable disable DSTATISTICS between builds to know if the build accepted my changes.

juj, 2019-07-16:
> // #define NO_INTERLACING should be replaced with #define NO_INTERLACING , But by default it's already disabled.

By default interlacing should be enabled. It is found in

https://github.com/juj/fbcp-ili9341/blob/0b46b04f1d05a5f5088e750bfde28b2099b8da7d/config.h#L65-L68

and should be replaced with

```c
// If defined, progressive updating is always used (at the expense of slowing down refresh rate if it's 
// too much for the display to handle) 
#define NO_INTERLACING
```

(not to be confused with the other conditional directive for NO_INTERLACING that appears below)

> I also tried adding -NO_INTERLACE when configuring the build.

Yeah, that would not do anything, as interlacing is not set up to be an option in CMake, but it is configured directly in config.h

> I should enable disable DSTATISTICS between builds to know if the build accepted my changes.

Ah sorry, that was the intention indeed - toggling statistics would produce a visible change to the build to make sure that the build actually went through. Also you can try adding the `#error` directive from https://github.com/juj/fbcp-ili9341/issues/83#issuecomment-507031572 to triple check that the code is definitely getting built with `NO_INTERLACING` mode active. If that also concurs, then the artifact is due to something else.

Another thing to try may be to remove the line

https://github.com/juj/fbcp-ili9341/blob/0b46b04f1d05a5f5088e750bfde28b2099b8da7d/config.h#L191

which should cause the display to update in the opposite order (horizontal instead of vertical). That affects the visual quality of the updating artifact that may look more pleasing (or not, depending on subjective aesthetics).

mailmartinviljoen, 2019-07-16:
Ok thanks I think it's best I start from scratch, I may have messed with the config.h file too much

mailmartinviljoen, 2019-07-16:
Solved! . I feel like a total moron now :-) I was commenting out the "conditional directive for NO_INTERLACING" as mentioned by you.

Thanks for your patience

daviel, 2020-05-08:
> Solved! . I feel like a total moron now :-) I was commenting out the "conditional directive for NO_INTERLACING" as mentioned by you.
> 
> Thanks for your patience

I think this issue can be closed

I-like-stuff, 2020-05-21:
Is anyone here able to share how they managed to set up the display as I am not quite able to make it up from the comments.

juj, 2020-12-08:
Closing this as resolved. @I-like-stuff If you are still actively struggling, please let's follow up in a new issue thread, with the steps on what you have tried so far, what hardware you have, and what didn't work, etc.

closed 2020-12-08 by juj